Koenkk/zigbee2mqtt #7564: ZM-RT201 PIR  Motion Sensor

frankiha, 2021-05-22:
## Information about the device + link
https://www.aliexpress.com/item/1005002420495341.html?spm=a2g0s.9042311.0.0.248d4c4dEHPA1V
## data/database.db entry of the device
{"id":5,"type":"EndDevice","ieeeAddr":"0x00124b0005ad88b1","nwkAddr":52292,"manufId":0,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":160}},"ssIasZone":{"attributes":{"zoneStatus":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1621707189864}

bootuz-dinamon, 2021-05-23:
I also received this device today. The box says ZM-RT-201, the instructions are ZM-35ZH-Q. And he also did not want to connect - he writes that the device is unsupported.

I read on this forum that some devices throw out the modelid and manufacturerName too quickly and the zigbee2mqtt does not have time to receive this data.

But after numerous attempts to connect the device, one day instead of "undefined" I saw TS0202 and _TYZB01_dl7cejts.

Now trying to make a converter.
```

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const motionDetector = {
    cluster: 'ssIasZone',
    type: 'attributeReport',
    convert: (model, msg, publish, options, meta) => {
        const zoneStatus = msg.data.zonestatus;
        return {
            occupancy: (zoneStatus & 1) > 0,
            tamper: (zoneStatus & 1<<2) > 0,
            battery_low: (zoneStatus & 1<<3) > 0,
        };
    },
};

 const definition = {
    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TYZB01_dl7cejts'}],
    zigbeeModel: ['TS0202'],
    model: 'TS0202',
    vendor: 'TuYa',
    description: 'Motion sensor',
    fromZigbee: [motionDetector],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.tamper()],
};

module.exports = definition;
```

Atheos46, 2021-05-23:
Hello,

I received these sensors ZM-RT201 today but I can't get it to learn my zigbee. How can I do this? Am new and can not code. I have Proxmox and IoBroker running as a VM and have the CC26X2R1. All zigbee devices work without problems.

bootuz-dinamon, 2021-05-24:
I get this message when the motion sensor is triggered:

`Debug Received Zigbee message from '0x00124b0005b4bd43', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":1}' from endpoint 1 with groupID 0`

But I don't have enough knowledge to write a converter, and even with a timer.
But the device can work with a zigbee2mqtt.

MarcelJ28, 2021-05-24:
I also have this item. I hope it can be included so I can use it.

bootuz-dinamon, 2021-05-24:
it is not a complete converter, but it is enough for the device to work.
The delay is set to 10 seconds.
Be sure to replace the ieee_address in the code with the ieee_address of your device.

```

//       Converter for TS004F.        https://www.youtube.com/channel/UCJRaKbgv8u1O4OwG-0D3uRw
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;

const motionSensorDetector = {
    cluster: 'ssIasZone',
    type: ['attributeReport'],
    convert: (model, msg, publish, options, meta) => {
            if (msg.data.zoneStatus !== 1) {
                // In case of 0 no occupancy is reported.
                // https://github.com/Koenkk/zigbee2mqtt/issues/467
                return;
            }

            // The occupancy sensor only sends a message when motion detected.
            // Therefore we need to publish the no_motion detected by ourselves.
            const timeout = options && options.hasOwnProperty('occupancy_timeout') ?
                options.occupancy_timeout : 10;   // occupancy delay in seconds

            // Stop existing timers because motion is detected and set a new one.
            globalStore.getValue(msg.endpoint, 'timers', []).forEach((t) => clearTimeout(t));
            globalStore.putValue(msg.endpoint, 'timers', []);

            if (timeout !== 0) {
                const timer = setTimeout(() => {
                    publish({occupancy: false});
                }, timeout * 1000);

                globalStore.getValue(msg.endpoint, 'timers').push(timer);
            }

            // No occupancy since
            if (options && options.no_occupancy_since) {
                options.no_occupancy_since.forEach((since) => {
                    const timer = setTimeout(() => {
                        publish({no_occupancy_since: since});
                    }, since * 1000);
                    globalStore.getValue(msg.endpoint, 'timers').push(timer);
                });
            }

            if (options && options.no_occupancy_since) {
                return {occupancy: true, no_occupancy_since: 0};
            } else {
                return {occupancy: true};
            }
        
            }
};

 const definition = {
//    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TYZB01_dl7cejts'}],
    fingerprint: [{ieee_address: '0x00124b0005b4bd43'}],   // WARNING! Change this ieee_address for yours ieee_address!!!!
    zigbeeModel: ['TS0202'],
    model: 'TS0202',
    vendor: 'TuYa',
    description: 'Motion sensor',
    fromZigbee: [ motionSensorDetector ],
    toZigbee: [],
    exposes: [e.occupancy()],
};

module.exports = definition;
```

Koenkk, 2021-05-24:
Can you try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TYZB01_dl7cejts'}],
    zigbeeModel: ['TS0202'],
    model: 'TS0202',
    vendor: 'TuYa',
    description: 'Motion sensor',
    fromZigbee: [fz.ias_occupancy_alarm_1],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.tamper()],
};
```

If the modelID of your device is `undefined`, try pairing it again close to the coordinator.

bootuz-dinamon, 2021-05-24:
Unfortunately, the number of attempts to connect and get the correct modelID has reached 50-60.
and I honestly tried to do as you wrote, but either something is wrong or I don't have enough experience.

The ias_occupancy_alarm_1 converter has cluster: 'ssIasZone', and type: 'commandStatusChangeNotification'.
And this device  cluster: 'ssIasZone', and type: 'attributeReport'.

But I'll try again.
And z2m is 1.18.3-dev

Koenkk, 2021-05-24:
@bootuz-dinamon could you provide the herdsman debug log when pairing the device?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Atheos46, 2021-05-24:
I always get these messages in the log:

zigbee.0 | 2021-05-24 23:22:57.331 | info | (9750) Zigbee: stop joining
-- | -- | -- | --
zigbee.0 | 2021-05-24 23:22:28.325 | warn | (9750) Device '0x00124b00060ea04e' announced itself
zigbee.0 | 2021-05-24 23:22:27.581 | info | (9750) Starting interview of '0x00124b00060ea04e'
zigbee.0 | 2021-05-24 23:21:56.101 | info | (9750) Zigbee: allowing new devices to join.

MarcelJ28, 2021-05-25:
The one from  bootuz-dinamon works for me! The only thing that is not working is the battery level but for this moment I am happy.

Atheos46, 2021-05-25:
> The one from bootuz-dinamon works for me! The only thing that is not working is the battery level but for this moment I am happy.

Hey Marcel,
for me it would also be enough if it works. The battery display is not important for now. What exactly do I have to do? Have really no experience at all what I should do with the code? Would be happy if you help me.

MarcelJ28, 2021-05-25:
> Hey Marcel,
> for me it would also be enough if it works. The battery display is not important for now. What exactly do I have to do? Have really no experience at all what I should do with the code? [...]

Well I have only nailed it after very much searching and I am not sure anymore what I exactly did at the beginning. So I am sorry I cannot help you....

bootuz-dinamon, 2021-05-25:
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

chrisrock1984, 2021-05-25:
> it is not a complete converter, but it is enough for the device to work.
> The delay is set to 10 seconds.
> Be sure to replace the ieee_address in the code with the ieee_address of your device.
>
> ``` [...]

works for me. I was also not able to fully interview the PIR on 1.8.3 dev even on very close distance.

Atheos46, 2021-05-25:
> Well I have only nailed it after very much searching and I am not sure anymore what I exactly did at the beginning. So I am sorry I cannot help you....

Thanks anyway. I'll have to keep looking then.

> https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
> 
> This is an instruction for adding external converters.

I have already looked at this but unfortunately did not understand it. When I get off work, I'll have another look at it. I just do not know how to integrate the code with the CC26X2R1 what is plugged with usb on NUC Proxmox.

Koenkk, 2021-05-25:
Added the device. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Koenkk, 2021-05-25:
@Schnittcher if you see TS0202 _TYZB01_dl7cejts the model ID isn't undefined since that is the model, or do you mean something else?

Schnittcher, 2021-05-25:
I have repaired the device.
Yesterday i get the model and the manufacture. Today i get model and manufacture as undefined.
When i try to pair the device with tasmota, i get the model id and the manufacture.

So i think this is a problem with zigbee2mqtt. I have a debug: https://pastebin.com/xZFnh7gP

Can you help?

Koenkk, 2021-05-25:
@Schnittcher the log doesn't reveal much (device doesn't event connect). Can you make a sniff of the pairing attempt? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

Schnittcher, 2021-05-25:
very interesting, now the device has been found again!

I will try your new version. I used docker is this already online?

Koenkk, 2021-05-25:
Will be available in 1 hour from now (use latest-dev tag)

Schnittcher, 2021-05-25:
Yes this i know.
When i have this problem again, can i asked you again?

bootuz-dinamon, 2021-05-25:
Already with a battery (you need to remove the battery and insert it back).
This is not yet an official solution to the issue.
 If you still need an extended log, tell me, I will.
 
```
//       Converter for ZM-RT201 ZM-35ZH-Q with battery now.        https://www.youtube.com/channel/UCJRaKbgv8u1O4OwG-0D3uRw
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;

const motionSensorDetector = {
    cluster: 'ssIasZone',
    type: ['attributeReport'],
    convert: (model, msg, publish, options, meta) => {
            if (msg.data.zoneStatus !== 1) {
                // In case of 0 no occupancy is reported.
                // https://github.com/Koenkk/zigbee2mqtt/issues/467
                return;
            }

            // The occupancy sensor only sends a message when motion detected.
            // Therefore we need to publish the no_motion detected by ourselves.
            const timeout = options && options.hasOwnProperty('occupancy_timeout') ?
                options.occupancy_timeout : 10;   // occupancy delay in seconds

            // Stop existing timers because motion is detected and set a new one.
            globalStore.getValue(msg.endpoint, 'timers', []).forEach((t) => clearTimeout(t));
            globalStore.putValue(msg.endpoint, 'timers', []);

            if (timeout !== 0) {
                const timer = setTimeout(() => {
                    publish({occupancy: false});
                }, timeout * 1000);

                globalStore.getValue(msg.endpoint, 'timers').push(timer);
            }

            // No occupancy since
            if (options && options.no_occupancy_since) {
                options.no_occupancy_since.forEach((since) => {
                    const timer = setTimeout(() => {
                        publish({no_occupancy_since: since});
                    }, since * 1000);
                    globalStore.getValue(msg.endpoint, 'timers').push(timer);
                });
            }

            if (options && options.no_occupancy_since) {
                return {occupancy: true, no_occupancy_since: 0};
            } else {
                return {occupancy: true};
            }
        
            }
};

 const definition = {
//    fingerprint: [{modelID: 'TS0202', manufacturerName: '_TYZB01_dl7cejts'}],
    fingerprint: [{ieee_address: '0x00124b0005b4bd43'}],   // WARNING! Change this ieee_address for yours ieee_address!!!!
    zigbeeModel: ['TS0202'],
    model: 'TS0202',
    vendor: 'TuYa',
    description: 'Motion sensor',
    fromZigbee: [ motionSensorDetector, fz.battery, fz.ignore_basic_report ],
    toZigbee: [],
    exposes: [e.occupancy(), e.battery_low(), e.battery()],
};

module.exports = definition;
```

Schnittcher, 2021-05-25:
@bootuz-dinamon, with your code, the pir works.
So i get the occupancy value.
But after it clears, i get a long time no new occupied value.
Do you know how long is the timeout?

But the interview of this device failed again.

bootuz-dinamon, 2021-05-25:
options.occupancy_timeout : 10;   // occupancy delay in seconds

Schnittcher, 2021-05-25:
Thanks, so I must find out which timeout is the right one?
Is it possible to set this settings via frontend?

Thanks for your help! :-)

guilhermeaiolfi, 2021-05-26:
The interview fails for me too. But the device is working, so, thanks for the code.

chrisrock1984, 2021-05-26:
> The interview fails for me too. But the device is working, so, thanks for the code.

Same for me. It works with the converter but the interview completed status is still false. And yes, I have updated to the latest version.

bootuz-dinamon, 2021-05-26:
> Thanks, so I must find out which timeout is the right one?
> Is it possible to set this settings via frontend?
> 
> Thanks for your help! :-)

No. You can change timeout only in converter.

Schnittcher, 2021-05-26:
Now i get sometimes this:
`Debug No converter available for 'TS0202' with cluster 'ssIasZone' and type 'attributeReport' and data '{"zoneStatus":1}'`

So I think sometimes zigbee2mqtt found the converter and sometimes not. Can someone explain that?

Schnittcher, 2021-05-26:
And my Coordinator is now a TS0202 :-(

[image]
Can I undo this?

bootuz-dinamon, 2021-05-26:
i have exactly the same problem.
maybe I made a mistake in the converter.

haverland, 2021-05-26:
Hi bootuz-dinamon, thx!
Works for my ZM-35ZH-Q. 

Finds the device, occupancy and link quality is fine.
But batteylevel and battery_low is null:

[image]

Schnittcher, 2021-05-26:
@bootuz-dinamon, thanks!
Now my coordinator is "not supported".

[image]

bootuz-dinamon, 2021-05-26:
> @bootuz-dinamon, thanks!
> Now my coordinator is "not supported".
> 

With last converter?
Turn off it.

bootuz-dinamon, 2021-05-26:
> Finds the device, occupancy and link quality is fine.
> But batteylevel and battery_low is null:

try removing the battery and putting it back in. Or wait a couple of hours.

Schnittcher, 2021-05-26:
@bootuz-dinamon i think, that is right.
Ii think this coordinator was not supported?
It is the Slaesh's CC2652RB stick coordinator.

Or is something wrong?

bootuz-dinamon, 2021-05-27:
> @bootuz-dinamon i think, that is right.
> Ii think this coordinator was not supported?
> It is the Slaesh's CC2652RB stick coordinator.
> 
> Or is something wrong?

I have cc2562 stick too. Work for me.

Schnittcher, 2021-05-27:
Can you send a screenshot?
I think this was from day one.

bootuz-dinamon, 2021-05-27:
[image]
[image]

Schnittcher, 2021-05-27:
Thanks. And can you explain, why the interview was not completed?

bootuz-dinamon, 2021-05-27:
This is from my database.db

`{"id":9,"type":"EndDevice","ieeeAddr":"0x00124b0005b4bd43","nwkAddr":42294,"manufId":0,"epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280],"outClusterList":[0,3,1,1280],"clusters":{"ssIasZone":{"attributes":{"zoneStatus":0}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":160}}},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1622097210361}`

As you see - manufid is undefined, manufacturerName is undefined.
This device can't complite interview with coordinator.

Schnittcher, 2021-05-27:
This is from my database.db:

`{"id":54,"type":"EndDevice","ieeeAddr":"0x00124b0005b4ca90","nwkAddr":50314,"manufId":0,"manufName":"_TYZB01_dl7cejts","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280],"outClusterList":[0,3,1,1280],"clusters":{"ssIasZone":{"attributes":{"zoneStatus":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":33,"hwVersion":0,"dateCode":"20060831        ","zclVersion":3,"interviewCompleted":false,"meta":{},"lastSeen":1622097433020}`

Schnittcher, 2021-05-27:
@Koenkk, can you add this device?
I think this is missing:

`const motionSensorDetector = {
    cluster: 'ssIasZone',
    type: ['attributeReport'],
    convert: (model, msg, publish, options, meta) => {
            if (msg.data.zoneStatus !== 1) {
                // In case of 0 no occupancy is reported.
                // https://github.com/Koenkk/zigbee2mqtt/issues/467
                return;
            }

            // The occupancy sensor only sends a message when motion detected.
            // Therefore we need to publish the no_motion detected by ourselves.
            const timeout = options && options.hasOwnProperty('occupancy_timeout') ?
                options.occupancy_timeout : 60;   // occupancy delay in seconds

            // Stop existing timers because motion is detected and set a new one.
            globalStore.getValue(msg.endpoint, 'timers', []).forEach((t) => clearTimeout(t));
            globalStore.putValue(msg.endpoint, 'timers', []);

            if (timeout !== 0) {
                const timer = setTimeout(() => {
                    publish({occupancy: false});
                }, timeout * 1000);

                globalStore.getValue(msg.endpoint, 'timers').push(timer);
            }

            // No occupancy since
            if (options && options.no_occupancy_since) {
                options.no_occupancy_since.forEach((since) => {
                    const timer = setTimeout(() => {
                        publish({no_occupancy_since: since});
                    }, since * 1000);
                    globalStore.getValue(msg.endpoint, 'timers').push(timer);
                });
            }

            if (options && options.no_occupancy_since) {
                return {occupancy: true, no_occupancy_since: 0};
            } else {
                return {occupancy: true};
            }
        
            }
};`

Thanks for your work. :)

Koenkk, 2021-05-27:
@Schnittcher did you test with the latest dev branch (without any external converter)? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html). Let me know if you get an `occupancy: false` event after some time, if I will add that code.

bootuz-dinamon, 2021-05-27:
> @Schnittcher did you test with the latest dev branch (without any external converter)? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html). Let me know if you get an `occupancy: false` event after some time, if I will add that code.

I get this after 60s.

```
Debug Received Zigbee message from '0x00124b0005b4bd43', type 'attributeReport', cluster 'ssIasZone', data '{"zoneStatus":0}' from endpoint 1 with groupID 0
Warning Received message from unsupported device with Zigbee model 'undefined' and manufacturer name 'undefined'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

But my device still " model 'undefined' and manufacturer name 'undefined'".
It seems to me that the manufacturer did not think for a long time and took part of the code from the leakage sensor.

Because with this function it was possible to make a simpler converter:

`ias_water_leak_alarm_1_report: {        cluster: 'ssIasZone',        type: 'attributeReport',        convert: (model, msg, publish, options, meta) =&gt; {            const zoneStatus = msg.data.zoneStatus;            return {                water_leak: (zoneStatus &amp; 1) &gt; 0,                tamper: (zoneStatus &amp; 1&lt;&lt;2) &gt; 0,                battery_low: (zoneStatus &amp; 1&lt;&lt;3) &gt; 0,            };        },    },`

bootuz-dinamon, 2021-05-27:
https://pastebin.com/E7dVu83P

This is my zigbee-herdsman-log.

freitas666, 2021-05-27:
> //       Converter for ZM-RT201 ZM-35ZH-Q with battery now.        https://www.youtube.com/channel/UCJRaKbgv8u1O4OwG-0D3uRw
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const globalStore = require('zigbee-herdsman-converters/lib/store');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const motionSensorDetector = {
>     cluster: 'ssIasZone',
>     type: ['attributeReport'],
>     convert: (model, msg, publish, options, meta) => {
>             if (msg.data.zoneStatus !== 1) {
>                 // In case of 0 no occupancy is reported.
>                 // https://github.com/Koenkk/zigbee2mqtt/issues/467
>                 return;
>             }
> 
>             // The occupancy sensor only sends a message when motion detected.
>             // Therefore we need to publish the no_motion detected by ourselves.
>             const timeout = options && options.hasOwnProperty('occupancy_timeout') ?
>                 options.occupancy_timeout : 60;   // occupancy delay in seconds
> 
>             // Stop existing timers because motion is detected and set a new one.
>             globalStore.getValue(msg.endpoint, 'timers', []).forEach((t) => clearTimeout(t));
>             globalStore.putValue(msg.endpoint, 'timers', []);
> 
>             if (timeout !== 0) {
>                 const timer = setTimeout(() => {
>                     publish({occupancy: false});
>                 }, timeout * 1000);
> 
>                 globalStore.getValue(msg.endpoint, 'timers').push(timer);
>             }
> 
>             // No occupancy since
>             if (options && options.no_occupancy_since) {
>                 options.no_occupancy_since.forEach((since) => {
>                     const timer = setTimeout(() => {
>                         publish({no_occupancy_since: since});
>                     }, since * 1000);
>                     globalStore.getValue(msg.endpoint, 'timers').push(timer);
>                 });
>             }
> 
>             if (options && options.no_occupancy_since) {
>                 return {occupancy: true, no_occupancy_since: 0};
>             } else {
>                 return {occupancy: true};
>             }
>         
>             }
> };
> 
>  const definition = {
> //   fingerprint: [{modelID: 'TS0202', manufacturerName: '_TYZB01_dl7cejts'}],
>     fingerprint: [{type: 'EndDevice', ieee_address: '0x00124b0005b4bd43'}],   // WARNING! Change this ieee_address for yours ieee_address!!!!
>     zigbeeModel: ['TS0202'],
>     model: 'ZM-RT201',
>     vendor: 'TuYa',
>     description: 'Motion sensor',
>     fromZigbee: [ motionSensorDetector, fz.battery ],
>     //fromZigbee: [ motionSensorDetector, fz.battery, fz.ignore_basic_report ],
>     toZigbee: [],
>     exposes: [e.occupancy(), e.battery_low(), e.battery()],
> };
> 
> module.exports = definition;

@bootuz-dinamon i put this converter and works... i mean, works but show a exclamation on icon! Like screenshot in attached. Have you some ideia about? thank you!

[image]

bootuz-dinamon, 2021-05-27:
this icon only indicates that the device has not passed the interview. But this does not prevent him from working.

freitas666, 2021-05-27:
> 
> 
> this icon only indicates that the device has not passed the interview. But this does not prevent him from working.

And how can i fix it?

Schnittcher, 2021-05-27:
> @Schnittcher did you test with the latest dev branch (without any external converter)? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html). Let me know if you get an `occupancy: false` event after some time, if I will add that code.

I think the Pir works, but:
[image]

Koenkk, 2021-05-28:
@bootuz-dinamon / @Schnittcher  thanks for the log, but I need the full herdsman debug log starting from putting the device in pairing mode until you get interview failed (this causes the exclamation mark). 

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

bootuz-dinamon, 2021-05-28:
I tried to remove such a log, but some of the data is lost. Is it possible to enable writing to the file of such a log and write it either to a separate log, or to /zigbee2mqtt/log/data/log.txt?

Koenkk, 2021-05-28:
If you are using the hass addon, the easiest is to SSH into your machine, use `docker ps -a` to figure out the contained ID, put the device into pairing mode and wait until interview fails. Then save the log with `docker logs CONTAINER_ID > log.txt 2>&1`

freitas666, 2021-05-28:
@Koenkk  this command will help me too ... ! thank you!

Schnittcher, 2021-05-28:
@Koenkk, is this good?
https://pastebin.com/1shDTpFT

Thanks for your help!

Koenkk, 2021-05-29:
Fixed the failed to interview + `https://github.com/Koenkk/zigbee2mqtt/issues/7564`

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

bootuz-dinamon, 2021-05-30:
https://pastebin.com/AzaEHT4Q

This my zigbee-heardsman log.
Device still not working without converter.

chrisrock1984, 2021-05-31:
> https://pastebin.com/AzaEHT4Q
> 
> 
> 
> This my zigbee-heardsman log.
> 
> Device still not working without converter.  

Same for me
Cannot finish interview and it only works with an external converter

Koenkk, 2021-05-31:
@bootuz-dinamon @chrisrock1984 have you tried a few times and also close to the coordinator?

[42 more replies]

closed 2021-09-10 by github-actions[bot]