Koenkk/zigbee2mqtt #7567: 'Programming event notification command' seems to miss a value (?)

TheStigh, 2021-05-22:
On my lock I'm adding SERVICE MODE where the lock will generate a random key and return  as part of a 'Programming event notification command'. SET seems to do what it is supposed to, and a 'Programming event notification command' are returned with following values (I've formatted the JSON):
```
Received Zigbee message from '0x680ae2fffe6bb522', type 'commandProgrammingEventNotification', cluster 'closuresDoorLock', data '
{
   "data":255,
   "pin":4,
   "programeventcode":4,
   "programeventsrc":0,
   "userid":108,
   "userstatus":54,
   "usertype":53,
   "zigbeelocaltime":16856113
}
' from endpoint 1 with groupID 0
```

<span class="fontstyle0">A door lock programming event notification has the following format.<br></span>
Octets | 1 | 1 | 2 | 1 | 1 | 1 | 4 | Variable/0
-- | -- | -- | -- | -- | -- | -- | -- | --
Data Type | uint8 | uint8 | uint16 | uint8 | enum8 | uint8 | uint32 | string
Field Name | ProgramEvent Source | ProgramEvent Code | User ID | PIN | UserType | UserStatus | ZigBeeLocalTime | Data

<span class="fontstyle0">The fields have the following values:<br>- The program event source and program event code fields are described below.<br>- The User ID field contains the index, see chapter 3.4. In the case of duplicate code or if all user<br>pins are deleted this field is set 0x00.<br>- **The PIN contains the set pin, the format is described in chapter 3.7.4.1.**<br>- The values for User type and User status fields are described in chapter 3.7.4.2.<br>- The ZigBeeLocalTime is set to value 0xFFFFFFFF for all reports.<br>- The Data field is the same as PIN.</span> 
<br style=" font-style: normal; font-variant: normal; font-weight: normal; letter-spacing: normal; line-height: normal; orphans: 2; text-align: -webkit-auto; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-size-adjust: auto; -webkit-text-stroke-width: 0px; ">

**The PIN contains the set pin, the format is described in chapter 3.7.4.1.(7.3.2.16.27.4 in ZCL)**

<span class="fontstyle0">The PIN has the following format<br></span>
Octals | 1 | 4-10
-- | -- | --
Code length | Pin or RFID code

<span class="fontstyle0">Where the code length is a number between 4 and 10.<br>The code is ASCII encoded so the code 1234 is encoded as 0x31 32 33 34</span> 
<br style=" font-style: normal; font-variant: normal; font-weight: normal; letter-spacing: normal; line-height: normal; orphans: 2; text-align: -webkit-auto; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-size-adjust: auto; -webkit-text-stroke-width: 0px; ">

It return 4 which is the PIN Length, but no user PIN that are assigned.

What am I missing?

Koenkk, 2021-05-23:
Can you provide the herdman debug log when receiving the `commandProgrammingEventNotification` ?

TheStigh, 2021-05-23:
Here it is

[Herdsman debug log](https://pastebin.com/0G8UdzE9)

Koenkk, 2021-05-24:
Think I found the issue, `commandProgrammingEventNotification` should return the pin now. Can you check?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

TheStigh, 2021-05-24:
While I wait for the dev to update;

What about the recurring configuration - whenever I do something from Z2M to the lock, it starts to reconfigure the devic?

And the logs provided for the SET/GET pin ?

Koenkk, 2021-05-25:
- Regarding configure, can you: after seeing `Successfully configured '0x680ae2fffe6bb522'`, stop z2m, and save the database.db entry of this device, then start z2m again and trigger another configure, stop z2m and save the database.db entry again. Then post both entries here.
- get/set thing should be fixed. Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

TheStigh, 2021-05-26:
Here are the two database entries you asked for:

This is after pairing from clean database:
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e3012c","nwkAddr":0,"manufId":0,"epList":[242,47,13,12,110,11,10,8,6,5,4,3,2,1],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null}
{"id":2,"type":"EndDevice","ieeeAddr":"0x680ae2fffe6bb522","nwkAddr":2745,"manufId":4919,"manufName":"Datek","powerSource":"Battery","modelId":"ID Lock 150","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,257],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"ID Lock 150","manufacturerName":"Datek","powerSource":3,"zclVersion":3,"appVersion":7,"stackVersion":65,"swBuildId":"0.7"}},"closuresDoorLock":{"attributes":{"16384":1,"16385":1,"16388":0,"16389":0,"lockState":2,"soundVolume":2,"doorState":0}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":140}}},"binds":[{"cluster":257,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e3012c","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e3012c","endpointID":1}],"configuredReportings":[{"cluster":257,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":7,"stackVersion":65,"swBuildId":"0.7","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1621986537453}
```
Stopped Z2M, did a restart as instructed and triggered the  `Configuring '0x680ae2fffe6bb522'` by sending LOCK to the lock:
```
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0018e3012c","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"conf [... cut ...] d":"0.7","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1621987385105}
```

[Here are also the log](https://pastebin.com/mLvTTzJR) from pairing until `Successfully configured '0x680ae2fffe6bb522`
[Here are the log](https://pastebin.com/cqkQMwYQ) after restart and trigger a `Configuring '0x680ae2fffe6bb522'`
[And here are the log](https://pastebin.com/zDQuZUfJ) after a new restart and a fully  `Configuring '0x680ae2fffe6bb522'` by itself without any intervention by me

Koenkk, 2021-05-26:
Are you sure your external converter is **not** in place anymore and that you are on the latest dev branch? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

TheStigh, 2021-05-26:
> 
> 
> Are you sure your external converter is **not** in place anymore and that you are on the latest dev branch? 

Sorry, I don't what you mean with "external converter is **not** in place anymore"

I'm running the Edge addon, did an uninstall, reload and re-install last night of the Edge.

Koenkk, 2021-05-27:
You don't need the `idlock.js` anymore, everything is integrated already.

TheStigh, 2021-05-27:
> 
> 
> You don't need the `idlock.js` anymore, everything is integrated already (so remove it from your configuration.yaml). Also make sure you are using the correct HA addon repo: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt#installation

Ehhh? I don't need the local idlock.js when adding new functionality to the existing device? Now I'm totally off... 
Existing device for release:
```
    {
        zigbeeModel: ['ID Lock 150'],
        model: '0402946',
        vendor: 'Datek',
        description: 'Zigbee module for ID lock 150',
        fromZigbee: [fz.lock, fz.battery, fz.lock_operation_event, fz.idlock, fz.lock_programming_event, fz.lock_pin_code_response],
        toZigbee: [tz.lock, tz.lock_sound_volume, tz.idlock_master_pin_mode, tz.idlock_rfid_enable,
            tz.idlock_lock_mode, tz.idlock_relock_enabled],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            const options = {manufacturerCode: 4919};
            await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
            await reporting.lockState(endpoint);
            await endpoint.read('closuresDoorLock', ['lockState', 'soundVolume', 'doorState']);
            await endpoint.read('closuresDoorLock', [0x4000, 0x4001, 0x4004, 0x4005], options);
            await reporting.batteryPercentageRemaining(endpoint);
        },
        exposes: [e.lock(), e.battery(),
            exposes.enum('sound_volume', ea.ALL, constants.lockSoundVolume).withDescription('Sound volume of the lock'),
            exposes.binary('master_pin_mode', ea.ALL, true, false).withDescription('Allow master pin unlock'),
            exposes.binary('rfid_enable', ea.ALL, true, false).withDescription('Allow RFID to unlock'),
            exposes.enum('lock_mode', ea.ALL, ['auto_off_away_off', 'auto_on_away_off', 'auto_off_away_on',
                'auto_on_away_on']).withDescription('Lock Mode of the lock'),
            exposes.binary('relock_enabled', ea.ALL, true, false).withDescription( 'Allow auto re-lock')],
    },
```

Extended functionality I'm currently **_working on_** (and I assume I then need the local file, with extra converters, yes?):
```
const device = {
    zigbeeModel: ['ID Lock 150'],
    model: '0402946',
    vendor: 'Datek',
    description: 'Zigbee module for ID lock 150',
    fromZigbee: [fz.lock, fz.battery, fz.lock_operation_event, fz.lock_programming_event, 
        fzLocal.idlock, fzLocal.idlock_fw, fz.lock_pin_code_response],
    toZigbee: [tz.lock, tz.lock_sound_volume, tzLocal.idlock_master_pin_mode, tzLocal.idlock_rfid_enable,
        tzLocal.idlock_service_mode, tzLocal.idlock_lock_mode, tzLocal.idlock_relock_enabled, tz.pincode_lock],
    meta: {pinCodeCount: 109},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const options = {manufacturerCode: 4919};
        await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
        await reporting.lockState(endpoint);
        await reporting.batteryPercentageRemaining(endpoint);
        const payload = [{
            attribute: {ID: 0x4000, type: 0x10},
            minimumReportInterval: 0,
            maximumReportInterval: repInterval.HOUR,
            reportableChange: 1,
        },
        {
            attribute: {ID: 0x4001, type: 0x10},
            minimumReportInterval: 0,
            maximumReportInterval: repInterval.HOUR,
            reportableChange: 1,
        },
        {
            attribute: {ID: 0x4003, type: 0x20},
            minimumReportInterval: 0,
            maximumReportInterval: repInterval.HOUR,
            reportableChange: 1,
        },
[... 20 lines cut ...]
            data.cluster === 'closuresDoorLock' &&
            data.data &&
            data.data.userid !== undefined &&
            // Don't read RF events, we can do this with retrieve_state
            (data.data.programeventsrc === undefined || constants.lockSourceName[data.data.programeventsrc] != 'rf')
        ) {
            await device.endpoints[0].command('closuresDoorLock', 'getPinCode', {userid: data.data.userid}, {});
        }
    },
    exposes: [e.lock(), e.battery(),
        exposes.enum('sound_volume', ea.ALL, constants.lockSoundVolume).withDescription('Sound volume of the lock'),
        exposes.binary('master_pin_mode', ea.ALL, true, false).withDescription('Allow Master PIN Unlock'),
        exposes.binary('rfid_enable', ea.ALL, true, false).withDescription('Allow RFID to Unlock'),
        exposes.enum('lock_mode', ea.ALL, ['auto_off_away_off', 'auto_on_away_off', 'auto_off_away_on',
            'auto_on_away_on']).withDescription('Lock-Mode of the Lock'),
        exposes.enum('service_mode', ea.ALL, ['deactivated', 'random_pin_1x_use',
            'random_pin_24_hours']).withDescription('Service Mode of the Lock'),
        exposes.binary('relock_enabled', ea.ALL, true, false).withDescription( 'Allow Auto Re-Lock')],
};
```

And [this is the full file](https://pastebin.com/vZXWGUkj) with the NEW converters added

And yes Koen, I'm using the correct addon.
[image]

Koenkk, 2021-05-27:
The problem is that there are now 2 definitions for the `ID Lock 150` (the one from datatek.js and your external converter). I believe this is were things go wrong. I recommend to temporarily change the  `modelID` in your `database.db` from `ID Lock 150` to `ID Lock 150 CUSTOM`, in your external converter change `zigbeeModel: ['ID Lock 150'],` to `zigbeeModel: ['ID Lock 150 CUSTOM'],`.

TheStigh, 2021-05-27:
And as usual you're perfectly right! I assumed that if it existed locally it would 'overwrite' the online version. Now, the `configuring...` do not occur on usage. 

Though, now when I pair the device, the new converters/functionalities are not 'loaded' and thus unavailable...?

TheStigh, 2021-05-27:
And as usual you're perfectly right! I assumed that if it existed locally it would 'overwrite' the online version. Now, the `configuring...` do not occur on usage. 

I've found some issues regarding PIN, but I'll open this in a new issue after creating the PR for this update. Thanks so far @Koenkk , you are as always the Master! Closing this topic.

closed 2021-05-27 by TheStigh