Koenkk/zigbee2mqtt #7569: TuYa TS0043 battery property is always null

mrreview, 2021-05-23:
## What happened
Battery level always show null value. It should be battery level instead?
`May 23 14:53:08 zigbee2mqtt npm[10891]: Zigbee2MQTT:info  2021-05-23 14:53:08: MQTT publish: topic 'zigbee2mqtt/mando_despatx', payload '{"action":"1_single","battery":null,"linkquality":93}'`

## What did you expect to happen
Show properly the battery level

## Debug info
Zigbee2MQTT version: 1.18.3
Adapter hardware: zzh! CC2652R
Adapter firmware version: CC2652R_coordinator_20210120

playersct, 2021-06-18:
I have this same problem too in LoraTap SS600ZB remote

playersct, 2021-06-23:
I got today Losonho TS0043 (zigbee wireless switch ZBWS03) and does not show battery status. Only blinking "%" and battery status "null %"

biggle606, 2021-07-21:
LoraTap SS600ZB remote. Battery status "null %"
Zigbee2MQTT version: 1.20.0

bidulle95, 2021-08-02:
pareil j'ai %

MattWestb, 2021-09-02:
> Я разобрался с данной проблемой. При спаривании устройства постоянно нажимал на кнопку сопряжения. Раз в секунду примерно. И устройство залетело с нормальным статусом батареи.

Google translate:

> I figured out this problem. When pairing the device, I constantly pressed the pairing button. Approximately once a second. And the device flew with a normal battery status.
  
  
Great finding hope user can trying it out and getting there switches working OK :-))

closed 2021-10-11 by github-actions[bot]