Koenkk/zigbee2mqtt #7571: Want a way to ignore publish if `color`/`color_temp` payload on set.

sjorge, 2021-05-23:
## Is your feature request related to a problem? Please describe
When using reporting for currentX/currentY/currentHue/currentSatuartion/colorTemperature the color slider will jump around slightly because bulbs don't always land on the exact set color. (Probably a mix of gamma correction and bulb firmware).

It would be nice to be able to skip the publish for color specifically. I use a mix of device topics (single bulbs/light bars/...) and group topics (sets of bulbs).

## Describe the solution you'd like
A way to skip publishing of the `color` and `color_temp` payloads on set. (Or a way to specify a list of payloads to not publish would be fine and more generic too)

## Describe alternatives you've considered
- https://github.com/Koenkk/zigbee2mqtt/blob/f2e39af12cfaf46166b053ee43b28fb862d46731/lib/extension/publish.js#L227
- high denounce value (5sec), this mostly works but delays the update a lot.
- optimistic: false, but this doesn't have the desired effect as groups still get published regardless due to: https://github.com/Koenkk/zigbee2mqtt/blob/f2e39af12cfaf46166b053ee43b28fb862d46731/lib/extension/publish.js#L227

## Additional context
Add any other context or screenshots about the feature request here.

sjorge, 2021-05-23:
As mentioned in the closed PR optimistic=false is not working as group topic still get a publish.
I'm guessing that makes sense as without reporting the group would never get updated otherwise, the rejected PR avoid this entirely as it filtered out the payload in the toZigbee color and colortemp converters.

sjorge, 2021-05-24:
Would having a optimistic_exclude list be a acceptable solution? Listed properties will always be ignored, even for groups.

That would fix my issue and allow it to be somewhat flexible for other uses, the default of course being `[]`,

This could be done by chaging:
```
const publishOptimistic = isGroup || (!options.hasOwnProperty('optimistic') || options.optimistic);
```

To something like:
```
const publishOptimistic = isGroup || (!options.hasOwnProperty('optimistic') || options.optimistic) || !ioptions.optimistic_exclude.includes(_var_with_property);
```

Not i haven dug deeper, this is assuming the property name is available here.

Koenkk, 2021-05-24:
I think the group and device optimistic behaviour should be the same. So https://github.com/Koenkk/zigbee2mqtt/blob/f2e39af12cfaf46166b053ee43b28fb862d46731/lib/extension/publish.js#L227

should change to: 

`const publishOptimistic = (!options.hasOwnProperty('optimistic') || options.optimistic);`

Would that fix your issue?

sjorge, 2021-05-24:
I think so, let me make the change locally and double check tomorrow.

sjorge, 2021-05-25:
I thought it fixed it but it didn't it still publish the group before any of the attReport's

```
Zigbee2MQTT:debug 2021-05-25 17:26:47: Received MQTT message on 'zigbee2mqtt/office/light/main/set/color_temp' with data '370'
Zigbee2MQTT:debug 2021-05-25 17:26:47: Publishing 'set' 'color_temp' to 'office/light/main'
Zigbee2MQTT:info  2021-05-25 17:26:47: MQTT publish: topic 'zigbee2mqtt/office/light/main/panel1', payload '{"brightness":64,"color":{"hue":32,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"linkquality":105,"power_on_behavior":"off","state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:info  2021-05-25 17:26:47: MQTT publish: topic 'zigbee2mqtt/office/light/main', payload '{"brightness":64,"color":{"hue":32,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"state":"ON"}'
Zigbee2MQTT:info  2021-05-25 17:26:47: MQTT publish: topic 'zigbee2mqtt/office/light/main/panel2', payload '{"brightness":64,"color":{"hue":32,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"linkquality":96,"power_on_behavior":"off",
"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:info  2021-05-25 17:26:47: MQTT publish: topic 'zigbee2mqtt/office/light/main/panel3', payload '{"brightness":64,"color":{"hue":32,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"linkquality":87,"power_on_behavior":"off",
"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:info  2021-05-25 17:26:47: MQTT publish: topic 'zigbee2mqtt/office/light/main/panel4', payload '{"brightness":64,"color":{"hue":32,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"linkquality":99,"power_on_behavior":"off","state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:info  2021-05-25 17:26:47: MQTT publish: topic 'zigbee2mqtt/adaptive_ligthing', payload '{"brightness":254,"color":{"hue":32,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"state":"ON"}'
Zigbee2MQTT:debug 2021-05-25 17:26:47: Received Zigbee message from 'office/light/main/panel4', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":370}' from endpoint 1 with groupID 0                                                         Zigbee2MQTT:debug 2021-05-25 17:26:47: Received Zigbee message from 'office/light/main/panel2', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":370}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-25 17:26:47: Received Zigbee message from 'office/light/main/panel3', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":370}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-25 17:26:47: Received Zigbee message from 'office/light/main/panel1', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":370}' from endpoint 1 with groupID 0
```

sjorge, 2021-05-25:
Closed the PR again, testing with a group with 1 bulb and no debounce consistantly results in 2x publish of the group and device topic. It should be 1x of each after the attReport.

Yeah I'm not sure this is fixed, it still publishes multiple ones which I do not expect

```
Zigbee2MQTT:debug 2021-05-25 17:31:39: Received MQTT message on 'zigbee2mqtt/office/light/desk/set/color_temp' with data '250'
Zigbee2MQTT:debug 2021-05-25 17:31:39: Publishing 'set' 'color_temp' to 'office/light/desk'
Zigbee2MQTT:info  2021-05-25 17:31:40: MQTT publish: topic 'zigbee2mqtt/office/light/desk/bulb', payload '{"brightness":254,"color":{"hue":40,"saturation":49,"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_options":{"execute_if_off":true},"color_temp":250,"color_temp_startup":65535,"linkquality":87,"power_on_behavior":"off","state":"ON"}'
Zigbee2MQTT:info  2021-05-25 17:31:40: MQTT publish: topic 'zigbee2mqtt/office/light/desk', payload '{"brightness":254,"color":{"hue":40,"saturation":49,"x":0.3804,"y":0.3767},"color_mode":"color_temp","color_temp":250,"state":"ON"}'
Zigbee2MQTT:debug 2021-05-25 17:31:40: Received Zigbee message from 'office/light/desk/bulb', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":250,"currentHue":33,"currentSaturation":118,"currentX":25021,"currentY":24883}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-05-25 17:31:40: MQTT publish: topic 'zigbee2mqtt/office/light/desk', payload '{"brightness":254,"color":{"hue":47,"saturation":46,"x":0.3818,"y":0.3797},"color_mode":"color_temp","color_temp":250,"state":"ON"}'
Zigbee2MQTT:info  2021-05-25 17:31:40: MQTT publish: topic 'zigbee2mqtt/office/light/desk/bulb', payload '{"brightness":254,"color":{"hue":47,"saturation":46,"x":0.3818,"y":0.3797},"color_mode":"color_temp","color_options":{"execute_if_off":true},"color_temp":250,"color_temp_startup":65535,"linkquality":99,"power_on_behavior":"off","state":"ON"}'
```

So the change does not work sadly :(

sjorge, 2021-05-25:
```
device_options:
  retain: true
  legacy: false
  #debounce: 0.5
  optimistic: false
```

Are my global device options.

Koenkk, 2021-05-25:
With https://github.com/Koenkk/zigbee2mqtt/issues/7571#issuecomment-847248635 included, can you try adding below https://github.com/Koenkk/zigbee2mqtt/blob/fc22702bc1e0fc760eff2f0102be05b07e0ba629/lib/extension/groups.js#L79 `console.log(JSON.stringify(data))`, this should reveal what causes these updates.

sjorge, 2021-05-25:
I grepped on 0x14b457fffe2bd760 (the bulb) and XXX (added that to the log)
```
XXX: {"ID":"0x14b457fffe2bd760","from":{"state":"ON","brightness":254,"color":{"hue":47,"saturation":46,"x":0.3818,"y":0.3797},"color_temp":250,"color_options":{"execute_if_off":true},"color_temp_startup":65535,"color_mode":"color_temp","power_on_behavior":"off"},"to":{"state":"ON","brightness":254,"color":{"hue":27,"saturation":92,"x":0.5056,"y":0.4152},"color_temp":454,"color_options":{"execute_if_off":true},"color_temp_startup":65535,"color_mode":"color_temp","power_on_behavior":"off"},"reason":"group_optimistic","update":{"color_mode":"color_temp","color":{"x":0.5056,"y":0.4152,"hue":27,"saturation":92},"color_temp":454}}
XXX: {"ID":"0x14b457fffe2bd760","from":{"state":"ON","brightness":254,"color":{"hue":27,"saturation":92,"x":0.5056,"y":0.4152},"color_temp":454,"color_options":{"execute_if_off":true},"color_temp_startup":65535,"color_mode":"color_temp","power_on_behavior":"off"},"to":{"state":"ON","brightness":254,"color":{"hue":28,"saturation":91,"x":0.5081,"y":0.4136},"color_temp":454,"color_options":{"execute_if_off":true},"color_temp_startup":65535,"color_mode":"color_temp","power_on_behavior":"off"},"reason":null,"update":{"color_temp":454,"color":{"x":0.5081,"y":0.4136,"saturation":91,"hue":28}}}
```

Koenkk, 2021-05-26:
Above both https://github.com/Koenkk/zigbee2mqtt/blob/669489fe97c6d968cbeefe1d659e44119374bd90/lib/extension/publish.js#L240 and https://github.com/Koenkk/zigbee2mqtt/blob/669489fe97c6d968cbeefe1d659e44119374bd90/lib/extension/publish.js#L245 can you put: `console.log('PUBLISH LOCATION 1')` and `console.log('PUBLISH LOCATION 2')` and provide the log again.

sjorge, 2021-05-26:
```
Zigbee2MQTT:debug 2021-05-26 16:46:59: Received MQTT message on 'zigbee2mqtt/office/light/desk/set/color_temp' with data '370'
Zigbee2MQTT:debug 2021-05-26 16:46:59: Publishing 'set' 'color_temp' to 'office/light/desk'
PUBLISH LOCATION 1
{"ID":10112,"from":{"state":"ON","brightness":254,"color":{"hue":47,"saturation":46,"x":0.3818,"y":0.3797},"color_temp":250,"color_mode":"color_temp"},"to":{"state":"ON","brightness":254,"color":{"hue":32,"saturation":82,"x":0.4599,"y":0.4106},"color_temp":370,"color_mode":"color_temp"},"reason":null,"update":{"color_mode":"color_temp","color":{"x":0.4599,"y":0.4106,"hue":32,"saturation":82},"color_temp":370}}
{"ID":"0x14b457fffe2bd760","from":{"state":"ON","brightness":254,"color":{"hue":47,"saturation":46,"x":0.3818,"y":0.3797},"color_temp":250,"color_options":{"execute_if_off":true},"color_temp_startup":65535,"color_mode":"color_temp","power_on_behavior":"off"},"to":{"state":"ON","brightness":254,"color":{"hue":32,"saturation":82,"x":0.4599,"y":0.4106},"color_temp":370,"color_options":{"execute_if_off":true},"color_temp_startup":65535,"color_mode":"color_temp","power_on_behavior":"off"},"reason":"group_optimistic","update":{"color_mode":"color_temp","color":{"x":0.4599,"y":0.4106,"hue":32,"saturation":82},"color_temp":370}}
Zigbee2MQTT:info  2021-05-26 16:47:00: MQTT publish: topic 'zigbee2mqtt/office/light/desk/bulb', payload '{"brightness":254,"color":{"hue":32,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_options":{"execute_if_off":true},"color_temp":370,"color_temp_startup":65535,"linkquality":54,"power_on_behavior":"off","state":"ON"}'
Zigbee2MQTT:info  2021-05-26 16:47:00: MQTT publish: topic 'zigbee2mqtt/office/light/desk', payload '{"brightness":254,"color":{"hue":32,"saturation":82,"x":0.4599,"y":0.4106},"color_mode":"color_temp","color_temp":370,"state":"ON"}'
Zigbee2MQTT:debug 2021-05-26 16:47:00: Received Zigbee message from 'office/light/desk/bulb', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":370,"currentHue":24,"currentSaturation":202,"currentX":29991,"currentY":26872}' from endpoint 1 with groupID 0
{"ID":"0x14b457fffe2bd760","from":{"state":"ON","brightness":254,"color":{"hue":32,"saturation":82,"x":0.4599,"y":0.4106},"color_temp":370,"color_options":{"execute_if_off":true},"color_temp_startup":65535,"color_mode":"color_temp","power_on_behavior":"off"},"to":{"state":"ON","brightness":254,"color":{"hue":34,"saturation":80,"x":0.4576,"y":0.41},"color_temp":370,"color_options":{"execute_if_off":true},"color_temp_startup":65535,"color_mode":"color_temp","power_on_behavior":"off"},"reason":null,"update":{"color_temp":370,"color":{"x":0.4576,"y":0.41,"saturation":80,"hue":34}}}
{"ID":10112,"from":{"state":"ON","brightness":254,"color":{"hue":32,"saturation":82,"x":0.4599,"y":0.4106},"color_temp":370,"color_mode":"color_temp"},"to":{"state":"ON","brightness":254,"color":{"hue":34,"saturation":80,"x":0.4576,"y":0.41},"color_temp":370,"color_mode":"color_temp"},"reason":"group_optimistic","update":{"color_temp":370,"color":{"x":0.4576,"y":0.41,"saturation":80,"hue":34}}}
Zigbee2MQTT:info  2021-05-26 16:47:00: MQTT publish: topic 'zigbee2mqtt/office/light/desk', payload '{"brightness":254,"color":{"hue":34,"saturation":80,"x":0.4576,"y":0.41},"color_mode":"color_temp","color_temp":370,"state":"ON"}'
Zigbee2MQTT:info  2021-05-26 16:47:00: MQTT publish: topic 'zigbee2mqtt/office/light/desk/bulb', payload '{"brightness":254,"color":{"hue":34,"saturation":80,"x":0.4576,"y":0.41},"color_mode":"color_temp","color_options":{"execute_if_off":true},"color_temp":370,"color_temp_startup":65535,"linkquality":54,"power_on_behavior":"off","state":"ON"}'
```

sjorge, 2021-05-26:
I also added:
```
console.log(`DEBUG: publishOptimistic=${publishOptimistic}, optimistic=${options.hasOwnProperty('optimistic')}, optimistic_val=${options.optimistic}`);
```
```
[root@amethyst /opt/zigbee2mqtt]# grep optimistic data/configuration.yaml
  optimistic: false
```

```
Zigbee2MQTT:debug 2021-05-26 16:49:28: Received MQTT message on 'zigbee2mqtt/office/light/desk/set/color_temp' with data '454'
Zigbee2MQTT:debug 2021-05-26 16:49:28: Publishing 'set' 'color_temp' to 'office/light/desk'
DEBUG: publishOptimistic=true, optimistic=true, optimistic_val=true
PUBLISH LOCATION 1
{"ID":10112,"from":{"state":"ON","brightness":254,"color":{"hue":34,"saturation":80,"x":0.4576,"y":0.41},"color_temp":370,"color_mode":"color_temp"},"to":{"state":"ON","brightness":254,"color":{"hue":27,"saturation":92,"x":0.5056,"y":0.4152},"color_temp":454,"color_mode":"color_temp"},"reason":null,"update":{"color_mode":"color_temp","color":{"x":0.5056,"y":0.4152,"hue":27,"saturation":92},"color_temp":454}}
{"ID":"0x14b457fffe2bd760","from":{"state":"ON","brightness":254,"color":{"hue":34,"saturation":80,"x":0.4576,"y":0.41},"color_temp":370,"color_options":{"execute_if_off":true},"color_temp_startup":65535,"color_mode":"color_temp","power_on_behavior":"off"},"to":{"state":"ON","brightness":254,"color":{"hue":27,"saturation":92,"x":0.5056,"y":0.4152},"color_temp":454,"color_options":{"execute_if_off":true},"color_temp_startup":65535,"color_mode":"color_temp","power_on_behavior":"off"},"reason":"group_optimistic","update":{"color_mode":"color_temp","color":{"x":0.5056,"y":0.4152,"hue":27,"saturation":92},"color_temp":454}}
Zigbee2MQTT:info  2021-05-26 16:49:29: MQTT publish: topic 'zigbee2mqtt/office/light/desk/bulb', payload '{"brightness":254,"color":{"hue":27,"saturation":92,"x":0.5056,"y":0.4152},"color_mode":"color_temp","color_options":{"execute_if_off":true},"color_temp":454,"color_temp_startup":65535,"power_on_behavior":"off","state":"ON"}'
Zigbee2MQTT:info  2021-05-26 16:49:29: MQTT publish: topic 'zigbee2mqtt/office/light/desk', payload '{"brightness":254,"color":{"hue":27,"saturation":92,"x":0.5056,"y":0.4152},"color_mode":"color_temp","color_temp":454,"state":"ON"}'
Zigbee2MQTT:debug 2021-05-26 16:49:29: Received Zigbee message from 'office/light/desk/bulb', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":454,"currentHue":20,"currentSaturation":231,"currentX":33296,"currentY":27106}' from endpoint 1 with groupID 0
{"ID":"0x14b457fffe2bd760","from":{"state":"ON","brightness":254,"color":{"hue":27,"saturation":92,"x":0.5056,"y":0.4152},"color_temp":454,"color_options":{"execute_if_off":true},"color_temp_startup":65535,"color_mode":"color_temp","power_on_behavior":"off"},"to":{"state":"ON","brightness":254,"color":{"hue":28,"saturation":91,"x":0.5081,"y":0.4136},"color_temp":454,"color_options":{"execute_if_off":true},"color_temp_startup":65535,"color_mode":"color_temp","power_on_behavior":"off"},"reason":null,"update":{"color_temp":454,"color":{"x":0.5081,"y":0.4136,"saturation":91,"hue":28}}}
{"ID":10112,"from":{"state":"ON","brightness":254,"color":{"hue":27,"saturation":92,"x":0.5056,"y":0.4152},"color_temp":454,"color_mode":"color_temp"},"to":{"state":"ON","brightness":254,"color":{"hue":28,"saturation":91,"x":0.5081,"y":0.4136},"color_temp":454,"color_mode":"color_temp"},"reason":"group_optimistic","update":{"color_temp":454,"color":{"x":0.5081,"y":0.4136,"saturation":91,"hue":28}}}
Zigbee2MQTT:info  2021-05-26 16:49:29: MQTT publish: topic 'zigbee2mqtt/office/light/desk', payload '{"brightness":254,"color":{"hue":28,"saturation":91,"x":0.5081,"y":0.4136},"color_mode":"color_temp","color_temp":454,"state":"ON"}'
Zigbee2MQTT:info  2021-05-26 16:49:29: MQTT publish: topic 'zigbee2mqtt/office/light/desk/bulb', payload '{"brightness":254,"color":{"hue":28,"saturation":91,"x":0.5081,"y":0.4136},"color_mode":"color_temp","color_options":{"execute_if_off":true},"color_temp":454,"color_temp_startup":65535,"linkquality":54,"power_on_behavior":"off","state":"ON"}'
```

sjorge, 2021-05-26:
I can't figure out where it is getting this 'true' value from I grepped the entire /opt/zigbee2mqtt dir and aside from the one hit in data/configuration.yaml all I get is hits in the code itself.

Is what I get when I just log options
```json
 {"retain":true,"legacy":false,"optimistic":true,"devices":["0x14b457fffe2bd760/1"],"friendly_name":"office/light/desk","ID":10112,"friendlyName":"office/light/desk","type":"group"}
```

sjorge, 2021-05-26:
I did find this:
```
lib/util/settings.js:        return {optimistic: true, devices: [], ...byID, ID: Number(IDorName), friendlyName: byID.friendly_name};
lib/util/settings.js:            return {optimistic: true, devices: [], ...group, ID: Number(ID), friendlyName: group.friendly_name};
lib/util/settings.js:        return {optimistic: true, devices: [], ...group, ID: Number(ID), friendlyName: group.friendly_name};
```

Which looks a bit suspicious.

sjorge, 2021-05-26:
If I remove the `optimistic: true` in lib/util/settings.js it works as it should I think.

```
{"retain":true,"legacy":false,"optimistic":false,"devices":["0x14b457fffe2bd760/1"],"friendly_name":"office/light/desk","ID":10112,"friendlyName":"office/light/desk","type":"group"}
```

Is the options payload now and it does not hit the publish location 1 anymore, and only publishes after the attReport. I'm not sure they should be there, I think they can just be removed? As the check should still default to true if unset.

Koenkk, 2021-05-27:
I don't understand what goes wrong, the `...byID` or `...group` should be an object containing `{optimistic: false}` which override the `optimistic: true`. I've also added a test case for this which passes: https://github.com/Koenkk/zigbee2mqtt/pull/7615/files#diff-e5529dd6d95cc6412d41048a450c6d2b3647beaba2e5489a02094ab5b4441d4fR425

Could you debug what is returned in settings.js?

sjorge, 2021-05-27:
I will redo some of the testing, but AFAIK I figured out. the hardcoded true's from lib/util/settings.js get merged in after devices_options so it takes precedence.

sjorge, 2021-05-27:
So these get called:

```
XXX: getGroup - {"optimistic":true,"devices":["0x14b457fffe2bd760/1"],"friendly_name":"office/light/desk","ID":10112,"friendlyName":"office/light/desk"}
XXX: publish - {"retain":true,"legacy":false,"optimistic":true,"devices":["0x14b457fffe2bd760/1"],"friendly_name":"office/light/desk","ID":10112,"friendlyName":"office/light/desk","type":"group"}
```

That is the logging of what is returned in settings.js and the options object in publish

sjorge, 2021-05-27:
@Koenkk I traced it upstream some what, I was correct in the merge order. It goes wrong here:
https://github.com/Koenkk/zigbee2mqtt/blob/669489fe97c6d968cbeefe1d659e44119374bd90/lib/zigbee.js#L261

`settings: {...deviceOptions, ...entity}` entity here already has the output of settings.getGRoup() with the hardcoded optimistic=true. So it is currently impossible to disable optimistic for any group I think.

Koenkk, 2021-05-28:
Thanks, see we it goes wrong now  indeed, merged it :smile:

closed 2021-05-28 by Koenkk

sjorge, 2021-05-28:
Excellent, seems to do the trick with optimistic=false in device_options and setting it to true on the non light bulb devices.

sjorge, 2021-05-29:
@Koenkk how do you feel about adding a filter like option too?
I've been messing about with having `optimistic: false` on only my bulbs, as that gives odd results with it off for everything on stuff like a TRV or switches  that don't have reporting for genOnOff.

But now i hit another one that is not easy to fix:
[image]

For example here there is repoting configure for xy, hs, ct, state and brightness... but chaging color_temp_startup now does not update the value in the payload (as expected) and I need to manually read it, as the bulb does not seem to want to do reporting for this attribute.

As originally state I only really care about `color` and `color_temp` not being set optimistically.

We already have a [filtered_attributes](https://www.zigbee2mqtt.io/information/configuration.html#device-specific-configuration) so I was thinking of filterd_optimistic that would do the same, filter out certain attributes when optimistic is true.

My idea would be to just insert a check agains it here https://github.com/Koenkk/zigbee2mqtt/blob/3d00338bc89e33eef345c0f5fa3c84ebf5cf6ffe/lib/extension/publish.js#L239 and delete the attribute from the payload.

Should be able to take a jab at this later this weekend, probably tomorrow.

Koenkk, 2021-05-29:
@sjorge I'm fine with that, but what would it mean if a user sets only `filterd_optimistic`, would this automatically make the other attributes not optimistic? (it conflicts with the `optimistic` option)

sjorge, 2021-05-29:
No, it's in reverse filtered_optimistic will filter out attributes from the optimistic behavior (like fitlered_attributes does)

Disable optimistic for everything
```
optmistic: false
```

Enable optimistic for everything (Default)
```
optimistic: true
```

Enable optimistic for everything but filter out color and color_temp
```
optimistic: true
filtered_optimistic:
  - color
  - color_temp
```

Well you can drop the `optmistic: true` as that is the default.

Koenkk, 2021-05-29:
I see now, looks good

PR #7615 (2021-05-28): Honour optimisitic false for groups
Closes: #7571