Koenkk/zigbee2mqtt #7573: RaspBee II on Hoobs PI

Stetocasa, 2021-05-24:
Hi, I’m losing hours to understand how work Zigbee2Mqtt via hoobs trough a RaspBee II

Mqtt server and all packages for Zigbee2Mqtt was installed and the configuration also

The issue is to identity the right location of the RaspBee

Trough the common input ls -l /dev/ttyACM0 a message shows that no permission to access 
[image]

The service running 
[image]
 But npm instal fail

[image]

Somebody could help me to solve and check if the raspbee is right connected? I already tried all FAQ

robertalexa, 2021-05-24:
Hi,

The error message about permission is not actually about permission but rather the device is not at that address

[image]

I think your adapter might actually be at /dev/ttyUSB0 and not /dev/ttyACM0.

Try that in your config instead.

If that still fails, can you run `lsusb -t` and paste the output?

Stetocasa, 2021-05-24:
> Hi,
> 
> The error message about permission is not actually about permission but rather the device is not at that address
> 
> [image]
> 
> I think your adapter might actually be at /dev/ttyS0 and not /dev/ttyACM0.
> 
> Try that in your config instead.
> 
> If that still fails, can you run `lsusb -t` and paste the output?

hoobs@hoobs:~ $ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M

robertalexa, 2021-05-24:
Sorry dude, I think I have edited my post while you were at it

Can you try in your config /dev/ttyS0 ?

PS: My bad aboud lsusb, that won't give you anything good in this case cause your device is not usb.

Stetocasa, 2021-05-24:
Seems that the device in not found
hoobs@hoobs:~ $ sudo GCFFlasher_internal -l
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee

Stetocasa, 2021-05-24:
No detected 

hoobs@hoobs:~ $ ls -l /dev/ttyS0
crw-rw---- 1 root dialout 4, 64 Feb 14  2019 /dev/ttyS0

robertalexa, 2021-05-24:
Could you have a look at the last post in this thread
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3898

I have a feeling that this is not a problem just for you. You might need to do some changes on your raspberry pi.

PS: i do not own your adapter, just trying to help

Stetocasa, 2021-05-24:
I did that:

sudo nano config.txt

enable_uart=1                                                                 
dtoverlay=miniuart-bt                                                     
force_turbo=1  

Save
Reboot

Nothing changed

robertalexa, 2021-05-24:
Silly question, but have you followed the instruction for this device?
https://phoscon.de/en/raspbee2/install#raspbian

In your case, you probable need everything UP TO deCONZ Installation

Stetocasa, 2021-05-24:
Deconz is already installed
[image]

robertalexa, 2021-05-24:
I meat the drivers for the board - basically RTC Installtion.

If that is all done, I am afraid I do not know what to suggest. Maybe someone else with this board can have some input.

Though I suggest you dig online with regards to why the board is not recognised rather than hoobs or z2m specific. They will all fail if the device is not found basically.

zhapwey, 2021-05-24:
Hello:
You may not have done the step:
[image]
Attention that the first question is no and the second yes.

Stetocasa, 2021-05-24:
> Hello:
> 
> You may not have done the step:
> 
> [image]
> 
> Attention that the first question is no and the second yes.

Already done! (And after your message checked again)
No way to detect the device

closed 2021-07-01 by github-actions[bot]