Koenkk/zigbee2mqtt #7581: Swapping lfl coordinator results in database.db being backed up then wiped

Morphy99, 2021-05-24:
## What happened
Swapped a CC2530 for another on the same firmware. 
database.db is backed up to database.db.backup and a fresh database.db created. Not sure why this happens?

According to the [FAQ](https://www.zigbee2mqtt.io/information/FAQ.html#what-does-and-does-not-require-repairing-of-all-devices) this should be possible without disruption but I also had problems with two types of devices that dropped off the network:
[Konke Motion Sensor](https://www.zigbee2mqtt.io/devices/2AJZ4KPBS.html#konke-2ajz4kpbs) 
[Osram Motion Sensor](https://www.zigbee2mqtt.io/devices/AC01353010G.html#osram-ac01353010g)
[Eurotronic Spirit TRV](https://www.zigbee2mqtt.io/devices/SPZB0001.html#eurotronic-spzb0001)

Once I enabled joining the Konke rejoined but the Osram ones failed to join the new coordinator. I went back to the previous coordinator in the end to save repairing and everything was back working as normal.

I suspect this has something to do with a routing device as I have several other Spirit TRV's which were joined OK.

## What did you expect to happen

Swap adapters and Z2M restart with devices joined and working.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.18.2 commit: abd8a09
Adapter hardware: CC2530
Adapter firmware version: 20201128 (Source Routing)

As this isn't a major bug and could likely be due to some device not adhering to standards I think this should be mentioned in the FAQ that some devices might not rejoin.  Also, to save users from panicking when they swap coordinators and the DB gets wiped a note about the backup would be good. Or just removing that behaviour if acceptable?

Koenkk, 2021-05-24:
Please test again with the latest dev branch, the restore/backup procedure has been heavily refactored (https://github.com/Koenkk/zigbee-herdsman/pull/303).

Dev branch: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

Morphy99, 2021-06-24:
Still to test this, is it in the latest stable or still dev?

Koenkk, 2021-06-24:
@Morphy99 it is in 1.19.1

closed 2021-07-02 by github-actions[bot]

Morphy99, 2021-07-07:
> @Morphy99 it is in 1.19.1

I'm on 1.19.1 and just swapped coordinator, same behaviour, database is backed up to database.db.backup and a fresh database.db is created.

Koenkk, 2021-07-07:
Could you provide the herdsman debug log when starting for the first time after swapping the coordinator?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.