Koenkk/zigbee2mqtt #7583: The curtain switch doesn't send the correct position/status

jvsystelab, 2021-05-24:
Hi and thanks in advance.

**- Scenario**

I have a curtain switch paired, LoraTap SC400W-US

https://www.loratap.com/products/tuya-smart-life-zigbee-30-curtain-blind-switch-for-roller-shutter-electric-motor-google-home-alexa-echo-voice-control-diy-1

But the zigbe2mqtt detects a TuYa TS130F

https://www.zigbee2mqtt.io/devices/TS130F.html

For me, it isn't a problem, all works fine less this:

**- What happened**

I can open and close de curtain (from HA and device) and calibrate the switch but, the device never sends the correct position.
HA and zigbe2mqtt frontend shows that the curtain is closed when it is actually open and viceversa.

**- What did I expect to happen**

I expect see the correct position of the curtain in the apps frontend.

**- How to reproduce it**

Pair and configure the device.
Open the curtain

**- Device pairing log**

https://pastebin.com/SysWRCBs

**- Log after down (motor_reversal: on)** 

https://pastebin.com/LTtqzE4v

**- Log after down (motor_reversal: off)** 

https://pastebin.com/RJcAar5q

**- Debug info**

Zigbee2MQTT version: 1.18.3
Adapter hardware: zig-a-zig-ah
Adapter firmware version: Latest

sjorge, 2021-05-25:
You might need to set the device option to invert the value:
https://www.zigbee2mqtt.io/devices/TS130F.html#device-type-specific-configuration

jvsystelab, 2021-05-25:
Hi,

How I change this value "invert_cover"? From zigbee2mqtt frontend?

I can't this option:

{
    "backlight_mode": "LOW",
    "calibration": "OFF",
    "linkquality": 54,
    "motor_reversal": "ON",
    "moving": "STOP",
    "position": 100
}

sjorge, 2021-05-25:
You add it under the device in configuration.yaml

https://www.zigbee2mqtt.io/information/configuration.html#example

jvsystelab, 2021-05-25:
Working!!! You're my new hero!!

Thank you very much!!

closed 2021-05-31 by jvsystelab