Koenkk/zigbee2mqtt #7589: ikea e1766 stopped working on my blinds after paring with zigbee2mqtt

ofirdagan, 2021-05-24:
## What happened
Hey, I managed to run zigbee2mqtt with my Texas Instruments CC2531. I also managed to pair my ikea blind remote e1766 with it. The problem is.. since then the remote stopped working.  I can see the messages in the mqtt but it doesn't control the blind anymore. and I can't make it bind to blind anymore. 

Please help :)
## What did you expect to happen
for the control to keep working

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: flashed your hardware

sjorge, 2021-05-25:
You need to pair both the remote and the blind to zigbee2mqtt, then you can bind the remote to the blind.

More info here: https://www.zigbee2mqtt.io/information/binding.html
Or just use the frontend to do it, thats just a lot easier than doing it via mqtt manually.

ofirdagan, 2021-05-25:
I'll try that thanks @sjorge. However the remote control had stopped working completely now. Can I reset it somehow so it'll work again? (i.e control the blind again regardless to the zigbee dongle)

sjorge, 2021-05-25:
You can remove it from the dongle and do the setup as usual, or pair both and then bind it.
Once bound, it also works when the dongle is offline.

ofirdagan, 2021-05-25:
I'm putting it for future reference..

I found a way to make the remote working again. I had to pair it first to the repeater and then to the blinds. I saw it here:
https://www.youtube.com/watch?v=72oF2KdQEqs

As for the zigbee2mqtt. I'll try using the front end soon and will update :)

MattWestb, 2021-05-25:
@ofirdagan Its recommended having the repeater or one good router near the blind so they always having contact with the network and the remote can sending the commands to them also then the coordinator is offline then you have doing the binding of the remote to the group the blind is added to.

ofirdagan, 2021-05-27:
finally made it all work :).. I'm writing my process here for future reference.
In order to make my ikea fyrtur blind work w/ zigbee2mqtt I had to:

1. pair the repeater to zigbee2mqtt [link](https://www.zigbee2mqtt.io/devices/E1746.html#pairing)
2. pair the blind to zigbee2mqtt [link](https://www.zigbee2mqtt.io/devices/E1757.html#pairing) while holding the repeater very close to the blind. It doesn't always works on the first try, be patient. 
3. pair my remote to zigbee2mqtt [link](https://www.zigbee2mqtt.io/devices/E1766.html#pairing)
4. unbind the remote from the default group
5. create a new group in the zigbee2mqtt UI
6. add the ikea blind device into said group
7. bind the remote to said group

Thank you for your help @MattWestb , @sjorge

closed 2021-05-27 by ofirdagan

MattWestb, 2021-05-27:
What i can see is you doing all things in the right way and if all is going well the remote have on "elusive" channel for sending the commands to the blinds.

Great done !!