Koenkk/zigbee2mqtt #7591: Latest-dev docker container does not start anymore

Calimerorulez, 2021-05-25:
Hi,

I just pulled the latest-dev docker image and the container failes to start:

```
Using '/app/data' as data directory
#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0xbecc125c
```

fludstank, 2021-05-27:
Did you get this solved? I'm getting the same issue

Calimerorulez, 2021-06-02:
> Did you get this solved? I'm getting the same issue

No I didn't, I migrated to my Intel NUC out of frustration... But I will reopen the issue :)

pgollor, 2021-06-02:
I got the same error by updating from docker image 1.18.3 to 1.19.0 on a Raspberry Pi4. But with another hex value:

```
#
# Fatal error in , line 0
# unreachable code
#
#
#
#FailureMessage Object: 0xbe86d1bc
```

Koenkk, 2021-06-02:
Closing this in favour of #7662 , please let me know if the latest dev works (https://github.com/Koenkk/zigbee2mqtt/issues/7662#issuecomment-853251828)

closed 2021-06-02 by Koenkk