Koenkk/zigbee2mqtt #7604: 2021 New Gledopto Dimmer is Unsupported

ildmgithub, 2021-05-25:
Hello,
I ordered the new gledopto dimmer but it is not supported by zigbee2mqtt. I tried to follow the instructions for adding a new device, but I am stuck.  The instruction to add new devices is not clear for me.

[image]

## Information about the device + link
https://www.gledopto.eu/zigbee-flush-mounted-triac-dimmer-control-unit-controller-230v-eng

ildmgithub, 2021-07-17:
Still actually problem

Koenkk, 2021-07-17:
Added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-07-17 by Koenkk

bernd-uhl, 2021-08-05:
Still actual problem

Koenkk, 2021-08-13:
Should be fixed now thanks to @Julusian 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

ildmgithub, 2021-09-05:
Hello 
When I run zigbee2mqtt edge I see that it is still unsupported.
[image]

xdmore70, 2024-10-08:
I have the same problem, Gledopto device works fine but the frontend report unsopported
[image]

I use last release frontend and last coordinator firmware
[image]

I find this old post, I hope some can help me