Koenkk/zigbee2mqtt #7605: Gledopto Firmware Update

narc792, 2021-05-25:
## What happened

Trying to update a Gledopto GL-B-008P but Zigbee2mqtt reports that the “device did not request any image blocks.” The update seems to be in the database so I’m but sure why it won’t go through. 

## What did you expect to happen

## How to reproduce it (minimal and precise)

Clicking Update OTA next to the light in the home assistant integration UI results in this issue every time. 

## Debug info
Zigbee2MQTT version:
Adapter hardware: zzh CC2652R
Adapter firmware version: CC2652R_coordinator_20210120

narc792, 2021-06-26:
This issue is still occurring on the latest Zigbee2MQTT release.

narc792, 2021-07-29:
No one care to help with this? I'm guessing the bulb is slightly old enough that it wont support an update.

closed 2021-09-06 by github-actions[bot]

jmgutierrezm, 2022-08-29:
Hi All, the issue still remains:
Zigbee2mqtt version: 1.27.0-1
Device: Gledopto GL-B-008P
Firmware version installed: V_1_4

Log:
debug 2022-08-29 12:58:46Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/update' with data '{"id":"Bombilla Burbujita","transaction":"axraa-1"}'
info 2022-08-29 12:58:46Updating 'Bombilla Burbujita' to latest firmware
info 2022-08-29 12:58:46MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Updating 'Bombilla Burbujita' to latest firmware","meta":{"device":"Bombilla Burbujita","status":"update_in_progress"},"type":"ota_update"}'
debug 2022-08-29 12:58:46Received Zigbee message from 'Bombilla Burbujita', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20201228","swBuildId":"V_1_4"}' from endpoint 11 with groupID 0
debug 2022-08-29 12:58:49Received Zigbee message from 'Bombilla Burbujita', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":6,"imageType":0,"manufacturerCode":4687}' from endpoint 11 with groupID 0
debug 2022-08-29 12:58:49Received Zigbee message from 'Bombilla Burbujita', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":6,"imageType":0,"manufacturerCode":4687}' from endpoint 11 with groupID 0
debug 2022-08-29 13:01:19Update of 'Bombilla Burbujita' failed (Error: Timeout: device did not request any image blocks)
info 2022-08-29 13:01:19MQTT publish: topic 'zigbee2mqtt/Bombilla Burbujita', payload '{"brightness":203,"color":{"x":0.37,"y":0.37},"color_mode":"color_temp","color_temp":235,"linkquality":141,"state":"OFF","update":{"state":"available"},"update_available":true}'
info 2022-08-29 13:01:19MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Update of 'Bombilla Burbujita' failed (Timeout: device did not request any image blocks)","meta":{"device":"Bombilla Burbujita","status":"update_failed"},"type":"ota_update"}'
info 2022-08-29 13:01:19MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Bombilla Burbujita"},"error":"Update of 'Bombilla Burbujita' failed (Timeout: device did not request any image blocks)","status":"error","transaction":"axraa-1"}'
error 2022-08-29 13:01:19Update of 'Bombilla Burbujita' failed (Timeout: device did not request any image blocks)