Koenkk/zigbee2mqtt #7606: Unable to start zigbee2mqtt after Synology DSM update to 6.2.4

henrikwils, 2021-05-26:
## What happened
I upgraded Synology DSM from 6.2.1 to 6.2.4 and after it restarted, zigbee2mqtt would not start up. I tried to upgrade the coordinator and follow all the steps in the troubleshooting guide, but nothing helps. It does look like it cannot properly communicate with the port, but the port has root as owner, like all the files in the zigbee2mqtt folder, so I'm pretty sure it's correct.

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.18.3
Adapter hardware: CC26X2R1
Adapter firmware version: 20210120

debug 2021-05-26 08:13:06: Loaded state from file /app/data/state.json
info  2021-05-26 08:13:06: Logging to console and directory: '/app/data/log/2021-05-26.08-13-04' filename: log.txt
debug 2021-05-26 08:13:06: Removing old log directory '/app/data/log/2021-05-26.08-04-41'
info  2021-05-26 08:13:06: Starting Zigbee2MQTT version 1.18.3 (commit #f2e39af)
info  2021-05-26 08:13:06: Starting zigbee-herdsman (0.13.92)
debug 2021-05-26 08:13:06: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[15],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6755},"serialPort":{"path":"/dev/ttyACM0"}}'
error 2021-05-26 08:13:17: Failed to call 'OnEvent' 'stop' (TypeError: Cannot read property 'getEntries' of null
    at Function.loadFromDatabaseIfNecessary (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:221:55)
    at Function.all (/app/node_modules/zigbee-herdsman/dist/controller/model/device.js:241:16)
    at Controller.getDevices (/app/node_modules/zigbee-herdsman/dist/controller/controller.js:289:31)
    at Zigbee.getClients (/app/lib/zigbee.js:154:30)
    at OnEvent.stop (/app/lib/extension/onEvent.js:23:42)
    at Controller.callExtensionMethod (/app/lib/controller.js:384:44)
    at async Controller.stop (/app/lib/controller.js:188:9)
    at async stop (/app/index.js:47:5)
    at async process.handleQuit (/app/index.js:53:9))
debug 2021-05-26 08:13:17: Saving state to file /app/data/state.json
error 2021-05-26 08:13:17: Not connected to MQTT server!
error 2021-05-26 08:13:17: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
info  2021-05-26 08:13:17: Disconnecting from MQTT server

rindlerblabla, 2021-05-27:
Am running 6.2.4 on Synology as well and it's working just fine. Are you sure the container is executed with high privilege?

henrikwils, 2021-05-28:
I have no idea what exactly fixed it, but I reflashed my CC26X2R1 with newest zigbee2mqtt coordinator firmware, and messed around with various things (swapping USB, recreating the container etc). Ended up restoring the entire system configuration, which keeps the DSM version. I'm pretty sure this did the trick, but flashing the CC26X2R1 now required me to change the panid, and re-pair every single device.... SIGH! (I have more than 100!)

closed 2021-07-06 by github-actions[bot]