Koenkk/zigbee2mqtt #7609: no more devices can be paired to cc2652r zzh coordinator

kitopopo, 2021-05-26:
## dear @Koenkk ,

I have 39 devices added to my zigbee network. 8 zigbee devices are paired directly to coordinator cc2652r and the other ones are paired in 3 zigbee repeaters.
When i try to add a new device is impossible pair this new device to cc2562r, this new device will be working very near to coordinator, but always is paired to another zigbee repeater. Looks like the coordinator doesn't accept new devices. The only way that the new device good be paired with the coordinator is unplugging my 3 repeaters. Do yo know this problem? Can you help me?

kitopopo, 2021-06-27:
Dear @Koenkk , any solution with this problem? thanks in advanced

Koenkk, 2021-06-28:
Investigating this in #7762

closed 2021-06-28 by Koenkk