Koenkk/zigbee2mqtt #7611: zigbee2mqtt stopping/hanging

finzec, 2021-05-26:


finzec, 2021-05-27:
I switched logging on to debug hoping to see more...

finzec, 2021-05-27:
This is the log in debug... nothing to see from my end:
debug 2021-05-27 16:15:53: Received Zigbee message from '0x00158d00032b0b02', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"rmsCurrent":403}' from endpoint 1 with groupID 0
info  2021-05-27 16:15:53: MQTT publish: topic 'zigbee2mqtt/0x00158d00032b0b02', payload '{"current":0.4,"linkquality":0,"power":89,"state":"ON","voltage":234}'
debug 2021-05-27 16:15:54: Received Zigbee message from '0x00158d00032b0b02', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":89,"rmsCurrent":403}' from endpoint 1 with groupID 0
info  2021-05-27 16:15:54: MQTT publish: topic 'zigbee2mqtt/0x00158d00032b0b02', payload '{"current":0.4,"linkquality":0,"power":89,"state":"ON","voltage":234}'
debug 2021-05-27 16:15:55: Received Zigbee message from '0x00158d00032b0b02', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":86}' from endpoint 1 with groupID 0
info  2021-05-27 16:15:55: MQTT publish: topic 'zigbee2mqtt/0x00158d00032b0b02', payload '{"current":0.4,"linkquality":0,"power":86,"state":"ON","voltage":234}'
debug 2021-05-27 16:15:58: Received Zigbee message from '0x00158d00032b0b02', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":285474818,"imageType":768,"manufacturerCode":4454}' from endpoint 1 with groupID 0
debug 2021-05-27 16:15:58: Received Zigbee message from '0x00158d00032b0b02', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":285474818,"imageType":768,"manufacturerCode":4454}' from endpoint 1 with groupID 0
debug 2021-05-27 16:15:58: Received Zigbee message from '0x00158d00032b0b02', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":285474818,"imageType":768,"manufacturerCode":4454}' from endpoint 1 with groupID 0
debug 2021-05-27 16:16:52: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2021-05-27 16:21:52: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2021-05-27 16:26:52: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2021-05-27 16:31:52: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2021-05-27 16:36:52: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2021-05-27 16:41:52: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2021-05-27 16:46:52: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2021-05-27 16:51:52: Saving state to file /opt/zigbee2mqtt/data/state.json

finzec, 2021-05-28:
some more info from log:
debug 2021-05-28 11:12:25: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2021-05-28 11:17:25: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2021-05-28 11:22:25: Saving state to file /opt/zigbee2mqtt/data/state.json
debug 2021-05-28 11:27:14: Received MQTT message on 'zigbee2mqtt/0x60a423fffeff4ed9/set' with data 'OFF'
debug 2021-05-28 11:27:14: Publishing 'set' 'state' to '0x60a423fffeff4ed9'
debug 2021-05-28 11:27:15: Received MQTT message on 'zigbee2mqtt/0x60a423fffe6f1d25/set' with data 'OFF'
debug 2021-05-28 11:27:15: Publishing 'set' 'state' to '0x60a423fffe6f1d25'
debug 2021-05-28 11:27:17: Received MQTT message on 'zigbee2mqtt/0x60a423fffe6f1d25/set' with data 'ON'
debug 2021-05-28 11:27:17: Publishing 'set' 'state' to '0x60a423fffe6f1d25'
error 2021-05-28 11:27:20: Publish 'set' 'state' to '0x60a423fffeff4ed9' failed: 'Error: Command 0x60a423fffeff4ed9/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
debug 2021-05-28 11:27:20: Error: Command 0x60a423fffeff4ed9/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:67:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
info  2021-05-28 11:27:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x60a423fffeff4ed9' failed: 'Error: Command 0x60a423fffeff4ed9/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"0x60a423fffeff4ed9"},"type":"zigbee_publish_error"}'
debug 2021-05-28 11:27:22: Received MQTT message on 'zigbee2mqtt/0x804b50fffe0a6a68/set' with data 'ON'
debug 2021-05-28 11:27:22: Publishing 'set' 'state' to '0x804b50fffe0a6a68'
debug 2021-05-28 11:27:25: Saving state to file /opt/zigbee2mqtt/data/state.json
error 2021-05-28 11:27:26: Publish 'set' 'state' to '0x60a423fffe6f1d25' failed: 'Error: Command 0x60a423fffe6f1d25/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
debug 2021-05-28 11:27:26: Error: Command 0x60a423fffe6f1d25/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:67:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7)
info  2021-05-28 11:27:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x60a423fffe6f1d25' failed: 'Error: Command 0x60a423fffe6f1d25/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null,\"writeUndiv\":false}) failed (SRSP - AF - dataRequest after 6000ms)'","meta":{"friendly_name":"0x60a423fffe6f1d25"},"type":"zigbee_publish_error"}'
error 2021-05-28 11:27:32: Publish 'set' 'state' to '0x804b50fffe0a6a68' failed: 'Error: Command 0x804b50fffe0a6a68/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)'
debug 2021-05-28 11:27:32: Error: Command 0x804b50fffe0a6a68/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:67:35)
    at listOnTimeout (internal/timers.js:554:17)

Koenkk, 2021-05-28:
This is because your CC2531 crashes, try using the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing , if that doesn't help upgrade to a more powerful adapter: https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter

finzec, 2021-05-29:
Thank's a lot for the hint!
Yesterday I patched with suggested firmware and since then itwas not hanging anymore.
As it seems to be fixed, I will close.

closed 2021-05-29 by finzec