Koenkk/zigbee2mqtt #7617: Cant bind/unbind from frontend

smulle48, 2021-05-27:
Hi

Just installed z2m on a fresh Debian 10 box. I have tried both the z2m HA add-on and the z2m bare metal with the same error.

When everything is installed and I want to use the bind and unbind button in the Device Bind page, it does not work.

If I check the checkboxes for ex a remote, then I press the remote before clicking on the bind / unbind button, but then all the selections on the page gets deleted. I used to be able to use the frontend to bind and unbind, but now I have to use an MQTT client for everything.

## Debug info
Zigbee2MQTT version: 1.18.3 commit: 669489fe
Adapter hardware: CC2652 (slae.sh)
Adapter firmware version: zStack3x0 / 20210120
Frontend: 0.4.0

rindlerblabla, 2021-05-27:
There is an open issue https://github.com/nurikk/z2m-frontend/issues/549 and a fixed issue https://github.com/nurikk/z2m-frontend/issues/489 (in the latest dev) that might be related to this.

smulle48, 2021-05-28:
> There is an open issue [nurikk/z2m-frontend#549](https://github.com/nurikk/z2m-frontend/issues/549) and a fixed issue [nurikk/z2m-frontend#489](https://github.com/nurikk/z2m-frontend/issues/489) (in the latest dev) that might be related to this.

Unfortunately it's still not working in the latest dev release.

I've also noticed that the "action" under "Exposes" don't show the action when pushing the button. If I don't define legacy: false, the field says N/A and when defining the legacy: false it says empty_string but in the log I can see:

MQTT publish: topic 'zigbee2mqtt/0x14b457fffe918632', payload '{"action":"on","battery":5,"linkquality":54,"update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/0x14b457fffe918632', payload '{"action":"","battery":5,"linkquality":54,"update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/0x14b457fffe918632/action', payload 'on'

https://user-images.githubusercontent.com/24297824/119960447-44389500-bfa5-11eb-8447-cd4513c71452.mov

levihb, 2021-05-31:
Just to add that I am also having this issue. It's very frustrating and makes it pretty much unusable :(

levihb, 2021-06-04:
@smulle48 Clear your cache, mine is working now on the latest version after clearing my browser cache.

levihb, 2021-06-04:
@smulle48 mine is mostly working on the latest release. Clear your browser cache, I did and it appears to have mostly fixed it.

@rindlerblabla @smulle48  to be clearer when I fill in a new bind settings it never refreshes, but if I go to a previous bind and tick something in clusters, after a few seconds it refreshes and resets. It seems the fix has only partially worked.

closed 2021-07-12 by github-actions[bot]