Koenkk/zigbee2mqtt #7622: MGM220P - Unable to get battery percentage remaining and battery voltage

pierremathelin, 2021-05-28:
Hello,

I am currently working with MGM220P board from silicon labs running as a simple switch end-device.
I have a Zigbee2MQTT server based on a Raspberry Pi 3 with a CC2538 as a coordinator.

With my ED, I am sending ON/OFF messages correctly, see logs below:
`
Zigbee2MQTT:debug 2021-05-28 09:47:42: Received Zigbee message from '0x588e81fffe43edfa', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2021-05-28 09:47:42: MQTT publish: topic 'zigbee2mqtt_snoc/0x588e81fffe43edfa', payload '{"action":"on","linkquality":57}'
Zigbee2MQTT:info  2021-05-28 09:47:42: Configuring '0x588e81fffe43edfa'
`

I am now trying to get battery percentage and battery voltage but I am getting the following message and I am not able to see whats the problem:
`
Zigbee2MQTT:debug 2021-05-28 09:43:33: Received Zigbee message from '0x588e81fffe43edfa', type 'read', cluster 'genPowerCfg', data '["batteryPercentageRemaining"]' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-28 09:43:33: No converter available for 'MGM220PC22HNA' with cluster 'genPowerCfg' and type 'read' and data '["batteryPercentageRemaining"]'
Zigbee2MQTT:error 2021-05-28 09:43:40: Failed to configure '0x588e81fffe43edfa', attempt 2 (Error: ConfigureReporting 0x588e81fffe43edfa/1 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 45172 - 1 - 20 - 1 - 7 after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
`

My external converter is configured as following:
`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: [''],
    model: 'MGM220PC22HNA',
    vendor: 'SNOC',
    description: 'Module de détection de fermeture de goupille',
    fromZigbee: [fz.battery, fz.on_off, fz.command_on],
    toZigbee: [],
    exposes: [e.linkquality(), e.battery(), e.switch()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint);
        await reporting.batteryVoltage(endpoint);
    },
 };

module.exports = definition;
`

Could you please help me to fix the issue?
Thank you ! :)

Koenkk, 2021-05-28:
The timeout means that your device does not respond to the configure reporting command. So my first guess is that the problem is on the MGM220P side. You should be able to investigate this further by sniffing: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

MattWestb, 2021-05-28:
Using network analyzer in SS4 or SS5 is easier if you is having one WSTK.
Also look what wot commands is revived in the devices debug CLI.

pierremathelin, 2021-05-31:
Thank you for your replies.
I would check on the MGM220P side.
However, does my configuration on Zigbee2MQTT side seems to be correct (at least for the battery management side)?

I don't really have understood how the battery information are sent through Zigbee and I am not sure of my solution.
I was expecting to get a similar payload as Xiaomi door sensor which sends directly "contact", "battery" and "voltage" values.

MattWestb, 2021-05-31:
I have not cooking SED in SS but it shall being normal attribute report / attribute reading ala Zigbee standard (that Xiaomi is normally not doing).

Try reading the attribute from the device and sniffing with network analyzer what receiving and what the device is  writing in the debug log (device CLI).

Then reading is working OK you can looking on the reporting function.

I have seen that the Z3Light is not doing all attribute OK (Some is missing / not implanted) but the principle is working the same on ZR and SED.

pierremathelin, 2021-05-31:
Thanks @MattWestb.
I finally tried to send a normal reading and sniffed the data going through Zigbee.
There are my results:
[image]

And on the Zigbee2MQTT side:
[image]

Seeing the debug log, it seems that there is a frame with the battery percentage remaining which is received on the Zigbee2MQTT side.
However, nothing is read as a payload.
Can it be because of empty data or whatever else?

MattWestb, 2021-05-31:
Great one step in the right direction.

Its looks like you is requesting  the coordinator (0x0000) to reporting attribute to your device (A2F4) that is little the wrong way.

I wast trying requesting battery from one IKEA remote from the coordinator (0x0000):
```
ZigBee Application Support [8 bytes]
   - Frame Control: 0x40
   -      Frame Type: Data (0)
   -      Delivery Mode: Direct (0)
   -      Indirect Address Mode: Dest Endpoint Present (0)
   -      Security Enabled: false
   -      Ack Required: true
   -      Extended Header Present: false
   - Destination Endpoint: 0x01
   - Cluster Identifier: 0x0001
   - Profile Identifier: 0x0104
   - Source Endpoint: 0x01
   - APS Counter: 0xB6
ZigBee Cluster Library [5 bytes]
   - ZCL Frame Control: 0x00
   -      Frame Type: Global command (0)
   -      Manufacturer Specific: false
   -      Command Direction: Client to Server (0)
   -      Disable default response: false
   -      Reserved: 0x00
   - Sequence number: 0xBE
   - Command Identifier: ReadAttributes (0x00)
   - Attribute Ids: 0x0021
```
And the remote is replaying to the request with  the same sequence number (0xBE):
```
 ZigBee Application Support [8 bytes]
   - Frame Control: 0x40
   -      Frame Type: Data (0)
   -      Delivery Mode: Direct (0)
   -      Indirect Address Mode: Dest Endpoint Present (0)
   -      Security Enabled: false
   -      Ack Required: true
   -      Extended Header Present: false
   - Destination Endpoint: 0x01
   - Cluster Identifier: 0x0001
   - Profile Identifier: 0x0104
   - Source Endpoint: 0x01
   - APS Counter: 0x77
ZigBee Cluster Library [8 bytes]
   - ZCL Frame Control: 0x18
   -      Frame Type: Global command (0)
   -      Manufacturer Specific: false
   -      Command Direction: Server to Client (1)
   -      Disable default response: true
   -      Reserved: 0x00
   - Sequence number: 0xBE
   - Command Identifier: ReadAttributesResponse (0x01)
   - Attribute Id: battery percentage remaining (0x0021)
   - Status: SUCCESS (0x00)
   - Attribute Type: Unsigned 8-bit integer (0x20)
   - Int8u_0: 0x2F
```
Reading attribute is the coordinator that is sending to the device and the device replaying with one read attribute response.

If have all working and configured attribute reporting the device is sending report attribute to the coordinator that is repaying with default response but that is coming later.

Trying sending one read attribute from the coordinator to you device and sniffing if / what is replaying.

pierremathelin, 2021-06-02:
Hi, just to mention: I finally manage to make it work. I made a mistake while sending the battery level from the end device to the coordinator, which was the origin of my issue.
I finally use the "classic" fromZigbee "battery" which work perfectly for this kind of application.

Thank you guys for your time and you help.

closed 2021-06-02 by pierremathelin

MattWestb, 2021-06-02:
Great to here Pierre !!!

Then i coming so fare and cooking SED i asking you for advice !! ;-))

I still fighting with NCP and GP for the devs getting it working on the host side and also Zigbee Control Bridge for sending commands to other devices in the network from small controllers like ESP devices.

I wold like "patching" IKEA E1743 (2 button remote) for having 2 pare of dimmers on it and putting it inside normal power switches but i have not coming so  fare :-((