Koenkk/zigbee2mqtt #7623: Failed to configure supported device Eco-Dim.07 Zigbee & Z-wave dimmer

NitramGB, 2021-05-28:
## What happened
Device paired ok:

MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee & Z-wave dimmer ","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"Eco-Dim.07","supports_ota":false,"vendor":"EcoDim"},"friendly_name":"0x086bd7fffe5d35ca","ieee_address":"0x086bd7fffe5d35ca","status":"successful","supported":true},"type":"device_interview"}'

but then failed to configure:  

Failed to configure '0x086bd7fffe5d35ca', attempt 1 (TypeError: extend.light_onoff_brightness(...).configure is not a function
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/ecodim.js:16:51)
   at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.js:136:45)
   at Configure.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/configure.js:118:18)
   at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:384:44)
   at process._tickCallback (internal/process/next_tick.js:68:7))

NB ecodim.js currently contains:

    
{        zigbeeModel: ['Dimmer-Switch-ZB3.0'],        
    model: 'Eco-Dim.07',        
    vendor: 'EcoDim',        
    description: 'Zigbee &amp; Z-wave dimmer ',        
    extend: extend.light_onoff_brightness({noConfigure: true}),        
    whiteLabel: [{vendor: 'Iolloi', model: 'ID-EU20FW09'}],        
    configure: async (device, coordinatorEndpoint, logger) =&gt; {
            await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
            await reporting.onOff(endpoint);
            await reporting.brightness(endpoint);
        },
    },<br class="Apple-interchange-newline">

## What did you expect to happen
Before I upgraded from 1.18.1 the device configured ok and z2m received on/off and dimming value updates, and also 30 minute beacons. Now the device can be controlled OK by sending payloads to "state" and "brightness", but doesn't report on/off state or brightness when changed on the device, and beacons are no longer received every 30 minutes.

## How to reproduce it (minimal and precise)
Pair device with latest software.

## Debug info
Zigbee2MQTT version: 1.18.3
Adapter hardware: CC2652R
Adapter firmware version: CC2652R_coordinator_20210120.hex

Koenkk, 2021-05-28:
This device has accidentally  been removed in 1.18.3, the ecodim.js from the dev branch is not compatible with 1.18.3. You should update to the latest dev branch for this device to work. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html) (#7325 )

closed 2021-05-28 by Koenkk

NitramGB, 2021-05-29:
Thanks Koenkk, I can confirm this fixed things, didn't even have to repair anything.
BTW, this dimmer is of high quality, a trailing-edge dimmer that doesn't buzz, fades linearly, is a very reliable zigbee device, and of course because it is mains powered it acts as a router as well.