Koenkk/zigbee2mqtt #7625: Fail to support Aqara new switch E1, successful paired but nothing output with trigger

qukefu, 2021-05-28:
## Information about the device + link
Aqara new switch that support Zigbee 3.0 and has a better price about $10!

I followed the how_to and creat an new js file an enable Debug in config. 

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.switch.b1lc04'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'QBKG38LM', // Vendor model number, look on the device for a model number
    vendor: 'Aqara', // Vendor of the device (only used for documentation and startup logging)
    description: 'Aqara single key switch e1', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.humidity()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;

~`

**Repair the device I can See this below , But  when I trigged the switch nothing output. It seems the switch did not send any thing.**

`Received Zigbee message from '0x54ef4410000f2da6', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":20,"imageType":6536,"manufacturerCode":4447}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-28 23:54:51: Device '0x54ef4410000f2da6' announced itself
Zigbee2MQTT:info  2021-05-28 23:54:51: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410000f2da6","ieee_address":"0x54ef4410000f2da6"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-05-28 23:54:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x54ef4410000f2da6"},"type":"device_announced"}'
Zigbee2MQTT:debug 2021-05-28 23:54:51: Received MQTT message on 'zigbee2mqtt/bridge/config/groups' with data '1622217291791'
Zigbee2MQTT:info  2021-05-28 23:54:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[],"type":"groups"}'
Zigbee2MQTT:debug 2021-05-28 23:54:51: Received MQTT message on 'zigbee2mqtt/bridge/config/devices/get' with data '1622217291791'
Zigbee2MQTT:info  2021-05-28 23:54:51: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"dateCode":"20210120","friendly_name":"Coordinator","ieeeAddr":"0x00124b0022811e1c","lastSeen":1622217291799,"networkAddress":0,"softwareBuildID":"zStack3x0","type":"Coordinator"},{"dateCode":"12-23-2020","description":"Aqara single key switch e1","friendly_name":"0x54ef4410000f2da6","hardwareVersion":1,"ieeeAddr":"0x54ef4410000f2da6","lastSeen":1622217291783,"manufacturerID":4447,"manufacturerName":"LUMI","model":"QBKG38LM","modelID":"lumi.switch.b1lc04","networkAddress":49103,"powerSource":"Mains (single phase)","type":"EndDevice","vendor":"Aqara"}]'
Zigbee2MQTT:debug 2021-05-28 23:54:51: Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data '[{"dateCode":"20210120","friendly_name":"Coordinator","ieeeAddr":"0x00124b0022811e1c","lastSeen":1622217291799,"networkAddress":0,"softwareBuildID":"zStack3x0","type":"Coordinator"},{"dateCode":"12-23-2020","description":"Aqara single key switch e1","friendly_name":"0x54ef4410000f2da6","hardwareVersion":1,"ieeeAddr":"0x54ef4410000f2da6","lastSeen":1622217291783,"manufacturerID":4447,"manufacturerName":"LUMI","model":"QBKG38LM","modelID":"lumi.switch.b1lc04","networkAddress":49103,"powerSource":"Mains (single phase)","type":"EndDevice","vendor":"Aqara"}]'
Zigbee2MQTT:info  2021-05-28 23:54:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[{"dateCode":"20210120","friendly_name":"Coordinator","ieeeAddr":"0x00124b0022811e1c","lastSeen":1622217291804,"networkAddress":0,"softwareBuildID":"zStack3x0","type":"Coordinator"},{"dateCode":"12-23-2020","description":"Aqara single key switch e1","friendly_name":"0x54ef4410000f2da6","hardwareVersion":1,"ieeeAddr":"0x54ef4410000f2da6","lastSeen":1622217291783,"manufacturerID":4447,"manufacturerName":"LUMI","model":"QBKG38LM","modelID":"lumi.switch.b1lc04","networkAddress":49103,"powerSource":"Mains (single phase)","type":"EndDevice","vendor":"Aqara"}],"type":"devices"}'`

qukefu, 2021-05-28:
Here is the pairing log
`Zigbee2MQTT:warn  2021-05-29 00:26:41: Device '0x54ef4410000f2da6' left the network
Zigbee2MQTT:info  2021-05-29 00:26:41: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"ieee_address":"0x54ef4410000f2da6"},"type":"device_leave"}'
Zigbee2MQTT:info  2021-05-29 00:26:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x54ef4410000f2da6"},"type":"device_removed"}'
Zigbee2MQTT:debug 2021-05-29 00:26:41: Received MQTT message on 'zigbee2mqtt/bridge/config/groups' with data '1622219201185'
Zigbee2MQTT:debug 2021-05-29 00:26:41: Received MQTT message on 'zigbee2mqtt/bridge/config/devices/get' with data '1622219201185'
Zigbee2MQTT:info  2021-05-29 00:26:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[],"type":"groups"}'
Zigbee2MQTT:info  2021-05-29 00:26:41: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"dateCode":"20210120","friendly_name":"Coordinator","ieeeAddr":"0x00124b0022811e1c","lastSeen":1622219201191,"networkAddress":0,"softwareBuildID":"zStack3x0","type":"Coordinator"}]'
Zigbee2MQTT:debug 2021-05-29 00:26:41: Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data '[{"dateCode":"20210120","friendly_name":"Coordinator","ieeeAddr":"0x00124b0022811e1c","lastSeen":1622219201191,"networkAddress":0,"softwareBuildID":"zStack3x0","type":"Coordinator"}]'
Zigbee2MQTT:info  2021-05-29 00:26:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[{"dateCode":"20210120","friendly_name":"Coordinator","ieeeAddr":"0x00124b0022811e1c","lastSeen":1622219201197,"networkAddress":0,"softwareBuildID":"zStack3x0","type":"Coordinator"}],"type":"devices"}'
Zigbee2MQTT:info  2021-05-29 00:26:43: Device '0x54ef4410000f2da6' joined
Zigbee2MQTT:info  2021-05-29 00:26:43: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410000f2da6","ieee_address":"0x54ef4410000f2da6"},"type":"device_joined"}'
Zigbee2MQTT:info  2021-05-29 00:26:43: Starting interview of '0x54ef4410000f2da6'
Zigbee2MQTT:info  2021-05-29 00:26:43: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410000f2da6","ieee_address":"0x54ef4410000f2da6","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2021-05-29 00:26:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x54ef4410000f2da6"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-05-29 00:26:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x54ef4410000f2da6"},"type":"pairing"}'
Zigbee2MQTT:debug 2021-05-29 00:26:43: Device '0x54ef4410000f2da6' announced itself
Zigbee2MQTT:info  2021-05-29 00:26:43: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x54ef4410000f2da6","ieee_address":"0x54ef4410000f2da6"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-05-29 00:26:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x54ef4410000f2da6"},"type":"device_announced"}'
Zigbee2MQTT:debug 2021-05-29 00:26:43: Received MQTT message on 'zigbee2mqtt/bridge/config/groups' with data '1622219203649'
Zigbee2MQTT:debug 2021-05-29 00:26:43: Received MQTT message on 'zigbee2mqtt/bridge/config/devices/get' with data '1622219203649'
Zigbee2MQTT:info  2021-05-29 00:26:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[],"type":"groups"}'
Zigbee2MQTT:info  2021-05-29 00:26:43: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"dateCode":"20210120","friendly_name":"Coordinator","ieeeAddr":"0x00124b0022811e1c","lastSeen":1622219203657,"networkAddress":0,"softwareBuildID":"zStack3x0","type":"Coordinator"},{"description":"-","friendly_name":"0x54ef4410000f2da6","ieeeAddr":"0x54ef4410000f2da6","lastSeen":1622219203641,"networkAddress":38425,"type":"Unknown","vendor":"-"}]'
Zigbee2MQTT:debug 2021-05-29 00:26:43: Received MQTT message on 'zigbee2mqtt/bridge/config/devices' with data '[{"dateCode":"20210120","friendly_name":"Coordinator","ieeeAddr":"0x00124b0022811e1c","lastSeen":1622219203657,"networkAddress":0,"softwareBuildID":"zStack3x0","type":"Coordinator"},{"description":"-","friendly_name":"0x54ef4410000f2da6","ieeeAddr":"0x54ef4410000f2da6","lastSeen":1622219203641,"networkAddress":38425,"type":"Unknown","vendor":"-"}]'
Zigbee2MQTT:info  2021-05-29 00:26:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[{"dateCode":"20210120","friendly_name":"Coordinator","ieeeAddr":"0x00124b0022811e1c","lastSeen":1622219203703,"networkAddress":0,"softwareBuildID":"zStack3x0","type":"Coordinator"},{"description":"-","friendly_name":"0x54ef4410000f2da6","ieeeAddr":"0x54ef4410000f2da6","lastSeen":1622219203641,"networkAddress":38425,"type":"Unknown","vendor":"-"}],"type":"devices"}'
Zigbee2MQTT:debug 2021-05-29 00:26:45: Received Zigbee message from '0x54ef4410000f2da6', type 'readResponse', cluster 'genBasic', data '{"modelId":"lumi.switch.b1lc04"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-05-29 00:26:45: Received Zigbee message from '0x54ef4410000f2da6', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"LUMI"}' from endpoint 1 with groupID 0
[... cut ...] 4ef4410000f2da6","name":"0x54ef4410000f2da6 linkquality","state_topic":"zigbee2mqtt/0x54ef4410000f2da6","unique_id":"0x54ef4410000f2da6_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:debug 2021-05-29 00:26:58: Received Zigbee message from '0x54ef4410000f2da6', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":20,"imageType":6536,"manufacturerCode":4447}' from endpoint 1 with groupID 0
`

GmasteR, 2021-05-28:
I have Aqara E1 2 gang no netural, this code works for me (copy from D1 code).
But Decoupled mode not supported:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;	
const ea = exposes.access;

const preventReset = async (type, data, device) => {
    if (
        // options.allow_reset ||
        type !== 'message' ||
        data.type !== 'attributeReport' ||
        data.cluster !== 'genBasic' ||
        !data.data[0xfff0] ||
        // eg: [0xaa, 0x10, 0x05, 0x41, 0x87, 0x01, 0x01, 0x10, 0x00]
        !data.data[0xFFF0].slice(0, 5).equals(Buffer.from([0xaa, 0x10, 0x05, 0x41, 0x87]))
    ) {
        return;
    }
    const options = {manufacturerCode: 0x115f};
    const payload = {[0xfff0]: {
        value: [0xaa, 0x10, 0x05, 0x41, 0x47, 0x01, 0x01, 0x10, 0x01],
        type: 0x41,
    }};
    await device.getEndpoint(1).write('genBasic', payload, options);
};

const definition = {
    zigbeeModel: ['lumi.switch.b2lc04'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'QBKG39LM', // Vendor model number, look on the device for a model number
    vendor: 'Aqara', // Vendor of the device (only used for documentation and startup logging)
    description: 'Aqara 2 gang switch without zero line', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off, fz.xiaomi_multistate_action, fz.xiaomi_operation_mode_basic],
    toZigbee: [tz.on_off, tz.xiaomi_operation_mode_basic],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'left': 1, 'right': 2};
    },
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right'), e.action([
        'hold_left', 'single_left', 'double_left', 'release_left', 'hold_right', 'single_right',
        'double_right', 'release_right', 'hold_both', 'single_both', 'double_both', 'release_both'])],
    onEvent: preventReset
};

module.exports = definition;
```

qukefu, 2021-05-29:
Hi GmasteR

Thanks for replay, you give me hope.

I tried to copy from D1 single key code. But nothing changed. 

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const preventReset = async (type, data, device) => {
    if (
        // options.allow_reset ||
        type !== 'message' ||
        data.type !== 'attributeReport' ||
        data.cluster !== 'genBasic' ||
        !data.data[0xfff0] ||
        // eg: [0xaa, 0x10, 0x05, 0x41, 0x87, 0x01, 0x01, 0x10, 0x00]
        !data.data[0xFFF0].slice(0, 5).equals(Buffer.from([0xaa, 0x10, 0x05, 0x41, 0x87]))
    ) {
        return;
    }
    const options = {manufacturerCode: 0x115f};
    const payload = {[0xfff0]: {
        value: [0xaa, 0x10, 0x05, 0x41, 0x47, 0x01, 0x01, 0x10, 0x01],
        type: 0x41,
    }};
    await device.getEndpoint(1).write('genBasic', payload, options);
};

const definition = {
    zigbeeModel: ['lumi.switch.b1lc04'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'QBKG38LM', // Vendor model number, look on the device for a model number
    vendor: 'Aqara', // Vendor of the device (only used for documentation and startup logging)
    description: 'Aqara single key switch e1', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.xiaomi_battery, fz.xiaomi_on_off_action, fz.xiaomi_multistate_action],
    toZigbee: [],
    exposes: [e.battery(),e.action(['single', 'double', 'hold']),e.battery_voltage()],
    onEvent: **preventReset**
};

module.exports = definition;`

I found that the log stucked in 'Zigbee2MQTT:debug 2021-05-29 09:15:41: Received Zigbee message from '0x54ef4410000f2da6', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":20,"imageType":6536,"manufacturerCode":4447}' from endpoint 1 with groupID 0
```

Maybe I should upgrade the FW first. I will buy  an aqara gateway and try again. Thanks any way.

GmasteR, 2021-05-29:
Try to use endpoint 1 - single

Ant1q, 2021-06-14:
> I have Aqara E1 2 gang no netural, this code works for me (copy from D1 code).
> But Decoupled mode not supported:
>
> ```
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz [...]

Hello! How about feedback? I mean that with this convector I can control E1 from z2m UI, but when I click physically on switch there are no changes in state in z2m.

And i checked it with Xiaomi Gateway 3 and there are no double\long\release action, only left, right and both.

GmasteR, 2021-06-14:
This code work better, return state when physically switch
`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;	
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.switch.b2lc04'],
    model: 'QBKG39LM',
    vendor: 'Aqara',
    description: 'Aqara 2 gang switch without zero line',
    fromZigbee: [fz.on_off, fz.xiaomi_multistate_action, fz.xiaomi_operation_mode_basic, fz.aqara_opple_multistate, fz.aqara_opple_report, fz.command_toggle],
    toZigbee: [tz.on_off, tz.xiaomi_operation_mode_basic, tz.aqara_opple_operation_mode],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'left': 1, 'right': 2};
    },
    exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right'), e.action(['left_single', 'left_double', 'left_triple', 'left_hold', 'left_release',
            'right_single', 'right_double', 'right_triple', 'right_hold', 'right_release'])],
	configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        const endpoint2 = device.getEndpoint(2);
        // set "event" mode
        await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f});
        // turn on the "multiple clicks" mode, otherwise the only "single click" events.
        // if value is 1 - there will be single clicks, 2 - multiple.
        await endpoint1.write('aqaraOpple', {0x0125: {value: 0x02, type: 0x20}}, {manufacturerCode: 0x115f});
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genPowerCfg']);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
    },
};
module.exports = definition;
`

Ant1q, 2021-06-14:
`        await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f});
        // turn on the "multiple clicks" mode, otherwise the only "single click" events.
        // if value is 1 - there will be single clicks, 2 - multiple.
        await endpoint1.write('aqaraOpple', {0x0125: {value: 0x02, type: 0x20}}, {manufacturerCode: 0x115f});
        await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genPowerCfg']);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);`
This part produces a lot of errors. It can be deleted without any effect for basic functional.

I added this to converter 
`    onEvent: async (type, data, device) => {
         // The position (genAnalogOutput.presentValue) reported via an attribute contains an invaid value
         // however when reading it will provide the correct value.
         if (data.type === 'attributeReport' && data.cluster === 'genAnalogOutput') {
             await device.endpoints[0].read('genAnalogOutput', ['presentValue']);
         }
     },
`
and now get this in debug log, maybe it can be useful for someone

`Received Zigbee message from 'Aqara E1 Bedroom', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[100,16,0,101,16,0,3,40,30,5,33,7,0,154,32,0,8,33,20,1,11,32,0,10,33,8,12,12,32,1],"type":"Buffer"}}' from endpoint 1 with groupID 0`

kirovilya, 2021-06-16:
simple converter for both switches:
```
    {
        zigbeeModel: ['lumi.switch.b2lc04'],
        model: 'QBKG39LM',
        vendor: 'Aqara',
        description: 'Aqara E1 2 gang switch without zero line',
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
        meta: {multiEndpoint: true},
        endpoint: (device) => {
            return {'left': 1, 'right': 2};
        },
        exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('right')],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            // Change to 1 mode
            await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, 
                disableDefaultResponse: true, disableResponse: true});
        },
    },
```

when the operating mode is changed, the device is re-announced! but in this (1) mode the keystrokes are sent.
there is also mode 2 and 3 - completely different data is sent in these modes

kirovilya, 2021-06-16:
The worst thing is that when working with a native gateway, this switch uses its own command and response format, like Tuya
[image]

closed 2021-07-24 by github-actions[bot]

andreyazbarov, 2021-08-08:
Dear @Koenkk , thank you to add my implementation for **Aqara E1 1 gang smart switch QBKG38LM**. 
Not sure I'm sharing the code in a right way, nevertheless kindly find it below:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const preventReset = async (type, data, device) => { // PLEASE READ: can be deleted as /devices/xiaomi.js already have it. I used it only in my external converter - QBKG38LM.js
    if (
        // options.allow_reset ||
        type !== 'message' ||
        data.type !== 'attributeReport' ||
        data.cluster !== 'genBasic' ||
        !data.data[0xfff0] ||
        // eg: [0xaa, 0x10, 0x05, 0x41, 0x87, 0x01, 0x01, 0x10, 0x00]
        !data.data[0xFFF0].slice(0, 5).equals(Buffer.from([0xaa, 0x10, 0x05, 0x41, 0x87]))
    ) {
        return;
    }
    const options = {manufacturerCode: 0x115f};
    const payload = {[0xfff0]: {
        value: [0xaa, 0x10, 0x05, 0x41, 0x47, 0x01, 0x01, 0x10, 0x01],
        type: 0x41,
    }};
    await device.getEndpoint(1).write('genBasic', payload, options);
}; // PLEASE READ: THE END OF THE BLOCK THAT CAN BE DELETE WHEN MERGING WITH XIAOMI.JS

const definition = {
    zigbeeModel: ['lumi.switch.b1lc04'], // The model ID from: Device with modelID 'lumi.switch.b1lc04' is not supported.
    model: 'QBKG38LM', // Vendor model number, look on the device for a model number
    vendor: 'Xiaomi', // Vendor of the device (only used for documentation and startup logging)
    description: 'Aqara E1 1 gang smart switch', // Description of the device, copy from vendor site. 
    fromZigbee: [fz.on_off, fz.xiaomi_multistate_action], // 
    toZigbee: [tz.on_off, tz.xiaomi_switch_operation_mode_opple], // 
    exposes: [ e.switch(),
            exposes.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled']).withDescription('Decoupled mode'),
            e.action(['single', 'double'])], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
        onEvent: preventReset,
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            // set "event" mode
            await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f,
                disableDefaultResponse: true, disableResponse: true});
            await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'genPowerCfg']);
    },
};

module.exports = definition;
```

Sorry for any possible inconvinience. Just sharing the code if somebody need it.

Best regards,
Andrey

AndreyBar, 2021-08-08:
@andreyazbarov it works like a charm, thank you very much!

Koenkk, 2021-08-09:
@andreyazbarov added, thanks! 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

OrBin, 2022-10-18:
Hi @Koenkk, do you think it'd be possible to add `hold` action ("long click") support for `QBKG38LM`/`QBKG39LM`?
Looking at the logs, I see the following line when making a long click on the switches:
```
Debug 2022-10-19 00:08:32Received Zigbee message from 'my switch', type 'attributeReport', cluster 'aqaraOpple', data '{"252":0}' from endpoint 1 with groupID 0
```

Note: not sure if that's important or not, but - when I make a long click, the switch state still changes (immediately at the click, not "waiting" for the click to "become a long click").