Koenkk/zigbee2mqtt #7630: Battery Level for Tuya Blind Motor - Recognised as TS0601

rbrueckner82, 2021-05-29:
Information about the device + link

https://ko.aliexpress.com/item/1005001775235402.html?spm=a2g0s.9042311.0.0.24644c4dOe24TO

The Device don show Battery Level, what can i do to fix them?

robertalexa, 2021-06-01:
@Koenkk @sjorge

I have 2 different devices that report as ts0601_cover once is the AM43 and the other is M515EGBZTN.

They are both battery operated and allegedly they both report their battery levels but I have the same issue as above.

I am willing to fix it, if you could suggest an approach.

Here is what I have in mind, with potential downsides.
1. Add battery level support to the whole group of devices. But some of them are not battery, but wired. I am unsure if this could mess anything up for those device. This also seems like the simplest fix
2. Add support for each device individually (in this case just these 2 that I can test myself) - I guess set zigbeeModel and one single fingerprint?
3.  Split the current support into 2 zigbeeModels basically - battery and non battery - i can probably work out which one is which? 

These solutions would lead to code duplication

4. Another option would be to add battery to the exposes array if the device power source is Battery, but I feel that is not achievable inside the object itself.

Jorge sorry for randomly tagging you, but I felt you might have some valuable input and together we could help some people.

Thanks in advance

sjorge, 2021-06-01:
IIRC you can fingerprint based on cluster too (@Koenkk would be able to confirm), so you could nearly split these into 2 groups one that has the genPower for battery reporting and one that doesn't.

That would be the cleanest solution I think.

robertalexa, 2021-06-01:
@sjorge I think fingerprinting by just manufacturerName might be enough as I do not think you could find both wired and battery operated with the same manufacturer name.

Some difficulties I am envisioning while doing this:
- do not have any of the wired devices to test
- I am unsure how well i will be able to find devices based on manufacturerName in order to decide if they are wired or not.
- we also have the issue with motor_speed and that it appears to only be working with am43, as far as I have learnt from a different issue - should we consider this too while splitting it?

I guess the alternative is to create a new group for the battery operated ones, move the 2 devices that I have and can test with, and then with time more devices can be migrated as people flag issues with battery reporting? This sounds like a sane approach with the least risk of breaking people's devices - but I am open to discussions here.

I will wait for Koen's input and go from there.
Thanks

sjorge, 2021-06-01:
It's probably also worth splitting that am43 device then, looks like a bunch of blinds based on TuYa design were grouped together, while not being 100% identical in feature set.

robertalexa, 2021-06-08:
@Koenkk Any feedback on the above before I make a start? Thanks

stevetz, 2021-06-09:
Happy to help test - I have a generic AM43, this one came with a solar panel/light sensor. I'd also hope to be able to bring the light sensor value into z2m if possible down the line - but battery is my main concern now.

DomiStyle, 2021-06-09:
@robertalexa Did you find out how these devices report their battery level yet [since the last issue](https://github.com/Koenkk/zigbee2mqtt/issues/7160#issuecomment-824249648)?

As far as I know they don't use the genPowerCfg cluster but manuSpecificTuya. However, you can check for both in the debug logs.

There's a battery type report in [tuya.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/36cf29c23c1d8bfd08fb080d8947acecb99aca3e/lib/tuya.js#L204) but I haven't seen it being used by Tuya covers yet.

robertalexa, 2021-06-09:
@DomiStyle Hey Domi, glad to see you join this, I think you are the "guardian" of the covers? haha.

To be perfectly honest I did not spend any time looking at this yet, I was hoping I would get Keonk's opinion on grouping these devices before making a start. What is your view on how to group them anyway?

I am hoping to have the time this week to have a play, I might be successful or I might fail misserably, we shall see.

DomiStyle, 2021-06-09:
@robertalexa I guess you could say that for the Tuya covers. :)

I don't think they need to be split unless the functionality differs too much. Of course it's up to @Koenkk to determine how to group them. The biggest problem with the Tuya devices is that all of them are really similar but never exactly the same.

Since the device itself already reports whether or not they are a battery device or not I don't think it matters all that much since there just won't be a battery report for devices that have no batteries.

It seems like the Tuya thermostats do report their battery as shown [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/7cccaae2e714e67dff4f24bfd82e5a6116797754/converters/fromZigbee.js#L3300). Although I haven't seen any covers send that report as of now. If they do support it, it can easily be added [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/7cccaae2e714e67dff4f24bfd82e5a6116797754/converters/fromZigbee.js#L1683).

You can add some logging before the switch/case to check if anything arrives that might point at a battery readout. Assuming it does regular reporting it should just appear but you might have to trigger it by moving the cover or depleting the battery.

robertalexa, 2021-06-09:
@DomiStyle Thanks for your input. I take it you had a go at this but were unsuccessful?

I have been having  a play since your last reply, but it is damn hard without specs to know what the device is meant to do. I also don't have Tuya Gateway so there is no option for sniffing.

I started by asking Zemismart for specs for AM43 and M515EGBZTN, we shall see what they come up with, they have usually been friendly and helpful when I have asked technical questions.

Looking at M515, the only DP I have managed to see in logs are 1 (state) and 2 (cover position). Nothing else.

But what gets me super confused (and that may be because I am still learning about z2m and herdsman)  is that we are not listening to cluster genPowerCfg, unless there would be some magic that should but I am unsure. Maybe you can comments on this for me?

But equally we are not listening to genBasic, yet I am seeing the data for it - which adds to my confusion and probably shows that I am missing something

This is a JSON.stringify dump of the msg
```
{"type":"commandGetData","device":{"ID":2,"_type":"EndDevice","_ieeeAddr":"0x5c0272fffeedd7a5","_networkAddress":45179,"_manufacturerID":4098,"_endpoints":[{"ID":1,"profileID":260,"deviceID":81,"inputClusters":[0,4,5,61184],"outputClusters":[25,10],"deviceNetworkAddress":45179,"deviceIeeeAddress":"0x5c0272fffeedd7a5","clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_xuzcvlku","powerSource":3,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"_binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e32e48","endpointID":1}],"_configuredReportings":[],"meta":{},"pendingRequests":[]}],"_manufacturerName":"_TZE200_xuzcvlku","_powerSource":"Battery","_modelID":"TS0601","_applicationVersion":68,"_stackVersion":0,"_zclVersion":3,"_hardwareVersion":1,"_dateCode":"","_interviewCompleted":true,"_interviewing":false,"_skipDefaultResponse":false,"meta":{},"_lastSeen":1623245252121,"_linkquality":23},"endpoint":{"ID":1,"profileID":260,"deviceID":81,"inputClusters":[0,4,5,61184],"outputClusters":[25,10],"deviceNetworkAddress":45179,"deviceIeeeAddress":"0x5c0272fffeedd7a5","clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_xuzcvlku","powerSource":3,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"_binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0018e32e48","endpointID":1}],"_configuredReportings":[],"meta":{},"pendingRequests":[]},"data":{"status":0,"transid":0,"dp":2,"datatype":2,"fn":0,"data":{"type":"Buffer","data":[0,0,0,65]}},"linkquality":23,"groupID":0,"cluster":"manuSpecificTuya","meta":{"zclTransactionSequenceNumber":91,"manufacturerCode":null,"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0}}}
```

Can you please shed some light for me @DomiStyle ?

The files I am looking at are:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/7cccaae2e714e67dff4f24bfd82e5a6116797754/converters/fromZigbee.js#L1673

https://github.com/Koenkk/zigbee-herdsman-converters/blob/7cccaae2e714e67dff4f24bfd82e5a6116797754/devices/tuya.js#L484

DomiStyle, 2021-06-09:
@robertalexa I only have a Tuya curtain, which is powered by mains so I don't have any devices to test with batteries.

There are only converters for information that is useful, for example:

If you have debug output enabled you will see data that is received with messages like these:
```zigbee2mqtt_1    | Zigbee2MQTT:debug 2021-06-03 21:20:50: Received Zigbee message from 'livingroom_curtain_main', type 'attributeReport', cluster 'genBasic', data '{"appVersion":73}' from endpoint 1 with groupID 0'```

My curtain reports the appVersion over and over again, which is no info we need, hence why there is no converter for it.

z2m already takes care of setting the basic device information like manufacturer name and so on, even without any converters for a device.

If you receive a genPowerCfg from your motor you know that you need to add genPowerCfg to your converters in order to read the battery state, but I still think the battery readout is probably hidden in manuSpecificTuya since the other Tuya devices do it like this.

So the easiest way would be to enable debug logs, add more logging to the tuya_cover converter [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/7cccaae2e714e67dff4f24bfd82e5a6116797754/converters/fromZigbee.js#L1683). Wait 24-48 hours and see if there are any unhandled clusters or datapoints. If not you might have to trigger them by depleting the battery or moving the motor up/down.

robertalexa, 2021-06-09:
@DomiStyle I am currently having logging with debugging on, but no such information is present unfortunately. Running the motor does not respond with anything but position.

0 messages from the blind after 2.30 hours of stand by connection.

Zemismart asked for my email address so i might be getting some specs, but who knows.

I will try to continue monitoring and see what happens. Thanks for your input

robertalexa, 2021-06-11:
After a few messages, Zemismart refused to share any specifications with me. So i guess that is a dead end, or at least as far as I am able to do anything at my end.

Maybe someone with a Tuya gateway can sniff the traffic and tell us more.

Sorry guys

bachoo786, 2021-07-12:
@robertalexa any updates with the battery on this tuya blind motor?

robertalexa, 2021-07-16:
> @robertalexa any updates with the battery on this tuya blind motor?

zemismart kept telling me that it isn't possible to report battery level but the "factory" is also refusing to share the zigbee specification docs, so can't check for myself either.

I won't pursue this any further i'm afraid

elgerg, 2021-08-04:
Thats a shame. I ended up here as well.

I was really impressed with the driver in terms of reporting open/closed. Position etc but the icing on the cake would be batter reporting.

I can get a gateway and attempt a sniff if someone can point me in the direction required? I do have a spare ccxx dongle lying around and a cc debugger so i can re-flash it if need be. I'm assuming the gateways are cheap like everywhere else?

DomiStyle, 2021-08-04:
There's a guide for sniffing traffic here: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

elgerg, 2021-08-04:
Thanks!
I've ordered one of these:
https://www.aliexpress.com/item/1005002248204422.html?spm=a2g0s.9042311.0.0.68cb4c4diMxdkU
Bundle A
We will see what happens in a couple of weeks when it arrives..

Kind regards

robertalexa, 2021-08-04:
> Thanks!
> I've ordered one of these:
> https://www.aliexpress.com/item/1005002248204422.html?spm=a2g0s.9042311.0.0.68cb4c4diMxdkU
> Bundle A
> We will see what happens in a couple of weeks when it arrives..
> 
> Kind regards

Keep us updated man. I have a tuya gateway in my aliexpress basket too, just never decided to get it to look into it.

Let me know if you need help with what you are finding, testing or implementing

elgerg, 2021-08-05:
@robertalexa have you seen this?
https://drive.google.com/file/d/1dxoNUbEzo9nRPfDgW8SK9VMfC8qud5ou/view
[image]

Thanks!

robertalexa, 2021-08-05:
@elgerg No I have not, but can't see anything battery/power related though?

elgerg, 2021-08-05:
@robertalexa no. I searched for bat and power and found nothing.. :(

elgerg, 2021-08-05:
That file also has a link to this:
https://github.com/iquix/Smartthings/blob/master/devicetypes/iquix/tuya-window-shade.src/tuya-window-shade.groovy
It seems to describe the different TS601s..

WRLPDZ, 2021-08-16:
Hi,

I paste debug messages from AM43 in my Z2M.  
https://pastebin.pl/view/4ad8a900

I made 4 runs with changes:
- power cable plugged
- power cable unplugged
- solar panel plugged
- solar panel and power cable plugged
In each run blind was changed from open to close and from close to open.

I do not see any difference between each run. Always reported power source is a battery.
In Z2M logs I do not see info about battery % or light sensor info.
I will try to sniff tuya app when I will find somewhere my gateway.

robertalexa, 2021-08-16:
> I will try to sniff tuya app when I will find somewhere my gateway.

Please do let us know if you find anything.

@elgerg have you managed to sniff anything battery related?

Thanks

elgerg, 2021-08-16:
@robertalexa 

I have received my hub and have started to look through the detail on how to sniff the traffic.

Is there a particular firmware that I need to load onto the CC2531 or will any do?

Thanks

Barneybaer84, 2021-09-02:
Maybe can help this to add battery.
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5238#issuecomment-907350161

Ramshield, 2022-01-09:
> Ok, I connected my motor up with the app/tuya gateway and there is no mention of battery level. I guess it does not report it.
> 
> What a shame. How daft to produce a battery operated device that does not advertise the amount of power left... Sometimes I think these people do not think about what they are doing!

So I wasn't the only one who tried to add this. 
Worth a try to contact TuYa and ask if (and if yes how) they expose this over UART protocol? (https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#_6-bonus-contacting-the-manufacturer)

robertalexa, 2022-01-09:
I have done the same yesterday and got nothing about battery using the tuya gateway. So yeah...

It appears like the Aqara E1 which is the same AM43 device at least on the outside, might have different firmware, is reporting battery level using the aqara hub. But looking at the z2m files that has not been implemented yet, so i guess no one tried to sniff it yet. Koen tried to help some people with an external converter, the battery was reported but the value was 0.

I have lost faith with this device

Ramshield, 2022-01-09:
@robertalexa I don't mind writing support for Aqara E1. How sure are you about the exposing battery? I'd like to know before spending $60...

robertalexa, 2022-01-09:
@Ramshield if you check their website it shows in their screenshots of the app. It also shows on demo videos and product pages. But you will also need an aqara hub in order to sniff the original traffic

robertalexa, 2022-01-09:
But i am not sure it supports motor speed. So we might win something and lose something else....

Ramshield, 2022-01-09:
@robertalexa Thanks for checking! I ordered the device, should arrive tomorrow. Once I know more (should be either tomorrow or the day after)  I'll update here.

robertalexa, 2022-01-09:
@Ramshield this is the ongoing thread for the E1 driver
https://github.com/Koenkk/zigbee2mqtt/issues/9447

I would suggest once you get some more facts you post them there to keep things together. Feel free to tag me there :)

AF360, 2022-01-17:
@robertalexa The linked thread #9447 is not "ongoing". 
Instead it has been closed and the last post links exactly here.
So no matter how this device is called - E1 or TS0601 - there is still no battery status until today and it looks like no one is workling on it?

robertalexa, 2022-02-04:
@Ramshield It seems in the latest version of z2m the E1 is now properly showing battery levels. So i guess it is times for us all to buy the E1?

https://github.com/Koenkk/zigbee2mqtt/discussions/11170#discussioncomment-2091251
https://github.com/Koenkk/zigbee2mqtt/issues/9447#issuecomment-1021071646

Ramshield, 2022-02-04:
@robertalexa I got mine and realized it didn't come with a solar panel so I returned mine. I am planning on having about 5-10 devices through my house, keeping all the batteries full while some of them are used every day seemed too much of a hassle. 

Unrelated to Z2m I also haven't gotten it working perfectly, since the chain keeps slips through the gear, so not sure if this device is my solution...

luisgarcia87, 2022-03-16:
> @robertalexa I got mine and realized it didn't come with a solar panel so I returned mine. I am planning on having about 5-10 devices through my house, keeping all the batteries full while some of them are used every day seemed too much of a hassle.
> 
> Unrelated to Z2m I also haven't gotten it working perfectly, since the chain keeps slips through the gear, so not sure if this device (or similar) is my solution...

I had the same problem, check if your motor is as tense as possible to move the chain successfully (I had to adjust the distance a few times before securing it to the wall) Also add some WD40 to the gear of your blind (not to the motor), it will make it move smoother, it worked for me.

Aepx, 2023-03-09:
Just found this thread, I have a few of these and the battery shows no status, has there been any updates on this?

cokeman0, 2023-05-01:
just recived 2 of these, and also like to get battery and light sensor from the solar panel data

bubez81, 2023-10-23:
there are news about this problem maybe?

arrikhan, 2024-06-24:
I have both a bluetooth model and the zigbee model. The Bluetooth model connects via ESPHome setup and has battery/light sensor exposed. I don't know how similar these things are but can the knowledge from esphome help with the zigbee model? They are both AM43 models.

Bluetooth AM43 code is here --> https://github.com/esphome/esphome/tree/dev/esphome/components/am43/sensor

trizmark, 2024-07-02:
@arrikhan Could you post a link to the model you've got? I bought the zigbee one ages ago to test, but the lack of battery level reporting is a deal-breaker. I need another 6 to automate my downstairs curtains.

arrikhan, 2024-07-02:
> @arrikhan Could you post a link to the model you've got? I bought the zigbee one ages ago to test, but the lack of battery level reporting is a deal-breaker. I need another 6 to automate my downstairs curtains.

The Bluetooth model? It's no longer available to purchase. The new bluetooth models don't seem to ship with solar panel for whatever reason..

Here is the link [https://www.aliexpress.com/item/4000156401936.html](url)

bubez81, 2025-08-27:
Are you sure worked before??? I never saw the battery status on this courtain motor

Feriman22, 2025-08-27:
@bubez81, yes. It's working for me with Z2M 2.6.0, check this:

[image]