Koenkk/zigbee2mqtt #7636: Light entity data ending up in other entity attributes for specific lights?

levihb, 2021-05-30:
## What happened

When I connect a few specific Hue bulbs, I only end up with "linkquality", "update available", and "update state" entities. There is no entity for the light itself. But it appears as though Zigbee2MQTT does send the config, e.g. here are the messages I found using `mosquitto_sub`:

```json
homeassistant/light/0x00178801084237d6/light/config
{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"brightness":true,"brightness_scale":254,"color_mode":true,"command_topic":"zigbee2mqtt/0x00178801084237d6/set","device":{"identifiers":["zigbee2mqtt_0x00178801084237d6"],"manufacturer":"Philips","model":"Hue White and Color Ambiance GU10 (929001953101)","name":"0x00178801084237d6","sw_version":"Zigbee2MQTT 1.18.3"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"json_attributes_topic":"zigbee2mqtt/0x00178801084237d6","max_mireds":500,"min_mireds":150,"name":"0x00178801084237d6","schema":"json","state_topic":"zigbee2mqtt/0x00178801084237d6","supported_color_modes":["xy","color_temp"],"unique_id":"0x00178801084237d6_light_zigbee2mqtt"}

homeassistant/sensor/0x00178801084237d6/linkquality/config
{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00178801084237d6"],"manufacturer":"Philips","model":"Hue White and Color Ambiance GU10 (929001953101)","name":"0x00178801084237d6","sw_version":"Zigbee2MQTT 1.18.3"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00178801084237d6","name":"0x00178801084237d6 linkquality","state_topic":"zigbee2mqtt/0x00178801084237d6","unique_id":"0x00178801084237d6_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}

homeassistant/sensor/0x00178801084237d6/update_state/config
{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00178801084237d6"],"manufacturer":"Philips","model":"Hue White and Color Ambiance GU10 (929001953101)","name":"0x00178801084237d6","sw_version":"Zigbee2MQTT 1.18.3"},"icon":"mdi:update","json_attributes_topic":"zigbee2mqtt/0x00178801084237d6","name":"0x00178801084237d6 update state","state_topic":"zigbee2mqtt/0x00178801084237d6","unique_id":"0x00178801084237d6_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}

homeassistant/binary_sensor/0x00178801084237d6/update_available/config
{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00178801084237d6"],"manufacturer":"Philips","model":"Hue White and Color Ambiance GU10 (929001953101)","name":"0x00178801084237d6","sw_version":"Zigbee2MQTT 1.18.3"},"json_attributes_topic":"zigbee2mqtt/0x00178801084237d6","name":"0x00178801084237d6 update available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x00178801084237d6","unique_id":"0x00178801084237d6_update_available_zigbee2mqtt","value_template":"{{ value_json.update_available}}"}
```

To me it looks like it correctly sends the information? But the light entity never shows up. The light does show up properly in the Z2M front-end, and I can easily control it. When trying to look for the light through the developer settings I found:

[image]

As you can see above, it appears as though all of the light information has ended up in the other entities? And this correctly updates, e.g. if I turn the light off:

[image]

This doesn't seem to happen with all of the bulbs, it does with the following:

[Hue white ambiance GU10 (8718696598283)](https://www.zigbee2mqtt.io/devices/8718696598283.html#philips-8718696598283)

[Hue White and Color Ambiance GU10 (929001953101)](https://www.zigbee2mqtt.io/devices/929001953101.html#philips-929001953101)

It however does not happen with:

[Hue white GU10 bluetooth (LWG004)](https://www.zigbee2mqtt.io/devices/LWG004.html#philips-lwg004)

[Hue white filament Edison E27 LED (929002241201)](https://www.zigbee2mqtt.io/devices/929002241201.html#philips-929002241201)

[Hue dimmer switch (324131092621)](https://www.zigbee2mqtt.io/devices/324131092621.html#philips-324131092621)

As for MQTT, I am using the Mosquitto broker.

## What did you expect to happen

Expected the light entity to show up and work normally, like the other examples above.

## How to reproduce it (minimal and precise)

Just add one of the bulbs I listed above on the latest version.

## Debug info
Zigbee2MQTT version: 1.18.3-1
Adapter hardware: zzh!
Adapter firmware version: 2021-01-20 (latest on git)

Koenkk, 2021-05-30:
Are you on the latest HA version?

levihb, 2021-05-30:
@Koenkk oh wow embarrassing. No I was on the previous 2021.4 version. It didn't seem like much of a problem as I had updated a few weeks ago.

I upgraded to 2021.5 and now they suddenly all appeared! At least I assume it was that, HassOS also had a minor update.

Was this a recent bug in HA then? Or do you know how to explain why this happened?

Koenkk, 2021-05-31:
See first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.18.3

closed 2021-05-31 by Koenkk