Koenkk/zigbee2mqtt #7637: Triple click WXKG11LM not working

dheuts90, 2021-05-30:
## What happened
I press the button of a Aqara quickly three or four times and nothing shows up in the logs. Single, double and hold are working.

In problem #5874 I've read it doesnt work with all models. But I'm sure it worked with the WXKG11LM I use. I Have 3 of them and it always worked fine. Until I found out last week it's not working anymore.

## What did you expect to happen
To make triple and quadruple click of the button work, and See triple and four clicks also back in the logs

## How to reproduce it (minimal and precise)
Triple or quadruple push the button and check the log.

## Debug info
Zigbee2MQTT version: Newest
Adapter hardware: CC2531
Adapter firmware version: Router

Koenkk, 2021-05-30:
As documented here, not all versions of this device send the triple event: https://www.zigbee2mqtt.io/devices/WXKG11LM.html#actions

dheuts90, 2021-05-31:
Good morning Koen,

Thanks a lot for your quick reply.
I changed the logging setting to debug, but I don't get any log when clicking 3 or 4 times fast.
When I click 1 or 2 times, it just works fine. Any idea how this is possible?

In the meantime I switched from default routing to source routing firmware. Could this be te problem?

Koenkk, 2021-05-31:
@dheuts90 I don't know how it could happen, but it seems the issue is at the button side. (since it should send messages on a tripple click). Changing z2m versions/fw versions will not fix it.

shirou93, 2021-06-01:
@Koenkk  i have same issue and 

'Xiaomi WXKG11LM Logs 'dict object' has no attribute 'click' when rendering '{{ value_json.click }} in home asssitant'

closed 2021-07-09 by github-actions[bot]

amilanov75, 2022-10-16:
What ever happened with this issue? It looks like it has never been resolved.

I have just installed the device and also can't get triple or quadruple click to work.
Release also didn't work, however after excluding and including again it started to work.

FYI, I have set legacy = false in the config file

TomAnks, 2024-03-10:
According to the Z2MQTT documentation, the WXKG11LM has never supported triple or quadruple clicks -  https://www.zigbee2mqtt.io/devices/WXKG11LM.html

Bjoernsen, 2024-07-27:
Today, I moved this button from deCONZ to Zigbee2MQTT. Using deCONZ, I was able to create a working automation using a triple click. I was also able to select quadruple clicks. Using Zigbee2MQTT, I can only select singe and double clicks.

So the button itself is able to handle triple and quadruple clicks.

The old automation config looked like

```json
device_id: ID
domain: deconz
platform: device
type: remote_button_triple_press
subtype: turn_on
```