Koenkk/zigbee2mqtt #7638: unit of measurement is missing (GZCGQ01LM)

Alfagek, 2021-05-30:
## What happened
No Unit of measurement
## What did you expect to happen

## How to reproduce it (minimal and precise)
Unit off measurment is missing from "illuminance".
"illuminance Lux", "Battery", "Signal" are good
See also: https://github.com/home-assistant/core/issues/50674
## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC2531
Adapter firmware version:

Koenkk, 2021-05-30:
the `illuminance` is a legacy sensor, this is a raw value and has no unit. the `illuminance_lux` is the illuminance in lux (calculated from the raw value).

closed 2021-05-30 by Koenkk

Alfagek, 2021-05-30:
Okay thanks for the explanation @Koenkk. But because the “illuminance” is more exact, I use that in Home Assistant. But because it has no unit I can’t use it in home assistant automation UI

Alfagek, 2021-06-04:
@Koenkk , is there a other posibillity to set a unit in my own config to use the "illuminance"?

Koenkk, 2021-06-04:
Adding this to your config should do it:

```
devices:
  "0x12345678":
    friendly_name: my_sensor
    homeassistant:
      illuminance:
        unit_of_measurement: 'lx'
```