Koenkk/zigbee2mqtt #7646: EchoStar SAGE Doorbell Sensor (SAGE206612) - Reporting issue (homeassistant)

tabnul, 2021-05-30:
## What happened
2 issues with this device
- The SAGE Doorbell sensor automatic reports, only containing LQI and last_seen, dont get propagated to home assistant.
- battery is always empty and i dont think the device supports this even. should be removed i think. as per https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3186

## What did you expect to happen
- To have every status update/report transferred to home assistant.
- to not have battery in this device since its empty anyway
- 
## How to reproduce it (minimal and precise)
The doorbell sensor automatic report messages dont get propagated to home assistant. (i am targeting last_seen primarily, want to see if the device is still alive)
The data DOES get propagated once i ring the doorbell.
Status in zigbee2mqtt always seems fine (gui)

I suspect it has something to do with the fact that automatic messages dont have all the properties inside. 
Doorbell ring:
```
{
    "action": bell1,
    "battery": null,
    "last_seen": "2021-05-30T18:25:45.236Z",
    "linkquality": 96
}
```
normal report:
```
{
    "last_seen": "2021-05-30T18:34:38.109Z",
    "linkquality": 96
}
```

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC1352P-2
Adapter firmware version: 2021-02

tabnul, 2021-05-30:
The strange thing is that while monitoring the MQTT topic of zigbee2mqtt, no updates are visible there. Basically it is the same as visible in home assistant.

However in the zigbee2mqtt GUI, the state does get updated!

fludstank, 2021-06-02:
> The strange thing is that while monitoring the MQTT topic of zigbee2mqtt, no updates are visible there. Basically it is the same as visible in home assistant.
> 
> However in the zigbee2mqtt GUI, the state does get updated!

Same. I do get reports under homeassistant/sensor/(IEEE Address) in my MQTT broker, but not under zigbee2mqtt/(Device Name)

closed 2021-07-10 by github-actions[bot]