Koenkk/zigbee2mqtt #7647: Tuya Smart Zigbee Circuit Breaker 2P integration

n00bsteam, 2021-05-30:
Hello! Can you help me with this device?

## Information about the device + link
https://www.aliexpress.com/item/1005002605097816.html?spm=a2g0s.9042311.0.0.735f33edR1e6Mp&_ga=2.207067290.861667168.1622402627-244894434.1577472696

Zigbee intelligent recloser circuit breaker

## data/database.db entry of the device
```{"id":4,"type":"Router","ieeeAddr":"0x804b50fffe2c223a","nwkAddr":12671,"manufId":4098,"manufName":"_TZ3000_1hwjutgo","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��'(\u0017��'(\u0013��'(\u0012\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012","65506":30,"modelId":"TS011F","manufacturerName":"_TZ3000_1hwjutgo","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1622406053567}```

n00bsteam, 2021-06-01:
Some data from device:
```
ug Received Zigbee message from '0x804b50fffe2c223a', type 'attributeReport', cluster 'genBasic', data '{"65506":30,"appVersion":65}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x804b50fffe2c223a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Debug Received Zigbee message from '0x804b50fffe2c223a', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
```

n00bsteam, 2021-06-01:
And my external_intergation config:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS011F',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZ3000_1hwjutgo'
        },
    ],
    model: 'ZJSB9-80Z-Zigbee',
    vendor: 'Mumubiz',
    description: 'Automatic Switch',
    supports: 'overload protection , short-circuit protection , isolating protection',
    fromZigbee: [
        // fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    // onEvent: tuya.setTime, // Add this if you are getting no converter for 'commandSetTimeRequest'
    meta: {
        configureKey: 1,
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
};

module.exports = definition;
```

n00bsteam, 2021-06-08:
Okay, looks like i fix it:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = {
    switch: (options={}) => {
        const exposes = [e.switch()];
        const fromZigbee = [fz.on_off, fz.ignore_basic_report];
        const toZigbee = [tz.on_off];
        return {exposes, fromZigbee, toZigbee};
    },
};
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_1hwjutgo'}],
        model: 'ZJSB9-80Z',
        vendor: 'Mumubiz',
        description: 'Automatic Switch',
        extend: extend.switch(),
        whiteLabel: [{vendor: 'Mumubiz', model: 'ZJSB9-80Z'}],
    };

module.exports = definition;
```

Koenkk, 2021-06-10:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-06-10 by Koenkk