Koenkk/zigbee2mqtt #7648: Add Support for Gen 2. Hybrid E-Wand

Leah-Anderson, 2021-05-31:
## Information about the device + link
Device Link - https://www.myewand.com/product-page/gen-2-hybrid-e-wand-single-unit 
Description: This is a blind tilt motor that has a Zigbee receiver for automating window blinds. 

## data/database.db entry of the device
{"id":12,"type":"EndDevice","ieeeAddr":"0x84fd27fffe80e0d5","nwkAddr":20283,"manufId":4707,"manufName":"Current Products Corp","powerSource":"Battery","modelId":"E-Wand","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,1,3,4,5,6,8,32,258,2821,64528],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"E-Wand","manufacturerName":"Current Products Corp","powerSource":3,"zclVersion":3,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1622423514179}

## Device Debug log during pairing
```
info  2021-05-27 19:16:46: Device '0x84fd27fffe80e0d5' joined
info  2021-05-27 19:16:46: Starting interview of '0x84fd27fffe80e0d5'
debug 2021-05-27 19:16:47: Device '0x84fd27fffe80e0d5' announced itself
debug 2021-05-27 19:16:47: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,3,80,80,24,0,0,0,4,255,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:16:48: Received Zigbee message from '0x84fd27fffe80e0d5', type 'readResponse', cluster 'genBasic', data '{"modelId":"E-Wand"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:16:49: Received Zigbee message from '0x84fd27fffe80e0d5', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Current Products Corp"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:16:50: Received Zigbee message from '0x84fd27fffe80e0d5', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
debug 2021-05-27 19:16:51: Received Zigbee message from '0x84fd27fffe80e0d5', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2021-05-27 19:16:54: Received Zigbee message from '0x84fd27fffe80e0d5', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2021-05-27 19:16:55: Received Zigbee message from '0x84fd27fffe80e0d5', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2021-05-27 19:16:56: Received Zigbee message from '0x84fd27fffe80e0d5', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2021-05-27 19:16:57: Received Zigbee message from '0x84fd27fffe80e0d5', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
debug 2021-05-27 19:16:58: Received Zigbee message from '0x84fd27fffe80e0d5', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
info  2021-05-27 19:16:58: Successfully interviewed '0x84fd27fffe80e0d5', device has successfully been paired
info  2021-05-27 19:16:58: Device '0x84fd27fffe80e0d5' is supported, identified as: Current Products Corp Gen. 2 Hybrid E-Wand (CP180335E-01)
debug 2021-05-27 19:17:27: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,7,128,200,0,0,0,0,2,0,0,0,186,232,64,0,0,0,0,255,255,0,0,179,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:18:06: Received Zigbee message from '0x84fd27fffe80e0d5', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":24,"imageType":80,"manufacturerCode":4707}' from endpoint 1 with groupID 0
debug 2021-05-27 19:18:32: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,18,128,200,0,0,0,0,2,0,0,0,240,95,64,0,0,0,0,255,255,0,0,180,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:19:37: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,28,128,200,0,0,0,0,2,0,0,0,40,39,65,0,0,202,65,255,255,0,0,180,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:20:42: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,38,128,200,0,0,0,0,2,0,0,0,208,109,64,0,0,202,65,255,255,0,0,180,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:21:48: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,40,128,200,0,0,0,0,2,0,0,0,16,137,65,0,0,217,65,255,255,0,0,179,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:22:53: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,41,128,200,0,0,0,0,2,0,0,0,196,193,64,0,0,228,65,255,255,0,0,178,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:23:59: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,42,128,200,0,0,0,0,2,0,0,0,248,7,64,0,0,223,65,255,255,0,0,177,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:25:04: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,43,128,200,0,0,0,0,2,0,0,0,208,103,64,0,0,216,65,255,255,0,0,176,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:26:09: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,44,128,200,0,0,0,0,2,0,0,128,194,146,65,0,0,224,65,255,255,0,0,175,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:27:15: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,45,128,200,0,0,0,0,2,0,0,0,232,223,63,0,0,224,65,255,255,0,0,174,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:28:20: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,52,128,200,0,0,0,0,2,0,0,128,148,187,65,0,0,221,65,255,255,0,0,180,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:29:25: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,74,128,200,0,0,0,0,2,0,0,0,226,160,64,0,0,210,65,255,255,0,0,180,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:30:31: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,82,128,200,0,0,0,0,2,0,0,0,149,126,65,0,0,208,65,255,255,0,0,180,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:31:36: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,84,128,200,0,0,0,0,2,0,0,128,203,136,65,0,0,212,65,255,255,0,0,179,0],"type":"Buffer"}' from endpoint 1 with groupID 0
warn  2021-05-27 19:32:15: Device '0x84fd27fffe80e0d5' left the network
info  2021-05-27 19:33:22: Device '0x84fd27fffe80e0d5' joined
info  2021-05-27 19:33:22: Starting interview of '0x84fd27fffe80e0d5'
debug 2021-05-27 19:33:22: Device '0x84fd27fffe80e0d5' announced itself
debug 2021-05-27 19:33:22: Received Zigbee message from '0x84fd27fffe80e0d5', type 'raw', cluster '64528', data '{"data":[13,99,18,86,80,80,24,0,0,0,4,255,0],"type":"Buffer"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:33:23: Received Zigbee message from '0x84fd27fffe80e0d5', type 'readResponse', cluster 'genBasic', data '{"modelId":"E-Wand"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:33:24: Received Zigbee message from '0x84fd27fffe80e0d5', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"Current Products Corp"}' from endpoint 1 with groupID 0
debug 2021-05-27 19:33:25: Received Zigbee message from '0x84fd27fffe80e0d5', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
debug 2021-05-27 19:33:26: Received Zigbee message from '0x84fd27fffe80e0d5', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2021-05-27 19:33:29: Received Zigbee message from '0x84fd27fffe80e0d5', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2021-05-27 19:33:30: Received Zigbee message from '0x84fd27fffe80e0d5', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
[... cut ...]
Where I got stuck was the only zigbee messages I'm seeing from the device look like the below. I'm not quite sure what to do with those raw messages though. Perhaps I'm missing something painfully obvious? If not I can reach out to the company to see if they are willing to provide any documentation on how to interact with the Zigbee radio on the device. I appreciate any pointers in the right direction :)  

```
2021-05-30 21:53:16: Received Zigbee message from 'Testing E-Wand', type 'raw', cluster '64528', data '{"data":[13,99,18,118,128,200,213,224,128,254,2,0,0,0,0,16,63,0,0,199,65,0,255,0,0,171,0],"type":"Buffer"}' from endpoint 1 with groupID 0
```

Leah-Anderson, 2021-06-01:
I just wanted to provide another update. I was able to get something that reasonably works. At this point the position slider works and responds correctly as well as an open/close command from Homeassistant. Additionally the battery percentage seems reasonable. I am still seeing this in the log though:
```
Zigbee2MQTT:debug 2021-05-31 23:25:57: No converter available for 'CP180335E-01' with cluster '64528' and type 'raw' and data '{"data":[13,99,18,103,128,199,213,224,128,254,2,0,0,0,0,128,63,0,0,210,65,0,255,134,1,179,0],"type":"Buffer"}'
```

I'll keep investigating but I'm not quite sure what that is. Additionally I had to add configureKey: 1 as I was seeing this in the logs:
```
Failed to configure 'Testing E-Wand', attempt 1 (TypeError: Cannot read property 'configureKey' of undefined
```

I'm not sure what that does but once I added that meta key it was able to successfully configure itself. The below is what I have so far. I'd appreciate any input that could be provided. It'll be a few days before I can pick this up again but will update if I find out more and ideally at some point submit a PR to get this added in. 

```
// Manufacturer: Current Products Corp
// Model: CP180335E-01.js (E-Wand)

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['E-Wand'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'CP180335E-01', // Vendor model number, look on the device for a model number
    vendor: 'Current Products Corp', // Vendor of the device (only used for documentation and startup logging)
    description: 'Gen. 2 Hybrid E-Wand', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.cover_position_tilt, fz.battery],
    toZigbee: [tz.cover_state, tz.cover_position_tilt],
    meta: {
        configureKey: 1,
    },
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'closuresWindowCovering']);
        await reporting.batteryPercentageRemaining(endpoint);
        await reporting.currentPositionTiltPercentage(endpoint);
    },
    exposes: [e.battery(), e.cover_position()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;

```

Leah-Anderson, 2021-07-02:
Initial support is there. Will continue working on fine tuning some of the functionality.

closed 2021-07-02 by Leah-Anderson

RMCob, 2021-08-19:
I am using zigbee2mqtt and z2m on Homebridge to expose this to Homekit. It works, sort of. Open (or tilt 100) does indeed open the blinds fully. Close (or tilt 0) closes the blinds in the 'Up' position. There is no way to specify closing the blinds in the 'Down' position. I reached out to the vendor and got this reply:

      The following command is used to send a signed percentage to the E-Wand allowing the full range of movement:
      Destination Endpoint: 1
      MFG Code: 0x1263
      Cluster ID: 0xFC10
      Command ID: 0x26
      Payload (1-byte signed): -100 to 100 for the percent of movement.

      The command must have the Mfg Specific flag set with the given Mfg Code. 
      Most of the issues we've seen with users adding custom capabilities is because 
      this flag wasn't added to the command.

      Charles Howard
      Customer Service

      Current Products Corp.
      1995 Hollywood Ave.
      Pensacola, FL  32505
      850.435.4994   Main Office
      850.332.7930   Desk
      850.332.7930   Cell

Is there any way this functionality could be implemented so that it is possible to control
the full range of motion.

Thanks in advance for your consideration.....
.....Rob Coben

RMCob, 2021-08-19:
More info: The Cluster ID specified by the manufacturer '0xFC10' is the '64528' cluster that keeps showing up in your logs. Not sure what that means but I wanted to point it out. :-)