Koenkk/zigbee2mqtt #7651: Network of Xiaomi devices gets slow and dies out

M1ndHunteR, 2021-05-31:
Hi Koenkk and other community members.

Please allow me to start this topic with a brief explanations of my problem and I'll try supplying whatever will be missing upon your requests or suggestions.

So, I'm running latest Zigbee2MQTT  using Electrolama zig-a-zig-ah! (zzh!) CC2652R with the latest coordinator firmware(20210120).
I am using an extension cable and I have tried tens and tens of different locations and orientations of the stick and antenna.

I have a network of about 50 Xiaomi devices, about third of which are constantly powered wall switches, which technically, are supposed to act like routers.

Now, the network works fine - all devices talk, very fast comms, life is good.
After approx 10-15 hours, many of battery powered devices stop communicate. Some wired wall switches become very slow to respond. If I do nothing, after a while Zigbee2MQTT log fills up with messages similar to one below:

`zigbee2mqtt:info  2021-05-31 08:25:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'light_hallway_front' failed: 'Error: Command 0x001788010279726f/11 genOnOff.on({},... blah blah blah`

And then everything just dies out. The only way to get things back to normal at this stage is to POWER CYCLE all the wall switches (circuit breaker is handy). Just restarting Zigbee2MQTT doesn't help at this stage.

However, if I restart Zigbee2MQTT  service just when the first signs of slowing down appear, it seems to rectify the upcoming disaster for some time.

I am ready to perform whatever experiments you can suggest, I went thru a lot of messages in this Issues board, and, well, I have no idea how to solve that or make sure anyone else can reproduce the issue because of the complexity of my setting.

Please help,
Gene.

M1ndHunteR, 2021-06-15:
Anyone? Anything?

Robertops, 2021-07-06:
Same problem here. Total of 38 Devices, 27 of then are Xiaomi wall switches (QBKG12LM and QBKG11LM) and 4 are gledopto rgb controllers (GL-C-008-1ID). A lot of routers.

I can get about 2 days before things start to get slow. In the next days they will stop responding and almost every device will time out in 3 or 4 days.

I tried updating all the switches with the new OTA fature but just got the same behavior.

Using zzh, but had exactly same problem with CC2531.

closed 2021-08-17 by github-actions[bot]

jinnerbichler, 2021-09-04:
I am experiencing the same behavior. I have Revision A of zzh!.

M1ndHunteR, 2021-10-16:
> 
> 
> @M1ndHunteR any solutions to this issue?

I think one of Xiaomi wall switches (routers), despite them all having same fw version, causing the problem. I have not yet figured which one. But I keep adding a couple more devices to the network every few days, and expecting it to fail eventually. So far so good, but there will definitely be a failure at some point and I hope to figure out what wall switch to blame.

dmitry-korolev, 2023-01-08:
I can confirm this issue. I ended up with two separate zigbee networks — the "normal" one connected directly to my HA instance, and another one based on xiaomi gateway