Koenkk/zigbee2mqtt #7655: Ikea Tradfri E2044 - not supported (?)

njordan77, 2021-05-31:
Hi,
i today bought an ikea remote control (looks like the E2049) in the list of supported devices. However inside close to the batteries there is written E2044. I tried many times to pair, but not working yet. Could it be the case that this item is not supported at all?

```
May 31 22:14:44 SmartPi npm[506]: Zigbee2MQTT:info  2021-05-31 22:14:44: Starting interview of '0x680aexxxxxx79e7bf'
May 31 22:14:44 SmartPi npm[506]: Zigbee2MQTT:info  2021-05-31 22:14:44: MQTT publish: topic '/z2m/bridge/event', payload '{"data":{"friendly_name":"0x68xxxxxfffe79e7bf","ieee_address":"0x680ae2fffe79e7bf","status":"started"},"type":"device_interview"}'
May 31 22:14:44 SmartPi npm[506]: Zigbee2MQTT:info  2021-05-31 22:14:44: MQTT publish: topic '/z2m/bridge/log', payload '{"message":{"friendly_name":"0x680ae2fffe79e7bf"},"type":"device_connected"}'
May 31 22:14:44 SmartPi npm[506]: Zigbee2MQTT:info  2021-05-31 22:14:44: MQTT publish: topic '/z2m/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x68xxxxxffe79e7bf"},"type":"pairing"}'
May 31 22:17:02 SmartPi npm[506]: Zigbee2MQTT:error 2021-05-31 22:17:02: Failed to interview '0x68xxxxx79e7bf', device has not successfully been paired
```

sjorge, 2021-06-01:
Can you grab the database.db entry for that device?

njordan77, 2021-06-04:
Sure if i know how to do that/what exactly you need to have....
IS there any short summary what i need to provide.

My general question is, will the device be in the DB already if i received a failed to interview /unsuccessful....
Kind Regards
Norbert

sjorge, 2021-06-04:
Yes, failed interviews also appear in the database.

You should be able to grep on the device_id and paste the line here.

njordan77, 2021-06-04:
Here is the entry from the DB.
`{"id":25,"type":"EndDevice","ieeeAddr":"0x680ae2fffe79e7bf","nwkAddr":24603,"manufId":4476,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"in$`

again the log of an unsuccessful try to pair. 
```
Jun 04 22:38:11 SmartPi npm[19338]: Zigbee2MQTT:info  2021-06-04 22:38:11: Starting interview of '0x680ae2fffe79e7bf'
Jun 04 22:38:11 SmartPi npm[19338]: Zigbee2MQTT:info  2021-06-04 22:38:11: MQTT publish: topic '/z2m/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe79e7bf","ieee_address":"0x680ae2fffe79e7bf","status":"started"},"type":"device_interview"}'
Jun 04 22:38:11 SmartPi npm[19338]: Zigbee2MQTT:info  2021-06-04 22:38:11: MQTT publish: topic '/z2m/bridge/log', payload '{"message":{"friendly_name":"0x680ae2fffe79e7bf"},"type":"device_connected"}'
Jun 04 22:38:11 SmartPi npm[19338]: Zigbee2MQTT:info  2021-06-04 22:38:11: MQTT publish: topic '/z2m/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x680ae2fffe79e7bf"},"type":"pairing"}'
Jun 04 22:38:26 SmartPi npm[19338]: Zigbee2MQTT:info  2021-06-04 22:38:26: MQTT publish: topic '/z2m/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe79e7bf","ieee_address":"0x680ae2fffe79e7bf"},"type":"device_announce"}'
Jun 04 22:38:26 SmartPi npm[19338]: Zigbee2MQTT:info  2021-06-04 22:38:26: MQTT publish: topic '/z2m/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x680ae2fffe79e7bf"},"type":"device_announced"}'
Jun 04 22:38:45 SmartPi npm[19338]: Zigbee2MQTT:error 2021-06-04 22:38:45: Failed to interview '0x680ae2fffe79e7bf', device has not successfully been paired
Jun 04 22:38:45 SmartPi npm[19338]: Zigbee2MQTT:info  2021-06-04 22:38:45: MQTT publish: topic '/z2m/bridge/event', payload '{"data":{"friendly_name":"0x680ae2fffe79e7bf","ieee_address":"0x680ae2fffe79e7bf","status":"failed"},"type":"device_interview"}'
Jun 04 22:38:45 SmartPi npm[19338]: Zigbee2MQTT:info  2021-06-04 22:38:45: MQTT publish: topic '/z2m/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x680ae2fffe79e7bf"},"type":"pairing"}'
```

From outside it looks like this:
`https://www.zigbee2mqtt.io/devices/W2049.html`

but inside there stands W2044.

MattWestb, 2021-06-05:
The model type is E2001/E2002.
The W2044/W2049 is the production year and week in short format as on all electronics.
My have W2045 and is one E2002 (with).

njordan77, 2021-06-05:
Ah thanks for the clarification. On the battery cover its written E2002.
But should it be already supported, as the device does not have any successful pairing after multiple tries.

I've the zzh! stick plus lattest release installed.

MattWestb, 2021-06-05:
IKEA device is very Zigbee standard and shall pairing with all Zigbee compatible system without problems and the rest is up to the host system.
Its in the Z2M docks you have linked so i supposing its supported.

njordan77, 2021-06-05:
OK, but does this mean it should already be supported?

njordan77, 2021-06-06:
I now removed manually the DB entry line, and restarted the pairing....now it worked and button reaction can be seen in the logs. BUT what is strange that with every button press i see two lines.....which annoys openhab when two times the same message comes along.....is this "normal" or something suspicious?
```

Jun 06 23:21:13 SmartPi npm[6514]: Zigbee2MQTT:info  2021-06-06 23:21:13: MQTT publish: topic '/z2m/0x680ae2fffe79e7bf', payload '{"action":"arrow_right_click","battery":100,"linkquality":69,"update":{"state":"idle"},"update_available":false}'
Jun 06 23:21:13 SmartPi npm[6514]: Zigbee2MQTT:info  2021-06-06 23:21:13: MQTT publish: topic '/z2m/0x680ae2fffe79e7bf', payload '{"action":"arrow_right_click","battery":100,"linkquality":72,"update":{"state":"idle"},"update_available":false}'
```

jack9603301, 2021-06-15:
This should be normal, because in the two pieces of data, the visible link quality indication changes, and it will report other states due to the existence of state cache

MattWestb, 2021-06-15:
The E2001 / E2002 is sending reports to all bonded device and if some have configured more bindings the devices is reporting to them all.

I have getting one spamming 8 transactions on one bottom pressed and sniffing how they is behaving.

njordan77, 2021-06-15:
ah, that makes sense....maybe its paired twice and i need to reset the device in order to re-pair it a single time.... will try this. Cheers

MattWestb, 2021-06-15:
If Z2M is setting up more bindings and have attribute reporting its reporting to all attribute to all bonded devices.
So if binding on/off and light level you is getting 2 reports of the attribute on for each binding.

closed 2021-07-24 by github-actions[bot]

gdp2000, 2021-12-16:
I have had a similar problem with a E2002 Ikea switch. The interview failed on the first pairing of the device. Subsequent pairings failed as well. Even after force deleting the device from a Zigbee2MQTT installation from wihtin Home Assistant. I had to manually remove the device from MQTTs "database.db". Afterwards the device paired succesfully.