Koenkk/zigbee2mqtt #7658: Moving my Z2M HA Addon from old to a new NUC - wont start ...

TheStigh, 2021-06-01:
Hi,

I've started migrating my HA from an old to a new NUC. Using the Addon version of Z2M. After restoring the backup on the new NUC, I can't get the Z2M to start. I did a clean Z2M install, after this I copied the entire /data folder from the old `/data/zigbee2mqtt` folder to the new proposed folder `/config/zigbee2mqtt` but Z2M just start up as default. It doesn't bother about the existing `configuration.yaml` or the `coordinator_backup.json`.

Using 1352-P2 adapter.

How can I resolve this?

Koenkk, 2021-06-01:
Could you provide the herdsman debug log when starting after the recovery?

TheStigh, 2021-06-01:
The 'open line' might be some data/log missing, hard to capture the entire log during startup :)

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] socat.sh: executing... 
[15:45:05] INFO: Socat not enabled, marking service as down
[cont-init.d] socat.sh: exited 0.
[cont-init.d] zigbee2mqtt.sh: executing... 
[15:45:06] INFO: MQTT available, fetching server detail ...
[15:45:06] INFO: Previous config file found, checking backup
[15:45:06] INFO: Creating backup config in '/config/zigbee2mqtt/.configuration.yaml.bk'
[15:45:06] WARNING: No devices.js file found in data path, starting with default devices.js
[15:45:06] INFO: Adjusting Zigbee2mqtt core yaml config with add-on quirks ...
[cont-init.d] zigbee2mqtt.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[15:45:06] INFO: Zigbee Herdsman debug logging enabled
[15:45:06] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.18.3 start /app
> node index.js
Zigbee2MQTT:debug 01-06-2021 15:45:07: Loaded state from file /config/zigbee2mqtt/state.json
Zigbee2MQTT:info  01-06-2021 15:45:07: Logging to console and directory: '/config/zigbee2mqtt/log/2021-06-01.15-45-06' filename: log.txt
Zigbee2MQTT:debug 01-06-2021 15:45:07: Removing old log directory '/config/zigbee2mqtt/log/2021-06-01.15-33-09'
Zigbee2MQTT:info  01-06-2021 15:45:07: Starting Zigbee2MQTT version 1.18.3 (commit #1.18.3)
Zigbee2MQTT:info  01-06-2021 15:45:07: Starting zigbee-herdsman (0.13.92)
Zigbee2MQTT:debug 01-06-2021 15:45:07: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/config/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","databasePath":"/config/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
2021-06-01T13:45:07.227Z zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT'
2021-06-01T13:45:07.228Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[9,12,13,6,1,13,2,13,8,15,13,5,0,11,4,3],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}'
2021-06-01T13:45:07.229Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2021-06-01T13:45:07.237Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2021-06-01T13:45:07.692Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2021-06-01T13:45:07.693Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]

2021-06-01T13:45:08.712Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}}
2021-06-01T13:45:08.713Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74]
2021-06-01T13:45:08.729Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62]
2021-06-01T13:45:08.729Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62]
2021-06-01T13:45:08.729Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62
2021-06-01T13:45:08.730Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}}
2021-06-01T13:45:08.730Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-01T13:45:08.731Z zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid
2021-06-01T13:45:08.732Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":96,"offset":0,"len":1,"value":{"type":"Buffer","data":[85]}}
2021-06-01T13:45:08.732Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,96,0,0,74]
2021-06-01T13:45:08.749Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62]
2021-06-01T13:45:08.749Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62]
2021-06-01T13:45:08.749Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62
2021-06-01T13:45:08.749Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}}
2021-06-01T13:45:08.750Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-01T13:45:08.750Z zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid
2021-06-01T13:45:08.751Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":132,"len":4,"offset":0,"value":{"type":"Buffer","data":[0,8,0,0]}}
2021-06-01T13:45:08.751Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,132,0,0,174]
2021-06-01T13:45:08.758Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,6,97,8,0,4,0,8,0,0,99]
2021-06-01T13:45:08.759Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,6,97,8,0,4,0,8,0,0,99]
[... cut ...]
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.3 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.18.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-06-01T13_46_08_895Z-debug.log
[15:46:09] INFO: Zigbee Herdsman debug logging enabled
[15:46:09] INFO: Handing over control to Zigbee2mqtt Core ...
```

TheStigh, 2021-06-01:
All good! Got it running again...!

Stopped Z2M
Disconnected ext antenna from 1352-P2
Moved backup back to new folder
Started Z2M
... failed for some time, after a few minutes it suddenly got 'contanct'
Stopped Z2M
Re-Connected antenna

Now devices are coming back to life!

closed 2021-06-01 by TheStigh