Koenkk/zigbee2mqtt #7664: HomeAssistant addon 1.19.0 Configuration is not consistent

radio-m, 2021-06-02:
After update z2m from 1.18 to 1.19 get some errors
`Zigbee2MQTT:info  2021-06-02 09:01:27: Starting zigbee-herdsman (0.13.111)
Zigbee2MQTT:error 2021-06-02 09:01:29: Configuration is not consistent with adapter state/backup!
Zigbee2MQTT:error 2021-06-02 09:01:29: - PAN ID: configured=6754, adapter=6754
Zigbee2MQTT:error 2021-06-02 09:01:29: - Extended PAN ID: configured=00124b0009d7fcae, adapter=00124b0009d7fcae
Zigbee2MQTT:error 2021-06-02 09:01:29: - Network Key: configured=01030507090b0d0f00020406080a0c0d, adapter=01030507090b0d0f00020406080a0c0d
Zigbee2MQTT:error 2021-06-02 09:01:29: - Channel List: configured=11, adapter=12,13,14,15,16,17,18,19,20,21,22,23,24,25,26
Zigbee2MQTT:error 2021-06-02 09:01:29: Please update configuration to prevent further issues.
Zigbee2MQTT:error 2021-06-02 09:01:29: If you wish to re-commission your network, please remove coordinator backup at /share/zigbee2mqtt/coordinator_backup.json.
Zigbee2MQTT:error 2021-06-02 09:01:29: Re-commissioning your network will require re-pairing of all devices!
Zigbee2MQTT:error 2021-06-02 09:01:29: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-06-02 09:01:29: Failed to start zigbee
Zigbee2MQTT:error 2021-06-02 09:01:29: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-06-02 09:01:29: Exiting...
Zigbee2MQTT:error 2021-06-02 09:01:29: Error: startup failed - configuration-adapter mismatch - see logs above for more information
    at ZnpAdapterManager.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/manager.js:201:35)
    at Generator.next (<anonymous>)
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/manager.js:24:58)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.19.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.19.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-06-02T06_01_29_607Z-debug.log`
RaspberryPi 4B, raspbian, HomeAssistant supervised in docker, adapter based on cc2652

Gooman-rus, 2021-06-02:
I have the same error.

DubhAd, 2021-06-02:
Looks related to [this report](https://github.com/Koenkk/zigbee-herdsman/pull/303#issuecomment-850989092)

tribalRu, 2021-06-02:
Same errors (cc2538)
```
Zigbee2MQTT:info  2021-06-02 10:36:57: Loaded  MyExampleExtension1614785288109
Zigbee2MQTT:info  2021-06-02 10:36:57: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-06-                           02.10-36-52' filename: log.txt
Zigbee2MQTT:info  2021-06-02 10:36:57: Starting Zigbee2MQTT version 1.19.0 (commit #89cb4f95)
Zigbee2MQTT:info  2021-06-02 10:36:57: Starting zigbee-herdsman (0.13.111)
Zigbee2MQTT:error 2021-06-02 10:37:01: Configuration is not consistent with adapter state/backup!
Zigbee2MQTT:error 2021-06-02 10:37:01: - PAN ID: configured=6754, adapter=6755
Zigbee2MQTT:error 2021-06-02 10:37:01: - Extended PAN ID: configured=00124b00062d5###, adapter=00124b00062d5                           ###
Zigbee2MQTT:error 2021-06-02 10:37:01: - Network Key: configured=c03b44e5860547710402b795ae1f6###, adapter=c                           03b44e5860547710402b795ae1f6###
Zigbee2MQTT:error 2021-06-02 10:37:01: - Channel List: configured=11, adapter=11
Zigbee2MQTT:error 2021-06-02 10:37:01: Please update configuration to prevent further issues.
Zigbee2MQTT:error 2021-06-02 10:37:01: If you wish to re-commission your network, please remove coordinator                            backup at /opt/zigbee2mqtt/data/coordinator_backup.json.
Zigbee2MQTT:error 2021-06-02 10:37:01: Re-commissioning your network will require re-pairing of all devices!
Zigbee2MQTT:error 2021-06-02 10:37:01: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-06-02 10:37:01: Failed to start zigbee
Zigbee2MQTT:error 2021-06-02 10:37:01: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqt                           t-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-06-02 10:37:01: Exiting...
Zigbee2MQTT:error 2021-06-02 10:37:01: Error: startup failed - configuration-adapter mismatch - see logs abo                           ve for more information
```

MudiStar, 2021-06-02:
I have the same problem ( CC2538 + CC2592 )
Rolled back to 1.18.3-1

Gooman-rus, 2021-06-02:
In my case error was in defferent PAN ID between zigbee2mqtt config and CC2538 apater:
`Zigbee2MQTT:error 2021-06-02 10:07:24: - PAN ID: configured=6754, adapter=6755`

What helped me:

1. On version 1.18.1-1 I have changed pan_id in z2m config to the 6755.
2. All of my devices lost connection with the network.
3. I have updated z2m to the last version (1.19), started it  and received an error because on the difference in pan_id. I have changed pan_id in z2m config back to the 6754. And this solved the error (in my case). **But I still have to repair all of my devices**.

MudiStar, 2021-06-02:
Thanks for the information
But repairing off all my 20 devices (some of them are hard to reach) will be the last option for me

tribalRu, 2021-06-02:
Change pan_id and all works fines

MudiStar, 2021-06-02:
I can confirm
Changing to pan_id: 6755 works without repairing

Wummeke, 2021-06-02:
I can confirm too that updating the configuration.yaml with the following line fixed the issue for me:

```YAML
advanced:
  pan_id: 6755
```
But I don't really understand why this solved it. Under 1.18.1 (and before) the pan_id was 6754 by default.

Koenkk, 2021-06-02:
With https://github.com/Koenkk/zigbee-herdsman/pull/303 there was a lot of refactoring done on the starting procedure. It seems that in some cases CC2538 sticks have been initialised with a different pan_id then specified (I guess the CC2538 firmware allows starting with another panID while the CC2652/CC1352/CC2531 don't). Changing the pan_id in your configuration.yaml to the adapter one should fix it without needing to repair (e.g. `PAN ID: configured=6754, adapter=6755`)

Palkin81, 2021-06-08:
There are several Livolo devices on my network. A configuration different from the standard one is required to connect them. After update 0.19.1-x, I can't start this configuration. Channel 26 must be installed. PAN ID=6756. What should I do?

marcelbrueckner, 2021-06-14:
I had the same issue after just upgrading from Home Assistant Add-On `1.18.3-1` to `1.19.1-1`. Restoring `1.18.3-1` and everything works again. 

For me, it's the `Extended PAN ID` configuration that mismatches with my [cod.m ZigBee CC2538 Raspberry Pi Module](https://shop.codm.de/automation/zigbee/29/zigbee-cc2538-raspberry-pi-modul)'s value. Although after initial setup, I've neither changed the (Extended) PAN ID nor the ZigBee hardware (which is connected to an other Raspberry Pi and made available to z2m via ser2net if that matters).

closed 2021-07-24 by github-actions[bot]

Gadzhi13, 2022-07-11:
@radio-m did you manage to find the root cause of the issue?

I am having the same on my end, with all the data being correct but the herdsman still complaining. And I did not even have any updates, all of this happened just after a reboot of the host system.