Koenkk/zigbee2mqtt #7665: Upgrading to 1.19 made all devices unjoin

hmmbob, 2021-06-02:
## What happened
I upgraded to 1.19.0 and had 0 devices joined. Rollback to 1.18.3 did not help - 0 devices joined. I joined all my devices manually again (wanted to do that anyway due to changed locations and some sticky bindings of battery powered sensors), but this was unexpected.

## What did you expect to happen
Smooth upgrade, keeping my devices

## How to reproduce it (minimal and precise)
I just upgraded from 1.18.3 to 1.19.0 by pulling the newest image in Docker.

## Debug info
Zigbee2MQTT version:
Adapter hardware: ~~CC2531~~, CC2530, ~~CC26X2R1, CC1352P-2, Conbee II~~
Adapter firmware version: 20190109

Squixx, 2021-06-02:
Same here, no further config changes

Koenkk, 2021-06-02:
What adapter/firmware are you using and are you able to reproduce this? If yes please share the herdsman debug log of this.

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

hmmbob, 2021-06-02:
I was using a CC2531 with 201901.. - was never updated before. But I've upgraded the firmware in the afternoon, so I don't think I can reproduce reliably....

madman2143, 2021-06-02:
I have the same problem, I normally have about 8 devices in Zigbee2MQTT, after the update there was only one device left. I also use the CC2531 dongle.

etzisim, 2021-06-02:
I upgraded to 1.19 and also all devices are lost in the Webinterface and the dont respond anymore. But i still have them in the config file

virby123, 2021-06-03:
All devices are lost after upgrade. Recovery from the backup did not solve the problem. Devices cannot be added to either a new or an older version. Adapter cc2531.

JDFS404, 2021-06-03:
Same for me, I had about 30 devices and all of them are gone... They still show up in `configuration.yaml` - I updated this night to the latest version and cannot rejoin with any of them.

olylak, 2021-06-03:
Same issue

Koenkk, 2021-06-03:
Please try to reproduce this and provide the log as asked in https://github.com/Koenkk/zigbee2mqtt/issues/7665#issuecomment-853258232 . Otherwise I cannot help.

virby123, 2021-06-03:
My previous message have all error from log with debug enabled.

JDFS404, 2021-06-03:
> Please try to reproduce this and provide the log as asked in [#7665 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/7665#issuecomment-853258232) . Otherwise I cannot help.

OK running a CC2531 here, running Zigbee2MQTT in docker with `latest-dev` branch because I got a lot of errors with the `latest` branch. Dev seemed to work fine. Added `zigbee-herdsman*` to environment variable with Key `DEBUG`.

Log:
```
debug 2021-06-03 10:02:47: Loaded state from file /app/data/state.json
info  2021-06-03 10:02:47: Logging to console and directory: '/app/data/log/2021-06-03.10-02-46' filename: log.txt
debug 2021-06-03 10:02:47: Removing old log directory '/app/data/log/2021-06-03.04-38-55'
info  2021-06-03 10:02:47: Starting Zigbee2MQTT version 1.19.0-dev (commit #c508155)
info  2021-06-03 10:02:47: Starting zigbee-herdsman (0.13.111)
debug 2021-06-03 10:02:47: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
info  2021-06-03 10:02:50: zigbee-herdsman started
info  2021-06-03 10:02:50: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20201127,"transportrev":2},"type":"zStack12"}'
debug 2021-06-03 10:02:50: Zigbee network parameters: {"channel":11,"extendedPanID":"0x00124b001cce0bc1","panID":6754}
info  2021-06-03 10:02:50: Currently 3 devices are joined:
info  2021-06-03 10:02:50: motion_hallway (0x0017880109170175): 9290012607 - Philips Hue motion sensor (EndDevice)
info  2021-06-03 10:02:50: 0x680ae2fffe3e0536 (0x680ae2fffe3e0536): Not supported (Unknown)
info  2021-06-03 10:02:50: bathroom (0x00178801096cb1d9): Not supported (Unknown)
warn  2021-06-03 10:02:50: `permit_join` set to  `true` in configuration.yaml.
warn  2021-06-03 10:02:50: Allowing new devices to join.
warn  2021-06-03 10:02:50: Set `permit_join` to `false` once you joined all devices.
info  2021-06-03 10:02:50: Zigbee: allowing new devices to join.
info  2021-06-03 10:02:50: Started frontend on port 0.0.0.0:8080
info  2021-06-03 10:02:50: Connecting to MQTT server at mqtt://192.168.1.101
debug 2021-06-03 10:02:50: Using MQTT anonymous login
info  2021-06-03 10:02:50: Connected to MQTT server
info  2021-06-03 10:02:50: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2021-06-03 10:02:50: MQTT publish: topic 'zigbee2mqtt/motion_hallway', payload '{"battery":100,"illuminance":16267,"illuminance_lux":42,"led_indication":null,"linkquality":null,"motion_sensitivity":"medium","occupancy":false,"occupancy_timeout":0,"temperature":21.52,"update":{"state":"available"},"update_available":true}'
info  2021-06-03 10:02:50: MQTT publish: topic 'zigbee2mqtt/0x680ae2fffe3e0536', payload '{"battery":74,"brightness":255,"rate":195,"update":{"state":"idle"},"update_available":false}'
info  2021-06-03 10:02:50: MQTT publish: topic 'zigbee2mqtt/bathroom', payload '{"brightness":3,"color_mode":"color_temp","color_temp":454,"state":"ON","update":{"state":"available"},"update_available":true}'
debug 2021-06-03 10:02:50: Received MQTT message on 'zigbee2mqtt/bridge/info' with data '{"commit":"c508155","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":0,"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"disable","legacy_api":true,"log_directory":"/app/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_symlink_current":false,"log_syslog":{},"pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{"0x00158d0006c3c7e3":{"friendly_name":"door_living_room"},"0x00158d0006d6c381":{"friendly_name":"humidity_bathroom"},"0x00158d0006e49bd1":{"friendly_name":"humidity_living_room"},"0x00178801020ebff1":{"friendly_name":"dimmer"},"0x00178801023d0fff":{"friendly_name":"storage"},"0x0017880108cb0555":{"friendly_name":"bedroom_2"},"0x001788010905d487":{"friendly_name":"tv_filament"},"0x001788010916e6e8":{"friendly_name":"motion_storage"},"0x0017880109170175":{"friendly_name":"motion_hallway"},"0x00178801091714a4":{"friendly_name":"motion_bathroom"},"0x00178801092a31cd":{"friendly_name":"living_room_1"},"0x00178801092a333d":{"friendly_name":"bedroom_1"},"0x00178801092a4fa3":{"friendly_name":"living_room_2"},"0x00178801092a501e":{"friendly_name":"dining_table_1"},"0x00178801092a5024":{"friendly_name":"dining_table_2"},"0x00178801092a50a1":{"friendly_name":"kitchen_2"},"0x00178801092adb7b":{"friendly_name":"desk"},"0x00178801092adfba":{"friendly_name":"kitchen_1"},"0x00178801092adfdf":{"friendly_name":"living_room_3"},"0x00178801092e1016":{"friendly_name":"bedroom_3"},"0x00178801096cb1d9":{"friendly_name":"bathroom"},"0x00178801096d06c7":{"friendly_name":"hallway"},"0x001788010b0032c0":{"friendly_name":"s [... cut ...]
debug 2021-06-03 10:02:50: Received MQTT message on 'homeassistant/binary_sensor/0x0017880109170175/update_available/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x0017880109170175"],"manufacturer":"Philips","model":"Hue motion sensor (9290012607)","name":"motion_hallway","sw_version":"Zigbee2MQTT 1.19.0-dev"},"json_attributes_topic":"zigbee2mqtt/motion_hallway","name":"motion_hallway_update_available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/motion_hallway","unique_id":"0x0017880109170175_update_available_zigbee2mqtt","value_template":"{{ value_json.update_available}}"}'
```

olylak, 2021-06-03:
[zigbee2mqtt_2021-06-03.11-04-05.log](https://github.com/Koenkk/zigbee2mqtt/files/6589986/zigbee2mqtt_2021-06-03.11-04-05.log)

hmmbob, 2021-06-03:
Are these all CC253x devices you are using?

virby123, 2021-06-03:
> Are these all CC253x devices you are using?

Yes

etzisim, 2021-06-03:
Yes I use a cc2531

JDFS404, 2021-06-03:
> Are these all CC253x devices you are using?

yes

Koenkk, 2021-06-03:
I was unclear about this in https://github.com/Koenkk/zigbee2mqtt/issues/7665#issuecomment-853258232, but I need this log when upgrading from 1.18.3 to 1.19.0 and losing your devices. Please post your logs on pastebin.com and link them here.
Also what firmware are you using on your adapter?

So try reproducing it by:
- Going to 1.18.3
- Enable herdsman debug log
- Upgrade to 1.19.0 -> now your devices should be lost.
- Post herdsman debug log here

JDFS404, 2021-06-03:
@Koenkk I downgraded back to 1.18.3 but all my devices are still lost. It looks like something happened with the upgrade that triggered the devices to be lost. Also I cannot rejoin them. But I will try to upgrade again, maybe that'll help!

virby123, 2021-06-03:
After several restores from the backup and updates to the latest version, the addition of devices on the new version worked. All devices added to the new, the names pulled up from the old base. Everything works.

JDFS404, 2021-06-03:
> After several restores from the backup and updates to the latest version, the addition of devices on the new version worked. All devices rejoined, the names pulled up from the old db. Everything works.

What did you do differently? I tried restoring from my backup from 2 days ago, but still all devices are unjoined (and still in `configuration.yaml`). It's driving me crazy, even tried factory resetting my Hue bulbs but they're not picked up. Since I changed all my physical switches for Hue Wall Modules, I cannot toggle the lights.

JDFS404, 2021-06-03:
Here is my log: 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/6591685/log.txt)

Koenkk, 2021-06-04:
@JDFS404 I need the log when upgrading from 1.18.3 to 1.19.0 like described here: https://github.com/Koenkk/zigbee2mqtt/issues/7665#issuecomment-853689437

JDFS404, 2021-06-04:
Well, I had some troubles again and lost ALL devices - especially the Hue Wall Modules are a pain in the ass to reset, so I reversed back to 1.18.3 and just set everything back. Even flashed CC2531 again and got another one from a friend of mine. So I will just stick with 1.18.3 now because I already spent a day resetting everything...

Maybe when I'll switch to ZZH I will update again and try to provide you with a debug log.

Ape, 2021-06-06:
I have the same issue, but all the devices worked for an hour or so with the new version before unjoining.

I cannot rejoin any of my devices. Downgrading does not help. I am using CC2531 with 20201127 firmware.

closed 2021-07-14 by github-actions[bot]