Koenkk/zigbee2mqtt #7666: Template variable warning: 'dict object' has no attribute 'power_outage_memory' when rendering '{{ value_json.power_outage_memory }}'

OmgImAlexis, 2021-06-02:
## What happened
My logs have this error.

## What did you expect to happen
Not to see the issue.

## How to reproduce it (minimal and precise)
Honestly not sure what's causing this but all google searches seem to lead back to this repo and other issues that look similar, most of which are closed by the OP with no answer to how/if they fixed it.

## Debug info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: CC2531 (I think)
Adapter firmware version: Not sure

m0wlheld, 2021-06-02:
Better report this Home Assistant issue at Home Assistant's issue tracker -> https://github.com/home-assistant/core/issues

Koenkk, 2021-06-02:
Already fixed in 1.18.3, see first note of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0 if you don't see the update.

closed 2021-06-02 by Koenkk