Koenkk/zigbee2mqtt #7671: SmartThings Arrival Sensor PGC410EU Battery Reporting

liamstears, 2021-06-02:
Although this sensor works perfectly battery reporting does not which makes it difficult to keep on top of battery changes as quite often they just die and end up settings off alarms without warning so having a functioning battery would be amazing

I have tried changing the device myself using the following code however I just received the errors below

'''
    {
        zigbeeModel: ['PGC410EU', 'PGC410'],
        model: 'STSS-PRES-001',
        vendor: 'SmartThings',
        description: 'Presence sensor',
        fromZigbee: [fz.PGC410EU_presence, fz.battery],
        exposes: [e.battery(), e.presence()],
        toZigbee: [],
		meta: {battery: {voltageToPercentage: '3V_2500'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genBinaryInput']);
            await reporting.batteryVoltage(endpoint);
            await reporting.presentValue(endpoint);
    },
'''

Errors:

'''
Zigbee2MQTT:info  2021-06-02 17:57:17: MQTT publish: topic 'zigbee2mqtt/PS1', payload '{"battery":null,"linkquality":93,"presence":true}'
2021-06-02T16:57:17.459Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,1,0,104,88,1,1,0,93,0,120,94,86,0,0,5,8,4,11,6,130,224,226,28,88]
2021-06-02T16:57:17.459Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,1,0,104,88,1,1,0,93,0,120,94,86,0,0,5,8,4,11,6,130,224,226,28,88]
2021-06-02T16:57:17.459Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,1,0,104,88,1,1,0,93,0,120,94,86,0,0,5,8,4,11,6,130,224,226,28] - 88
2021-06-02T16:57:17.459Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1,"srcaddr":22632,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":93,"securityuse":0,"timestamp":5660280,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[8,4,11,6,130]}}
2021-06-02T16:57:17.459Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":4,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":6,"statusCode":130},"Command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}},"address":22632,"endpoint":1,"linkquality":93,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2021-06-02T16:57:17.460Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-02T16:57:24.173Z zigbee-herdsman:adapter:zStack:adapter Response timeout (0xd052a81e8ca00005:22607,1)
2021-06-02T16:57:24.173Z zigbee-herdsman:controller:endpoint ConfigureReporting 0xd052a81e8ca00005/1 genPowerCfg([{"attribute":"batteryVoltage","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 22607 - 1 - 3 - 1 - 7 after 10000ms)
Zigbee2MQTT:error 2021-06-02 17:57:24: Failed to configure 'PS2', attempt 2 (Error: ConfigureReporting 0xd052a81e8ca00005/1 genPowerCfg([{"attribute":"batteryVoltage","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 22607 - 1 - 3 - 1 - 7 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
2021-06-02T16:57:25.513Z zigbee-herdsman:adapter:zStack:adapter Response timeout (0xd052a800fe520005:22632,1)
2021-06-02T16:57:25.513Z zigbee-herdsman:controller:endpoint ConfigureReporting 0xd052a800fe520005/1 genPowerCfg([{"attribute":"batteryVoltage","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 22632 - 1 - 4 - 1 - 7 after 10000ms)

Liam Stears, [02.06.21 17:58]
Zigbee2MQTT:error 2021-06-02 17:57:25: Failed to configure 'PS1', attempt 2 (Error: ConfigureReporting 0xd052a800fe520005/1 genPowerCfg([{"attribute":"batteryVoltage","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 22632 - 1 - 4 - 1 - 7 after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
'''

Please note I have 2 sensors (PS1 and PS2)

GSzabados, 2021-06-03:
@liamstears, just FYI, the presence sensors battery reporting are the same unreliable in SmartThings. They just die one day to another.

You are better if you set up a notification at some level and just replace at that time.

Or to keep track that it is still working, just set up notifications for arrival and leave, then you will always see when it is registered. You can even add the battery level to the notification so you will see what level at they will go dead.

Setting battery reporting with report every time when changes can cause rapid depletion.

liamstears, 2021-06-03:
> @liamstears, just FYI, the presence sensors battery reporting are the same unreliable in SmartThings. They just die one day to another.
>
> You are better if you set up a notification at some level and [...]

Hi and thank you for the reply

I think maybe my original post was not clear. What I mean is that the battery level is never reported or shown at all, it always just says "null" or "N/A"

It doesn't work at all and that is the problem that I am trying to fix here, I know the reporting can be unreliable but I would at least like something that I can monitor

GSzabados, 2021-06-04:
> It doesn't work at all and that is the problem that I am trying to fix here, I know the reporting can be unreliable but I would at least like something that I can monitor

Hi @liamstears,

I will try to pair one if my Arrival sensors to Z2M. (I haven't done that before and it is still connected to SmartThings.) I will try to have a look what is happening.

I am not really an expert with Z2M, but looking at your error messages, my guess is that the devices goes to sleep when you are trying to set reporting.

I would try to delete the device and re-pair it again with your modification, just to make sure that it does the configuration and the device is awake when your try to set attribute reporting.

GSzabados, 2021-06-04:
@liamstears, ok, I might have the newer model. I have mine from a v2 Hub starter kit.

But, I found your original issue, when you first tried to pair this arrival sensor.

https://github.com/Koenkk/zigbee2mqtt/issues/4055

What was missed is the periodic reporting. That reporting has the battery value. (I cannot see any reporting in the SmartThings DH for the 2015 model, it just receives a message in raw and parses the value from that.)

Koenkk, 2021-06-04:
@liamstears if it's indeed manufacturer specific this will be harder to support, but it's worth to try:
- ` await reporting.batteryPercentageRemaining(endpoint);` instead of ` await reporting.batteryVoltage(endpoint);`
- Make sure the device is awake when the configuring is done, you can manually trigger the configure via the frontend, click on the device and press the yellow refresh button.

GSzabados, 2021-06-04:
> @liamstears if it's indeed manufacturer specific this will be harder to support, but it's worth to try:

In the SmartThings DH there is no sign of configureReport and the battery report comes together with rssi and lqi, what is not something I have ever seen before.

Plus the beep works with a manufacturer specific cluster FC05 on endpoint 2 as well.

`// Used source endpoint of 0x02 because we are using smartthings manufacturer specific cluster.`

liamstears, 2021-06-10:
Wow there is a lot of information in here that goes above my head, thanks to all trying to help

What I don't get though is I get a linkquality reading but if battery, rssi and lqi are sent together surely I shouldn't get that reading?

liamstears, 2021-06-10:
> @liamstears, ok, I might have the newer model. I have mine from a v2 Hub starter kit.
> 
> But, I found your original issue, when you first tried to pair this arrival sensor.
> 
> #4055
> 
> What was missed is the periodic reporting. That reporting has the battery value. (I cannot see any reporting in the SmartThings DH for the 2015 model, it just receives a message in raw and parses the value from that.)
> 
> Edit:
> 
> It turned out, I have the same model. The reporting is terrible...
> 
> I've looked at the DH from SmartThings.
> 
> `private Boolean isBatteryMessage(String description) { // "raw:36EF1C, dni:36EF, battery:1B, rssi:, lqi:" description ==~ /.*battery:.*rssi:.*lqi:.*/}`
> 
> This code suggests that the device doesn't use the standard battery reporting.
> 
> It uses a manufacturer specific way to report battery, rssi and lqi.
> 
> To support this, a new cluster should be added to the zigbee-herdsman as it has happened with the manuSpecificSmartThingsArrivalSensor cluster.
> 
> Unfortunately the example in the SmartThings DH doesn't really show a clear message what the Hub receives from the sensor.
> 
> I am trying to catch it in the Live Logging what is exactly that message but reporting is absolutely zero. Maybe once a day...

Really appreciate your help and anything you can do to help resolve this problem

GSzabados, 2021-06-10:
@liamstears, I tried to get it from SmartThings live logging, but I couldn't catch it. Unfortunately I don't have time for it now.

I think LQI is translated from some other messages in Z2M.

But the device should report LQI, RSSI and battery in a messages.

To catch these messages you either need to run zigbee-herdsman logging for at least 24 hours or do sniffing for 24 hours. But you need to collect it and search for the right message afterward.

In SmartThings the device seems to report every 23 hours, plus 1 hour after that. (It doesn't makes sense but that is logged in my case...)

liamstears, 2021-06-10:
> @liamstears, I tried to get it from SmartThings live logging, but I couldn't catch it. Unfortunately I don't have time for it now.
>
> I think LQI is translated from some other messages in Z2M.
>
> But [...]

I can quite easily run zigbee-herdsman for 24 hours but is there an easy way to get that to log to a file for easy finding and reading?

GSzabados, 2021-06-10:
> I can quite easily run zigbee-herdsman for 24 hours but is there an easy way to get that to log to a file for easy finding and reading?

Not as I am aware of.

liamstears, 2021-06-14:
> > I can quite easily run zigbee-herdsman for 24 hours but is there an easy way to get that to log to a file for easy finding and reading?
> 
> Not as I am aware of.

I noticed when playing around some longer messages in herdsman than what I normally get, I don't know if anything in here is useful? https://pastebin.com/RZntHzn7

GSzabados, 2021-07-22:
I've asked an engineer at SmartThings, might can give some details on the device, but still hasn't received an answer.

closed 2021-08-31 by github-actions[bot]