Koenkk/zigbee2mqtt #7674: Add new device "Moes BRT-100_TRV _TZE200_b6wax7g0" Thermostat.

vladi1234, 2021-06-02:
Hello,
I'm new here, but I've finished my converter for Moes-BRT-100_TRV thermostat and already tested it, it works.
I created "devices.js" file and made changes to "/lib/tuya.js", "fromZigbee.js", "toZigbee.js" and "exposes.js" files.
Where should I go with this thermostat supported?
How should I add new "Moes BRT-100_TRV _TZE200_b6wax7g0" to the project?

vladi1234, 2021-06-02:
[image]

jmarfer, 2021-06-03:
Great your work. As soon as I have seen it I have wanted to try it. I'm sure I must do something wrong because it doesn't work for me, it won't start Zigbee2mqtt (502: Bad Gateway)

[image]

Thank you

wizu, 2021-06-03:
I've managed to start Z2M with a code from @vladi1234 with a little modification:
```
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
	{
		fingerprint: [
			{
				modelID: 'TS0601',
				manufacturerName: '_TZE200_b6wax7g0',
			},
		],
		model: 'BRT-100_TRV',
		vendor: 'Moes',
		description: 'Thermostatic Radiator Valve',
		supports: 'thermostat, temperature',
		fromZigbee: [
			fz.ignore_basic_report, fz.ignore_tuya_set_time,
			fz.tuya_data_point_dump,
			fz.moesS_thermostat,
		],
		toZigbee: [
			tz.tuya_data_point_test, // Another debug converter
			tz.moesS_thermostat_current_heating_setpoint, tz.moesS_thermostat_child_lock,
			tz.moesS_thermostat_window_detection, tz.moesS_thermostat_temperature_calibration,
			tz.moesS_thermostat_temperature_calibration, tz.moesS_thermostat_system_mode,
			tz.moesS_thermostat_boost_heating, tz.moesS_thermostat_boostHeatingCountdownTimeSet,
			tz.moesS_thermostat_eco_temperature, tz.moesS_thermostat_max_temperature, tz.moesS_thermostat_min_temperature,
			tz.moesS_thermostat_schedule, tz.moesS_thermostat_moesSecoMode,
			tz.moesS_thermostat_boost_heating_countdown,
		],
		onEvent: tuya.onEventSetLocalTime,
		meta: {
			configureKey: 1,
			tuyaThermostatSystemMode: tuya.thermostatSystemModes5,
		},
		configure: async (device, coordinatorEndpoint, logger) => {
			const endpoint = device.getEndpoint(1);
			await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic', 'manuSpecificTuya']);
		},
		exposes: [
			e.battery(), e.child_lock(),
			exposes.climate()
				.withLocalTemperature(ea.STATE).withSetpoint('current_heating_setpoint', 5, 30, 0.5, ea.STATE_SET)
				.withLocalTemperatureCalibration(ea.STATE_SET).withSystemMode(['auto', 'heat', 'off'], ea.STATE_SET),
			exposes.text('schedule', ea.STATE).withDescription('MANUAL MODE ☝ - In this mode, the device executes manual temperature setting. ' +
				'When the set temperature is lower than the "minimum temperature", the valve is closed (forced closed). ' +
				'PROGRAMMING MODE ⏱ -In this mode, the device executes a preset week programming temperature time and temperature. ' +
				'HOLIDAY MODE ⛱ - In this mode, for example, the vacation mode is set for 10 days and the temperature is set' +
				'to 15 degrees Celsius. After 10 days, the device will automatically switch to programming mode. ' +
				'TEMPORARY MANUAL MODE - In this mode, ☝ icon will flash. At this time, the device executes the manually set temperature' +
				'and returns to the weekly programming mode in the next time period. '),
			exposes.text('boost_heating', ea.STATE).withDescription('Boost Heating: Press and hold "+" for 3 seconds, the device will' +
				'enter the boost heating mode, and the ▷╵◁ will flash. The countdown will be displayed in the APP'),
			exposes.text('boost_heating_countdown', ea.STATE).withDescription('countdown in minutes'),
			e.boost_time(300, 900, 10, ea.STATE_SET).withUnit('sec'),
			exposes.text('eco_mode', ea.STATE).withDescription('ON/OFF state ECO MODE'), e.eco_temperature(),
			e.max_temperature(), e.min_temperature(), e.window_detection(), e.position(),
			// e.boost_heating(), e.eco_mode(),
		],
	}];
```

but I've got a lot of errors:
```
zigbee2mqtt_1    | Zigbee2MQTT:error 2021-06-03 15:40:04: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
zigbee2mqtt_1    |     at /app/lib/extension/receive.js:117:42
zigbee2mqtt_1    |     at Array.filter (<anonymous>)
zigbee2mqtt_1    |     at Receive.onZigbeeEvent (/app/lib/extension/receive.js:116:65)
zigbee2mqtt_1    |     at Controller.callExtensionMethod (/app/lib/controller.js:386:44))
```

vladi1234, 2021-06-03:
You still have to make changes in the four files "/lib/tuya.js", "fromZigbee.js", "toZigbee.js" and "exposes.js", and insert additional code.

vladi1234, 2021-06-03:
I apologize, I forgot the top line when copying / pasting.

**const fz = require('zigbee-herdsman-converters/converters/fromZigbee');**

jmarfer, 2021-06-03:
Great, I'm going to try right now and comment

vladi1234, 2021-06-03:
I have already corrected.

jmarfer, 2021-06-03:
Now it works. When changing some parameter of settings, this error appears:

Error Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'key' of undefined at /app/lib/extension/publish.js:185:56 at Array.find (<anonymous>) at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:185:42) at Controller.callExtensionMethod (/app/lib/controller.js:380:44) at processTicksAndRejections (internal/process/task_queues.js:93:5))

vladi1234, 2021-06-03:
---------------------------------------------------------------------------------
**add to node_modules/zigbee-herdsman-converters/lib/tuya.js**

```
    // MoesS Thermostat
    moesSsystemMode: 1,
    moesSheatingSetpoint: 2,
    moesSlocalTemp: 3,
    moesSboostHeating: 4,
    moesSboostHeatingCountdown: 5,
    moesSreset: 7,
    moesSwindowDetectionFunktion_A2: 8,
    moesSwindowDetection: 9,
    moesSchildLock: 13,
    moesSbattery: 14,
    moesSschedule: 101,
    moesSvalvePosition: 104,
    moesSboostHeatingCountdownTimeSet: 103,
    moesScompensationTempSet: 105,
    moesSecoMode: 106,
    moesSecoModeTempSet: 107,
    moesSmaxTempSet: 108,
    moesSminTempSet: 109,
```
------------------------------------------------------------------------------------------------------
**add to node_modules/zigbee-herdsman-converters/converters/fromZigbee.js**

```
   // ############################## fromZigbee.js moesS #######################################
    moesS_thermostat: {
        cluster: 'manuSpecificTuya',
        type: ['commandGetData', 'commandSetDataResponse'],
        convert: (model, msg, publish, options, meta) => {
            const dp = msg.data.dp; // First we get the data point ID
            const value = tuya.getDataValue(msg.data.datatype, msg.data.data);
            switch (dp) {
            case tuya.dataPoints.moesSsystemMode:
                switch (value) {
                case 0:
                    return {system_mode: 'auto'};
                case 1:
                    return {system_mode: 'manual'};
                case 2:
                    return {system_mode: 'temphand'};
                case 3:
                    return {system_mode: 'holiday'};
                }
                break;
            case tuya.dataPoints.moesSheatingSetpoint:
                // return {current_heating_setpoint: (value / 10).toFixed(1)};
                return {current_heating_setpoint: value};

            case tuya.dataPoints.moesSlocalTemp:
                return {local_temperature: (value / 10)};

            case tuya.dataPoints.moesSboostHeating:
                return {boost_heating: value ? 'ON' : 'OFF'};

            case tuya.dataPoints.moesSboostHeatingCountdown:
                return {boost_heating_countdown: value};

            case tuya.dataPoints.moesSreset:
                break;

            case tuya.dataPoints.moesSwindowDetectionFunktion_A2:
                return {window_detection: value ? 'ON' : 'OFF'};

            case tuya.dataPoints.moesSwindowDetection:
                return {window_detection: value ? 'ON' : 'OFF'};

[... 180 lines cut ...]
    // moesSminTempSet: 109,
    moesS_thermostat_min_temperature: {
        key: ['min_temperature'],
        convertSet: async (entity, key, value, meta) => {
            // const temp = Math.round(value * 10);
            await tuya.sendDataPointValue(entity, tuya.dataPoints.moesSminTempSet, value);
        },
    },
    // ########################### toZigbee.js ende moesS #########################################
```

------------------------------------------------------------------------------------------------------
**add to node_modules/zigbee-herdsman-converters/converters/lib/exposes.js**

Insert 'manual', 'temphand', 'holiday' as shown below.

```
    withSystemMode(modes, access=a.ALL) {
        assert(!this.endpoint, 'Cannot add feature after adding endpoint');
        const allowed = ['off', 'heat', 'cool', 'auto', 'dry', 'fan_only', 'sleep', 'manual', 'temphand', 'holiday'];
        modes.forEach((m) => assert(allowed.includes(m)));
        this.features.push(new Enum('system_mode', access, modes).withDescription('Mode of this device'));
        return this;
    }
```

jmarfer, 2021-06-03:
Sorry but I don't understand very well. I don't have those four files to modify. Do I have to create them? "/lib/tuya.js", "fromZigbee.js", "toZigbee.js" and "exposes.js"

vladi1234, 2021-06-03:
These four files are already there. You have to insert a piece of code, as I described above.

vladi1234, 2021-06-03:
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

wizu, 2021-06-03:
@vladi1234 try to make pull request to: https://github.com/Koenkk/zigbee-herdsman-converters
https://docs.github.com/en/github/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request

https://www.google.com/search?q=github+how+to+make+pull+request&oq=github+how+to+mak&aqs=chrome.1.0l2j69i57j0i22i30l7.3239j0j7&sourceid=chrome&ie=UTF-8

jmarfer, 2021-06-03:
Nothing, this is too complex for me, I am not able to get it.

I will buy one of the compatible valves that link by default.

Thank you very much for your patience and work

vladi1234, 2021-06-03:
> @vladi1234 try to make pull request to: https://github.com/Koenkk/zigbee-herdsman-converters [...]

Thanks for recommendation. I have to take more time to do right.

yarosman, 2021-06-03:
@vladi1234 if you use zigbee2mqtt hassio addon you don't have these files: "/lib/tuya.js", "fromZigbee.js", "toZigbee.js" and "exposes.js"

vladi1234, 2021-06-03:
I am not using the hassio addon. I favor zigbee2mqtt, Domoticz and Node-red.

jmarfer, 2021-06-03:
so hard to be able to do it

wizu, 2021-06-04:
@vladi1234 windows detection seems not to work. 
I can't wait for you pull request and when it will be merged and this thermostat will work out of the box ;)

@jmarfer single file version:
```
const tuyaDataPoints = {
	moesSsystemMode: 1,
	moesSheatingSetpoint: 2,
	moesSlocalTemp: 3,
	moesSboostHeating: 4,
	moesSboostHeatingCountdown: 5,
	moesSreset: 7,
	moesSwindowDetectionFunktion_A2: 8,
	moesSwindowDetection: 9,
	moesSchildLock: 13,
	moesSbattery: 14,
	moesSschedule: 101,
	moesSvalvePosition: 104,
	moesSboostHeatingCountdownTimeSet: 103,
	moesScompensationTempSet: 105,
	moesSecoMode: 106,
	moesSecoModeTempSet: 107,
	moesSmaxTempSet: 108,
	moesSminTempSet: 109,
};

const tuya = {...require('zigbee-herdsman-converters/lib/tuya')};
tuya.dataPoints = {...tuya.dataPoints, ...tuyaDataPoints}

const fzExt = {
	moesS_thermostat: {
		cluster: 'manuSpecificTuya',
		type: ['commandGetData', 'commandSetDataResponse'],
		convert: (model, msg, publish, options, meta) => {
			const dp = msg.data.dp; // First we get the data point ID
			const value = tuya.getDataValue(msg.data.datatype, msg.data.data);
			switch (dp) {
				case tuya.dataPoints.moesSsystemMode:
					switch (value) {
						case 0:
							return {system_mode: 'auto'};
						case 1:
							return {system_mode: 'manual'};
						case 2:
							return {system_mode: 'temphand'};
						case 3:
							return {system_mode: 'holiday'};
					}
					break;
				case tuya.dataPoints.moesSheatingSetpoint:
					// return {current_heating_setpoint: (value / 10).toFixed(1)};
					return {current_heating_setpoint: value};

				case tuya.dataPoints.moesSlocalTemp:
					return {local_temperature: (value / 10)};

				case tuya.dataPoints.moesSboostHeating:
					return {boost_heating: value ? 'ON' : 'OFF'};

				case tuya.dataPoints.moesSboostHeatingCountdown:
					return {boost_heating_countdown: value};

				case tuya.dataPoints.moesSreset:
					break;

				case tuya.dataPoints.moesSwindowDetectionFunktion_A2:
					return {window_detection: value ? 'ON' : 'OFF'};

				case tuya.dataPoints.moesSwindowDetection:
					return {window_detection: value ? 'ON' : 'OFF'};
[... 236 lines cut ...]
			const endpoint = device.getEndpoint(1);
			await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic', 'manuSpecificTuya']);
		},
		exposes: [
			e.battery(), e.child_lock(),
			climate
				.withLocalTemperature(ea.STATE).withSetpoint('current_heating_setpoint', 5, 30, 0.5, ea.STATE_SET)
				.withLocalTemperatureCalibration(ea.STATE_SET).withSystemMode(['auto', 'manual', 'temphand', 'holiday'], ea.STATE_SET),
			exposes.text('schedule', ea.STATE).withDescription('MANUAL MODE ☝ - In this mode, the device executes manual temperature setting. ' +
				'When the set temperature is lower than the "minimum temperature", the valve is closed (forced closed). ' +
				'PROGRAMMING MODE ⏱ -In this mode, the device executes a preset week programming temperature time and temperature. ' +
				'HOLIDAY MODE ⛱ - In this mode, for example, the vacation mode is set for 10 days and the temperature is set' +
				'to 15 degrees Celsius. After 10 days, the device will automatically switch to programming mode. ' +
				'TEMPORARY MANUAL MODE - In this mode, ☝ icon will flash. At this time, the device executes the manually set temperature' +
				'and returns to the weekly programming mode in the next time period. '),
			exposes.text('boost_heating', ea.STATE).withDescription('Boost Heating: Press and hold "+" for 3 seconds, the device will' +
				'enter the boost heating mode, and the ▷╵◁ will flash. The countdown will be displayed in the APP'),
			exposes.text('boost_heating_countdown', ea.STATE).withDescription('countdown in minutes'),
			e.boost_time(300, 900, 10, ea.STATE_SET).withUnit('sec'),
			exposes.text('eco_mode', ea.STATE).withDescription('ON/OFF state ECO MODE'), e.eco_temperature(),
			e.max_temperature(), e.min_temperature(), e.window_detection(), e.position(),
			// e.boost_heating(), e.eco_mode(),
		],
	}];
```

vladi1234, 2021-06-04:
I just made a pull request.

wizu, 2021-06-04:
@vladi1234 I cannot find it. Could you link to it?

vladi1234, 2021-06-04:
https://github.com/vladi1234/zigbee-herdsman-converters/pulls?q=is%3Apr+is%3Aclosed

wizu, 2021-06-04:
You should now create one Pull Request to main repo: https://github.com/Koenkk/zigbee-herdsman-converters/compare

vladi1234, 2021-06-04:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/2652

jmarfer, 2021-06-04:
Hello @wizu!

I just put the unique configuration. The following happens to me:
 
I have sensors aqara that when they change state I get tts notices by alexa. Well, when I touch any valve parameter, I get tts notifications from those sensors. I have verified that these warnings are due to these sensors losing connection (not available), which without this configuration does not happen to me

Also comment that the battery level is not shown

jmarfer, 2021-06-04:
Every time I modify any valve parameter, the zigbee sensors remain "unavailable" and reconnect. Any suggestion?

In the code I get this error (in case it could be the cause):

[image]

vladi1234, 2021-06-05:
Hallo @jmarfer !
I am not very familiar with HA, so I can't really help with HA, but I've already made a pull request and requested merging with zigbee2mqtt.
For you it would be best to wait until the merge is finished and the thermostat is supported.

jmarfer, 2021-06-05:
Ok @vladi1234, I will do so. Thank you so much!

jmarfer, 2021-06-08:
Any news about the definitive compatibility of this valve?

vladi1234, 2021-06-11:
The thermostat 'Moes BRT-100-TRV _TZE200_b6wax7g0' is now supported.

jmarfer, 2021-06-11:
Thank you very much @ vladi1234. A question. I'm on version 1.18 and I don't get an update on zigbbe2mqtt (1.19). I have added the repository in the component store and the new version does not appear either. How do I solve it?

jmarfer, 2021-06-11:
I have already been able to update!

jmarfer, 2021-06-11:
Sorry to say that by default it doesn't link me. When I say by default I mean that I have eliminated the .js file where the code was, because as it is supported I understand that it is no longer necessary

jmarfer, 2021-06-11:
Nothing, I have tried to put the .js file again and the communication problems of the other sensors continue, in addition to the valve not working well.

vladi1234, 2021-06-12:
Today the last malfunctions were solved and got into z2m.
Unfortunately I can't say when it's ready for HA.

https://github.com/Koenkk/zigbee-herdsman-converters/pull/2683

closed 2021-06-12 by vladi1234

jmarfer, 2021-06-12:
Do we have unique yaml code to test until it is available in HA?

ialtuhov, 2021-11-16:
Hi
I ask for help, I can't figure it out, I'm a beginner in this field.
I have the same device TS0601 '_TZE200_B6WAX7G0'. But it is not defined in zigbee2mqtt, Support status is Unsupported.

I am using HA Supervisor (core-2021.11.4) and Zigbee2mqtt (version: 1.18.1-1).
Please tell me how to make it work???

andiukas, 2021-11-16:
update zigbee2mqtt :)

ialtuhov, 2021-11-16:
how to do it? HA writes that the version is the latest

andiukas, 2021-11-16:
The last 1.22-0-2 version, update from homeassistant :)

ialtuhov, 2021-11-16:
Thanks. I didn't see that the repository was changed

KaHeMu, 2022-01-06:
@vladi1234 Great job! Thanks for creating this converter.  I'm using the BRT-100-TRV with Z2M 1.22.1 and zStack3x0 and it works (almost) perfectly. One thing I wonder is about the holiday mode. How can I change the setpoint temperature for the holiday mode or is it fixed to 15°C? And I also have some issues with valve position reporting. See [10384](https://github.com/Koenkk/zigbee2mqtt/issues/10384)  But anyway, thanks!

vladi1234, 2022-01-06:
Hi,
Holiday mode cannot be changed.

Siehe Manual:

[image]

KaHeMu, 2022-02-02:
@vladi1234 I've been using a few of these TRVs for a while and noticed that when 'preset' is set to 'HOLIDAY MODE' (was 'PROGAMMING' before), instead of switching to setpoint 15°C it switches to the last 'current_heating_setpoint'. Was there a change in your converter that caused this intentionally? I can reproduce it easily. I remember when the TRVs were new that it turned to 15°C when preset to HOLIDAY MODE (see my previous post above). I know it sounds a bit strange but this is what I'm observing repeatedly now. Any idea? 
(Z2M v1.22.1 commit c2b5229)