Koenkk/zigbee2mqtt #7679: Request to add a Niko motion detector device

Fafitude, 2021-06-03:
..Hello,

please add the Niko motion detector device reference 552-80401.
It is recognized by zigbee2mqtt but not supported.
Manufacturer link: https://www.niko.eu/fr-be/article/552-80401

If some information is missing, I can try to provide it to you (sorry this is my first request here).

thank you in advance.

sjorge, 2021-06-03:
Did you try to pair the device already? If so whats the database entry?
The technical docs available from Niko are purely user focussed so it's hard to guess what needs to be done converter side to get it to work.

Fafitude, 2021-06-03:
Yes, I paired it. Here is what I got in the logs:
`Info Successfully interviewed '0x90fd9ffffe61a657', device has successfully been paired
Attention Device '0x90fd9ffffe61a657' with Zigbee model 'Connectable motion sensor,Zigbee' and manufacturer name 'NIKO NV' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x90fd9ffffe61a657","ieee_address":"0x90fd9ffffe61a657","status":"successful","supported":false},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x90fd9ffffe61a657","supported":false},"type":"pairing"}'

Attention Received message from unsupported device with Zigbee model 'Connectable motion sensor,Zigbee' and manufacturer name 'NIKO NV'
Attention Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

	

0x90fd9ffffe61a657	0x90fd9ffffe61a657 ( 0xE209 )	Non pris en charge	Détecteur de mouvement connectable, Zigbee	234	
`

sjorge, 2021-06-03:
Can you grep '0x90fd9ffffe61a657' in database.db it should have a list of in and out clusters.

Fafitude, 2021-06-03:
Sorry, I'm a beginner, I don't know what to do?

sjorge, 2021-06-03:
There should be 'database.db' file in the data/ directory of your z2m install, it should have a line in it that contains 0x90fd9ffffe61a657, other info on that line are needed to know what the clusters and endpoint the device has. (WHich we need to write the device entry)

Here is an example from a Niko smart plug:
```
{"id":94,"type":"Router","ieeeAddr":"0x60a423fffef84050","nwkAddr":55994,"manufId":4703,"manufName":"Niko NV","powerSource":"Mains (single phase)","modelId":"Smart plug Zigbee PE","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,3,4,5,6,1794,2820,2821,4096,64642],"outClusterList":[3,10,25],"clusters":{"genBasic":{"attributes":{"modelId":"Smart plug Zigbee PE","manufacturerName":"Niko NV","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20200929","swBuildId":"2.03.01"}},"haElectricalMeasurement":{"attributes":{"acPowerMultiplier":1,"acPowerDivisor":1,"activePower":776}},"seMetering":{"attributes":{"multiplier":1,"divisor":100,"currentSummDelivered":[0,313]}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":255}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00228120b5","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b00228120b5","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b00228120b5","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":1},{"cluster":1794,"attrId":0,"minRepIntval":60,"maxRepIntval":3600,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20200929","swBuildId":"2.03.01","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1622731178015}
```

Fafitude, 2021-06-03:
je pense que c'est ça `{"id":58,"type":"EndDevice","ieeeAddr":"0x90fd9ffffe61a657","nwkAddr":5602,"manufId":4703,"manufName":"NIKO NV","powerSource":"Battery","modelId":"Connectable motion sensor,Zigbee","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1280,2821,64641],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"Connectable motion sensor,Zigbee","manufacturerName":"NIKO NV","powerSource":3,"appVersion":4,"stackVersion":2,"hwVersion":1,"dateCode":"20190702-70","swBuildId":"4.5"}},"ssIasZone":{"attributes":{"65531":1,"iasCieAddr":"0x00212effff04e808","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":4,"stackVersion":2,"hwVersion":1,"dateCode":"20190702-70","swBuildId":"4.5","interviewCompleted":true,"meta":{},"lastSeen":1622731962385}`

sjorge, 2021-06-03:
Yes! I'll see what I can do next time I have a bit of time, unless Koen beats me to it.

Fafitude, 2021-06-03:
Super, merci :)

sjorge, 2021-06-03:
No guarantees as I don't have the device but try adding this to devices/niko.js

```
    {
        zigbeeModel: ['Connectable motion sensor,Zigbee'],
        model: '552-80401',
        vendor: 'Niko',
        description: 'Wireless motion sensor',
        fromZigbee: [fz.ias_occupancy_alarm_1, fz.battery],
        toZigbee: [],
        meta: {battery: {voltageToPercentage: '3V_2100'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            const bindClusters = ['genPowerCfg'];
            await reporting.bind(endpoint, coordinatorEndpoint, bindClusters);
            await reporting.batteryVoltage(endpoint);
        },
        exposes: [e.occupancy(), e.battery_low(), e.battery()],
    },
```

Fafitude, 2021-06-03:
Je voudrais essayer mais je ne trouve pas le dossier devices dans Z2M.
Pouvez-vous m'indiquez le chemin svp ?

sjorge, 2021-06-03:
My install is in `/opt/zigbee2mqtt` and the file you need to edit is  `/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/niko.js`

Fafitude, 2021-06-03:
Merci.
Mais j'ai encore un problème, l'accès est refusé quand je sauvegarde le fichier.
J'utilise mobaxterm et je suis connecté en root.

sjorge, 2021-06-03:
How are you running zigbee2mqtt? Docter? Manual install?

Fafitude, 2021-06-03:
Oui, installer sur une vm.

sjorge, 2021-06-03:
Not sure how to do it on docker, I think the path is not writable in the container. I guess we wait for @Koenkk as I don't use docker and have no idea what the proper workaround for that is.

Fafitude, 2021-06-03:
D'accord, je ne sais pas non plus, je vais regarder de mon côté si je trouve une solution

Fafitude, 2021-06-04:
Bonjour,
j'ai essayé de voir si mes droits root sont bons et ça me parait ok, j'arrive pas à mettre un fichier dans le dossier.
Une idée ?
Merci d'avance.

Fafitude, 2021-06-06:
Bonjour,
j'ai réussi à l'intégrer, ça fonctionne.
Vous avez besoin de quelque chose ?

sjorge, 2021-06-06:
Ah nice you got it to work! I'll open a PR so it is in the next release.

Fafitude, 2021-07-07:
Merci ça fonctionne très bien.

closed 2021-07-07 by Fafitude