Koenkk/zigbee2mqtt #7680: Unable to control Zigbee2MQTT Covers in HA 2021.6.0

BarBaar44, 2021-06-03:
# Bug Report
Zigbee2MQTT covers no longer work in HA 2021.6.0 from HA. Not manually of by automation.

The web interface in HA of Zigbee2MQTT in HA doesn't show up either. May be related

## What did you expect to happen
The covers to keep working

## How to reproduce it (minimal and precise)
Upgrade to HA 2021.6.0

## Debug Info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: CC26X2R1
Adapter firmware version:

Logoutput from HA-logs:

`Logger: homeassistant.util.logging
Source: util/logging.py:105
First occurred: 8:25:19 (10 occurrences)
Last logged: 8:26:01

`Exception in async_discover when dispatching 'mqtt_discovery_new_cover_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/Cover Tim/set', 'device': {'identifiers': ['zigbee2mqtt_0x04cf8cdf3c738d2c'], 'manufacturer': 'Xiaomi', 'model': 'Aqara B1 curtain motor (ZNCLDJ12LM)', 'name': 'Cover Tim', 'sw_version': 'Zigbee2MQTT 1.18.1'}, 'json_attributes_topic': 'zigbee2mqtt/Cover Tim', 'name': 'Cover Tim', 'position_topic': 'zigbee2mqtt/Cover Tim', 'set_position_template': '{ "position": {{ position }} }', 'set_position_topic': 'zigbee2mqtt/Cover Tim/set', 'unique_id': '0x04cf8cdf3c738d2c_cover_zigbee2mqtt', 'value_template': '{{ value_json.position }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/cover.py", line 118, in validate_options raise vol.Invalid( voluptuous.error.Invalid: 'value_template' must be set together with 'state_topic'. During handling of the above exception, another exception occurred: Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 160, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 276, in __call__ raise er.MultipleInvalid([e]) voluptuous.error.MultipleInvalid: 'value_template' must be set together with 'state_topic'.`

pznamenskii, 2021-06-03:
same for my curtain - different model.
perhaps due to this change - https://github.com/home-assistant/core/pull/50263?

Logger: homeassistant.util.logging
Source: util/logging.py:105
First occurred: 9:46:20 AM (1 occurrences)
Last logged: 9:46:20 AM

Exception in async_discover when dispatching 'mqtt_discovery_new_cover_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/curtain_room/set', 'device': {'identifiers': ['zigbee2mqtt_0x00158d00039cdd57'], 'manufacturer': 'Xiaomi', 'model': 'Aqara curtain motor (ZNCLDJ11LM)', 'name': 'curtain_room', 'sw_version': 'Zigbee2MQTT 1.18.1'}, 'json_attributes_topic': 'zigbee2mqtt/curtain_room', 'name': 'curtain_room', 'position_topic': 'zigbee2mqtt/curtain_room', 'set_position_template': '{ "position": {{ position }} }', 'set_position_topic': 'zigbee2mqtt/curtain_room/set', 'unique_id': '0x00158d00039cdd57_cover_zigbee2mqtt', 'value_template': '{{ value_json.position }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable return schema(data) File "/usr/src/homeassistant/homeassistant/components/mqtt/cover.py", line 118, in validate_options raise vol.Invalid( voluptuous.error.Invalid: 'value_template' must be set together with 'state_topic'. During handling of the above exception, another exception occurred: Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 160, in async_discover config = schema(discovery_payload) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 218, in __call__ return self._exec((Schema(val) for val in self.validators), v) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 341, in _exec raise e if self.msg is None else AllInvalid(self.msg, path=path) File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 337, in _exec v = func(v) File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 276, in __call__ raise er.MultipleInvalid([e]) voluptuous.error.MultipleInvalid: 'value_template' must be set together with 'state_topic'.

DubhAd, 2021-06-03:
https://www.home-assistant.io/blog/2021/06/02/release-20216/#other-breaking-changes

* Using `value_template` to use for extracting position is no longer allowed an error will be raised if this configuration is used, Instead of using `value_template`, `position_template` should be used.
* `tilt_invert_state` is removed, instead `tilt_min` and `tilt_max` should be used, error will be raised if this is found in configuration.

Which is the change you identified

kirbo, 2021-06-03:
I'm also experiencing the same issue, with Ikea Fyrtur covers:
```
2021-06-03 09:39:12 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_cover_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'command_topic': 'zigbee2mqtt/Ikea Fyrtur - Nuutin huone/set', 'device': {'identifiers': ['zigbee2mqtt_0x842e14fffe75e80a'], 'manufacturer': 'IKEA', 'model': 'FYRTUR roller blind (E1757)', 'name': 'Ikea Fyrtur - Nuutin huone', 'sw_version': 'Zigbee2MQTT 1.18.1'}, 'json_attributes_topic': 'zigbee2mqtt/Ikea Fyrtur - Nuutin huone', 'name': 'Ikea Fyrtur - Nuutin huone', 'position_topic': 'zigbee2mqtt/Ikea Fyrtur - Nuutin huone', 'set_position_template': '{ "position": {{ position }} }', 'set_position_topic': 'zigbee2mqtt/Ikea Fyrtur - Nuutin huone/set', 'unique_id': '0x842e14fffe75e80a_cover_zigbee2mqtt', 'value_template': '{{ value_json.position }}', 'platform': 'mqtt'},)
Traceback (most recent call last):
File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__
return self._compiled([], data)
File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable
return schema(data)
File "/usr/src/homeassistant/homeassistant/components/mqtt/cover.py", line 118, in validate_options
raise vol.Invalid(
voluptuous.error.Invalid: 'value_template' must be set together with 'state_topic'.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 160, in async_discover
config = schema(discovery_payload)
File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 218, in __call__
return self._exec((Schema(val) for val in self.validators), v)
File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 341, in _exec
raise e if self.msg is None else AllInvalid(self.msg, path=path)
File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 337, in _exec
v = func(v)
File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 276, in __call__
raise er.MultipleInvalid([e])
voluptuous.error.MultipleInvalid: 'value_template' must be set together with 'state_topic'.
```

pznamenskii, 2021-06-03:
so the topic raised by zigbee2mqtt needs to be corrected

Koenkk, 2021-06-03:
This issue has already been fixed in 1.19.0, if you don't see the update see the first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2021-06-03 by Koenkk