Koenkk/zigbee2mqtt #7690: Losonho ZB-CL01 simillar GU10 don't turn off with transition enabled on YAML

raphaeleduardo42, 2021-06-03:
## What happened
After define the transition option for a No Brand GU10 recognized as Losonho ZB-CL01 is not possible to turn off the bulb. The issue occurs with two different units. When defined brightness below 1 not happens. All messages are sended, but the device don't turn off.

## What did you expect to happen
Turn off as with the transition option not set

## How to reproduce it (minimal and precise)
Define transition for all devices or to specific on configuration.yaml

## Debug info
Zigbee2MQTT version: 1.19.0 89cb4f95
Adapter hardware: CC2531
Adapter firmware version: 20201127

Thrasher2020, 2021-07-10:
I have the same issue.

Thrasher2020, 2021-07-13:
I am seeing the same issue with version 1.20.0

Koenkk, 2021-07-13:
Cannot be fixed, if this doesn't work it means the firmware of the light does not support transitions properly.

Thrasher2020, 2021-07-13:
> Cannot be fixed, if this doesn't work it means the firmware of the light does not support transitions properly.

Thanks - it's just odd that they transition to 'on' without an issue. I'll leave them with a scene transition to off as zero (as that works fine)

Thrasher2020, 2021-07-14:
> ## What happened
> After define the transition option for a No Brand GU10 recognized as Losonho ZB-CL01 is not possible to turn off the bulb. The issue occurs with two different units. When defined brightness below 1 not happens. All messages are sended, but the device don't turn off.
> 
> ## What did you expect to happen
> Turn off as with the transition option not set
> 
> ## How to reproduce it (minimal and precise)
> Define transition for all devices or to specific on configuration.yaml
> 
> ## Debug info
> Zigbee2MQTT version: 1.19.0 [89cb4f9](https://github.com/Koenkk/zigbee2mqtt/commit/89cb4f95368423f55047d2529ec96ab762b74ca0)
> Adapter hardware: CC2531
> Adapter firmware version: 20201127

Hi - I "fixed" this by using CHOICE in HA - basically if the light is on and it's saving the state and returning to another the transition works 100%. If it's going back to off it doesn't, so a simple check to see if it was already on = do a transition otherwise just issue an off with no transition. As above it seems to be a failing of the firmware (of which there is no updated version as of now).

Thanks to @Koenkk :)

closed 2021-08-22 by github-actions[bot]

cmiguelcabral, 2023-02-01:
I bought some Lonsonho ZB-RGBCW GU10 (reported as https://www.zigbee2mqtt.io/devices/ZB-RGBCW.html#lonsonho-zb-rgbcw).
They also do not turn off if I set some transition time.
- The transition works, and it respects the time I set on z2m configuration, but it does not turn off. To turn it off, I need to manually set the brightness to 1.
- If I remove the transition property from z2m, then I can turn them ON/OFF nicely and it returns to the last brightness value. The bulb has its own transition for ON/OFF, but then there is no transition when changing brightness or color. In this case, if I send the transition property when sending commands to z2m through mqtt, the transition works to change things, but again, not to turn them OFF (in the case I send transition in the json payload).
- Setting transition to the bulbs, and adding them to a group without transition, we can turn the bulbs off if we control the group, but individually we can't.

So my guess is, in the case we had, let's say, a new boolean property on z2m's configuration to disable transition only for ON/OFF commands, everything would work.
I think this could be related to this change: https://github.com/Koenkk/zigbee2mqtt/issues/2850

cmiguelcabral, 2023-02-05:
I was messing around with this, and created an external converter for those bulbs. I mixed it up with the inovelli bulb, and got it working.
I copied the `tzLocal.light_onoff_brightness_inovelli` portion of code, and changed the brightness to turn off from 0 to 1 and voila, the bulbs work as they should. 
Now I can use transition on the bulbs and turn them OFF/ON, they will restore to the last state.

Thrasher2020, 2023-02-16:
Thanks for this but I can't use on/off if I use the external converter. I just get this error : 

`Error 2023-02-16 16:49:52Publish 'set' 'state' to 'Utility' failed: 'ReferenceError: converters is not defined'

Debug 2023-02-16 16:49:52ReferenceError: converters is not defined at Object.convertSet (/app/data/extension/externally-loaded.js:110:28) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:52) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:121:27) at WebSocket.emit (node:events:513:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1180:20) at Receiver.emit (node:events:513:28) at Receiver.dataMessage (/app/node_modules/ws/lib/receiver.js:541:14)`

cmiguelcabral, 2023-02-16:
> Thanks for this but I can't use on/off if I use the external converter. I just get this error :
> 
> `Error 2023-02-16 16:49:52Publish 'set' 'state' to 'Utility' failed: 'ReferenceError: converters is not defined'
> 
> Debug 2023-02-16 16:49:52ReferenceError: converters is not defined at Object.convertSet (/app/data/extension/externally-loaded.js:110:28) at Publish.onMQTTMessage (/app/lib/extension/publish.ts:246:52) at EventEmitter.emit (node:events:525:35) at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:109:22) at MQTT.onMessage (/app/lib/mqtt.ts:140:27) at WebSocket. (/app/lib/extension/frontend.ts:121:27) at WebSocket.emit (node:events:513:28) at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1180:20) at Receiver.emit (node:events:513:28) at Receiver.dataMessage (/app/node_modules/ws/lib/receiver.js:541:14)`

That's because you do not have a `transition` set, in that case, you don't need this external converter.
Anyway, thank's for pointing it out, I'll try to fix it and make it work both ways.

cmiguelcabral, 2023-02-16:
[lonsonho.zip](https://github.com/Koenkk/zigbee2mqtt/files/10761865/lonsonho.zip)

@Thrasher2020, try this one.
It should work now with and without `transition`.

Regarding this external converter, I don't know if it makes sense to integrate it into the `zigbee-herdsman-converters` release as there might be some other bulbs recognized as the same device, and this code might not work on those others bulbs...