Koenkk/zigbee2mqtt #7691: Eria Smart Siren 81868

snakuzzo, 2021-06-03:
I'm trying to add this device...

https://zigbeealliance.org/zigbee_products/adurosmart-siren/

Followed instructions and I tried different configurations.
This is my last external converter...

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Smart Siren'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '81868', // Vendor model number, look on the device for a model number
    vendor: 'AduroSmart', // Vendor of the device (only used for documentation and startup logging)
    description: 'ERIA smart siren', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.ias_no_alarm],
    toZigbee: [tz.warning],
    exposes: [e.warning(), e.battery_low(), e.tamper()]
}

module.exports = definition;
```

This is the log when I try to pair...

```
Zigbee2MQTT:debug 2021-06-03 19:50:59: Device '0x00158d00054f1a28' announced itself
Zigbee2MQTT:info  2021-06-03 19:50:59: MQTT publish: topic 'zigbee2mqtt-dev/bridge/event', payload '{"data":{"friendly_name":"0x00158d00054f1a28","ieee_address":"0x00158d00054f1a28"},"type":"device_announce"}'
Zigbee2MQTT:info  2021-06-03 19:50:59: MQTT publish: topic 'zigbee2mqtt-dev/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d00054f1a28"},"type":"device_announced"}'
Zigbee2MQTT:debug 2021-06-03 19:50:59: Received Zigbee message from '0x00158d00054f1a28', type 'readResponse', cluster 'genBasic', data '{"modelId":"Smart Siren"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-03 19:50:59: Received Zigbee message from '0x00158d00054f1a28', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"AduroSmart Eria"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-03 19:50:59: Received Zigbee message from '0x00158d00054f1a28', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-03 19:50:59: Received Zigbee message from '0x00158d00054f1a28', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-03 19:50:59: Received Zigbee message from '0x00158d00054f1a28', type 'readResponse', cluster 'genBasic', data '{"appVersion":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-03 19:50:59: Received Zigbee message from '0x00158d00054f1a28', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-03 19:51:00: Received Zigbee message from '0x00158d00054f1a28', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-03 19:51:00: Received Zigbee message from '0x00158d00054f1a28', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20200629"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-03 19:51:00: Received Zigbee message from '0x00158d00054f1a28', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"01000003"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-03 19:51:00: Received Zigbee message from '0x00158d00054f1a28', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x00124b00076887d9","zoneState":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2021-06-03 19:51:00: No converter available for '81868' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x00124b00076887d9","zoneState":0}'
Zigbee2MQTT:error 2021-06-03 19:51:00: Failed to interview '0x00158d00054f1a28', device has not successfully been paired
```

snakuzzo, 2021-06-05:
Here the zigbee-herdsman debug log...

```
2021-06-05T20:08:22.075Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,74,163,40,26,79,5,0,141,21,0,0,0,138]
2021-06-05T20:08:22.075Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,74,163,40,26,79,5,0,141,21,0,0,0,138]
2021-06-05T20:08:22.076Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [74,163,40,26,79,5,0,141,21,0,0,0] - 138
2021-06-05T20:08:22.076Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":41802,"extaddr":"0x00158d00054f1a28","parentaddr":0}
2021-06-05T20:08:22.077Z zigbee-herdsman:controller:log Device '0x00158d00054f1a28' joined
2021-06-05T20:08:22.078Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-05T20:08:22.078Z zigbee-herdsman:controller:log Device '0x00158d00054f1a28' accepted by handler
2021-06-05T20:08:22.078Z zigbee-herdsman:controller:log New device '0x00158d00054f1a28' joined
2021-06-05T20:08:22.078Z zigbee-herdsman:controller:log Creating device '0x00158d00054f1a28'
Zigbee2MQTT:info  2021-06-05 22:08:22: Device '0x00158d00054f1a28' joined
Zigbee2MQTT:info  2021-06-05 22:08:22: MQTT publish: topic 'zigbee2mqtt-dev/bridge/event', payload '{"data":{"friendly_name":"0x00158d00054f1a28","ieee_address":"0x00158d00054f1a28"},"type":"device_joined"}'
2021-06-05T20:08:22.084Z zigbee-herdsman:controller:log Interview '0x00158d00054f1a28' start
Zigbee2MQTT:info  2021-06-05 22:08:22: Starting interview of '0x00158d00054f1a28'
Zigbee2MQTT:info  2021-06-05 22:08:22: MQTT publish: topic 'zigbee2mqtt-dev/bridge/event', payload '{"data":{"friendly_name":"0x00158d00054f1a28","ieee_address":"0x00158d00054f1a28","status":"started"},"type":"device_interview"}'
2021-06-05T20:08:22.088Z zigbee-herdsman:controller:device:log Interview - start device '0x00158d00054f1a28'
2021-06-05T20:08:22.090Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - nodeDescReq - {"dstaddr":41802,"nwkaddrofinterest":41802}
2021-06-05T20:08:22.090Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,2,74,163,74,163,35]
Zigbee2MQTT:info  2021-06-05 22:08:22: MQTT publish: topic 'zigbee2mqtt-dev/bridge/log', payload '{"message":{"friendly_name":"0x00158d00054f1a28"},"type":"device_connected"}'
Zigbee2MQTT:info  2021-06-05 22:08:22: MQTT publish: topic 'zigbee2mqtt-dev/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00158d00054f1a28"},"type":"pairing"}'
2021-06-05T20:08:22.594Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,2,0,102]
2021-06-05T20:08:22.594Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,2,0,102]
2021-06-05T20:08:22.594Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 2 - [0] - 102
2021-06-05T20:08:22.595Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - nodeDescReq - {"status":0}
2021-06-05T20:08:22.595Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-05T20:08:22.639Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,18,69,130,74,163,0,74,163,1,64,142,45,18,127,90,0,0,44,90,0,0,118]
2021-06-05T20:08:22.640Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,18,69,130,74,163,0,74,163,1,64,142,45,18,127,90,0,0,44,90,0,0,118]
2021-06-05T20:08:22.640Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 18 - 2 - 5 - 130 - [74,163,0,74,163,1,64,142,45,18,127,90,0,0,44,90,0,0] - 118
2021-06-05T20:08:22.640Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - nodeDescRsp - {"srcaddr":41802,"status":0,"nwkaddr":41802,"logicaltype_cmplxdescavai_userdescavai":1,"apsflags_freqband":64,"maccapflags":142,"manufacturercode":4653,"maxbuffersize":127,"maxintransfersize":90,"servermask":11264,"maxouttransfersize":90,"descriptorcap":0}
2021-06-05T20:08:22.640Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-05T20:08:22.643Z zigbee-herdsman:controller:device:log Interview - got node descriptor for device '0x00158d00054f1a28'
2021-06-05T20:08:22.644Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - activeEpReq - {"dstaddr":41802,"nwkaddrofinterest":41802}
2021-06-05T20:08:22.644Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,4,37,5,74,163,74,163,36]
2021-06-05T20:08:22.658Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,5,0,97]
2021-06-05T20:08:22.658Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,5,0,97]
2021-06-05T20:08:22.658Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 5 - [0] - 97
2021-06-05T20:08:22.658Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - activeEpReq - {"status":0}
2021-06-05T20:08:22.658Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-06-05T20:08:22.684Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,74,163,74,163,40,26,79,5,0,141,21,0,142,231]
2021-06-05T20:08:22.684Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,74,163,74,163,40,26,79,5,0,141,21,0,142,231]
2021-06-05T20:08:22.684Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [74,163,74,163,40,26,79,5,0,141,21,0,142] - 231
2021-06-05T20:08:22.684Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":41802,"nwkaddr":41802,"ieeeaddr":"0x00158d00054f1a28","capabilities":142}
2021-06-05T20:08:22.684Z zigbee-herdsman:controller:log Device announce '0x00158d00054f1a28'
Zigbee2MQTT:debug 2021-06-05 22:08:22: Device '0x00158d00054f1a28' announced itself
[... cut ...]
2021-06-05T20:08:23.700Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-06-05T20:08:23.700Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,23,209]
2021-06-05T20:08:23.701Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,23] - 209
2021-06-05T20:08:23.701Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":23}
2021-06-05T20:08:23.701Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

snakuzzo, 2021-07-06:
Hi all!
I'm stuck. I can't add the device. Any ideas ?

Koenkk, 2021-07-06:
Try adding `tz.warning` to `toZigbee` and use https://www.zigbee2mqtt.io/devices/AV2010_29A.html#controlling to control

snakuzzo, 2021-07-06:
Same result...

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Smart Siren'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '81868', // Vendor model number, look on the device for a model number
    vendor: 'AduroSmart', // Vendor of the device (only used for documentation and startup logging)
    description: 'ERIA smart siren', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.ias_no_alarm],
    toZigbee: [tz.warning],
    exposes: []
}

module.exports = definition;
```

```
2021-07-06 20:29:50: No converter available for '81868' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x00124b00076887d9","zoneState":0}'
```

Koenkk, 2021-07-07:
So did controlling according to https://www.zigbee2mqtt.io/devices/AV2010_29A.html#controlling not trigger the alarm? Don't worry about the `no converter available..` message for now.

snakuzzo, 2021-07-07:
Tried just now...

```shell
mosquitto_pub -t 'zigbee2mqtt-dev/0x00158d00054f1a28/set' -m '{"warning": {"level": "medium"}}'  -h 192.168.1.10
```

This is the log...

```
2021-07-07 20:13:27: Received MQTT message on 'zigbee2mqtt-dev/0x00158d00054f1a28/set' with data '{"warning": {"level": "medium"}}'
2021-07-07 20:13:27: Publishing 'set' 'warning' to '0x00158d00054f1a28'
```

...but no sound from siren

Koenkk, 2021-07-07:
Then try with `tz.on_off` and send to `zigbee2mqtt-dev/0x00158d00054f1a28/set` message `{"state": "ON"}`, if that also doesn't work you will need to sniff traffic with the original gateway to figure out how it triggers the warning

snakuzzo, 2021-07-07:
No success...

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Smart Siren'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '81868', // Vendor model number, look on the device for a model number
    vendor: 'AduroSmart', // Vendor of the device (only used for documentation and startup logging)
    description: 'ERIA smart siren', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.ias_no_alarm],
    toZigbee: [tz.on_off],
    exposes: []
}

module.exports = definition;
```

Tried the new publish...

```shell
mosquitto_pub -t 'zigbee2mqtt-dev/0x00158d00054f1a28/set' -m '{"state":"ON"}'  -h 192.168.1.10
```

...and this is the log...

```
2021-07-08 00:04:35: Received MQTT message on 'zigbee2mqtt-dev/0x00158d00054f1a28/set' with data '{"state":"ON"}'
2021-07-08 00:04:35: No converter available for 'state' (ON)
2021-07-08 00:04:51: Received Zigbee message from '0x00158d00054f1a28', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":1,"imageType":1027,"manufacturerCode":4653}' from endpoint 1 with groupID 0
```

TheStigh, 2021-08-10:
@snakuzzo Still stuck on this, or do you want help ?

TheStigh, 2021-08-11:
@snakuzzo 

Try this code:
```

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const utils = require('zigbee-herdsman-converters/lib/utils');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {

    ias_siren: {
        cluster: 'ssIasZone',
        type: 'commandStatusChangeNotification',
        convert: (model, msg, publish, options, meta) => {
            const zoneStatus = msg.data.zonestatus;
            return {
                alarm: (zoneStatus & 1) > 0,
                tamper: (zoneStatus & 1<<2) > 0,
                battery_low: (zoneStatus & 1<<3) > 0,
                supervision_reports: (zoneStatus & 1<<4) > 0,
                restore_reports: (zoneStatus & 1<<5) > 0,
                ac_status: (zoneStatus & 1<<7) > 0,
                test: (zoneStatus & 1<<8) > 0,
            };
        },
    },
}

const tzLocal = {
    
    warning: {
        key: ['warning'],
        convertSet: async (entity, key, value, meta) => {
            const mode = {'stop': 0, 'burglar': 1, 'fire': 2, 'emergency': 3, 'police_panic': 4, 'fire_panic': 5, 'emergency_panic': 6};
            const level = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};
            const strobeLevel = {'low': 0, 'medium': 1, 'high': 2, 'very_high': 3};
            const values = {
                mode: value.mode || 'police_panic',
                level: value.level || 'very_high',
                strobe: value.hasOwnProperty('strobe') ? value.strobe : false,
                duration: value.hasOwnProperty('duration') ? value.duration : 60,
                strobeDutyCycle: value.hasOwnProperty('strobe_duty_cycle') ? value.strobe_duty_cycle * 10 : 0,
                strobeLevel: value.hasOwnProperty('strobe_level') ? strobeLevel[value.strobe_level] : 1,
            };
            const info = (mode[values.mode]) + ((values.strobe ? 1 : 0) << 4) + (level[values.level] << 6);
            await entity.command(
                'ssIasWd',
                'startWarning',
                {startwarninginfo: info, warningduration: values.duration,
                    strobedutycycle: values.strobeDutyCycle, strobelevel: values.strobeLevel},
                utils.getOptions(meta.mapped, entity),
            );
        },
    },
}

module.exports = [
    {
        zigbeeModel: ['Smart Siren'],
        model: '81868',
        vendor: 'AduroSmart',
        description: 'ERIA smart siren', 
        fromZigbee: [fz.ias_enroll, fz.ias_wd, fzLocal.ias_siren],
        toZigbee: [tzLocal.warning, tz.ias_max_duration],
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic', 'ssIasZone', 'ssIasWd']);
            await endpoint.read('ssIasZone', ['zoneState', 'iasCieAddr', 'zoneId']);
            await endpoint.read('ssIasWd', ['maxDuration']);
        },
        exposes: [e.tamper(), e.warning()],
    },
];

```

snakuzzo, 2021-08-11:
Thank you @TheStigh.
I'm in vacation.
I will be able to test in a couple of weeks

closed 2021-10-02 by github-actions[bot]

jvanderzande, 2021-10-16:
> @jvanderzande I tried your JS and I confirm it pairs, but it doesn't pair correctly. It's just possible to change duration and mode

I know, as I stated ;) 
The siren works with and supports these options: mode/duration/strobe/strobe_duty_cycle  but level doesn't work.  
I am not familiar with the pair log info to be able to sort out what the issue is, but am happy the basics work for me.

KlavsKlavsen, 2022-01-31:
I am trying to get this siren to work with my home-assistant - but it won't even show up when I try to search for devices to pair with. there is a button on the back which I can "just reach" - when its plugged in, but I've tried holding it and everything I can think of - to get that green light to do anything but just "constant green" - and nothing happens and it doesn't show up.
Do you know how to get it into pairing mode? or if it cannot be found directly by a zigbee usb stick with home-assistant?

snakuzzo, 2022-01-31:
You just need to press for a couple of seconds (not a long press) with a small pin in the small  hole on the side

KlavsKlavsen, 2022-01-31:
Found it - thank you very much.

KlavsKlavsen, 2022-02-03:
I have added the device - and under configuration -> devices -> zigbee devices -> smart siren
I can see it has "configuration" - where there are settings such as siren_level, strobe, strobe_level and warning mode
But no matter what I change - nothing happens on actual siren :(
I also tried creating a scene which changed strobe to ON - but nothing happens..
How do I test it works? 
I've just updated to latest home-assistant version.

KlavsKlavsen, 2022-02-03:
I just got it to do something, using developer tools -> services -> "integrated 'turn-on'" of strobe .. Yay.. So it seems to work fine for me too.

KlavsKlavsen, 2022-03-08:
Did any of you get the siren working at anything but lowest siren sound level? I try to set the level to highest level, enable strobe - and then run the homeassistant "general home_assistant.turn_on".. which makes it sound alert for ~5-10 seconds - at lowest sound level - and nothing else - no matter what I do :(

KlavsKlavsen, 2022-03-08:
I tried using the homeassistant "siren" support - but that just says "unknown error" - when I try to call it: https://www.home-assistant.io/integrations/siren/

jvanderzande, 2022-03-14:
> Did any of you get the siren working at anything but lowest siren sound level? 

The siren is working fine but only has one level as I confirmed a couple of replies up: 

> I've checked with their support and the siren has a fix volume.

KlavsKlavsen, 2022-03-14:
wow.. they state it has 100+db siren - and the siren is NOT even close to that when I get it to fire :(

Heire, 2022-08-04:
Did anyone try to do more tests to have it integrated into Z2M?
Was hoping to use this one where I could activate the LED's without using the alarm. Most zigbee siren with LED are always in combination but would like to use it seperately.
Seems to be the same issue or just a matter of using another configuration i would hope.

jvanderzande, 2022-08-06:
Work fine here and you are able to just activate the RedLight by specifying Mode=Stop and setting the strobe and press apply  e.g.
[image]

snakuzzo, 2023-03-11:
Hi all.
Any news about PR to official support this device ?
To recap... Is [this](https://github.com/Koenkk/zigbee2mqtt/issues/7691#issuecomment-942725430) the correct js ?
@jvanderzande  did you create some PR ?

jvanderzande, 2023-04-22:
>  @jvanderzande did you create some PR ?

Sorry for the late reply but missed the notification: I haven't made any PR, just shared my experience here for others to use.

jvanderzande, 2024-11-06:
Just to share: 
I had to make some changes after I made an update to the latest version of zigbee2mqtt.
Changes made to **adurosmart_siren.js** :
- Made it work for latest releases by removing line "const extend = require('zigbee-herdsman-converters/lib/extend');"
- Added some config item to no show SoundLevel and StrobeLevel as those aren't supported by the device:
[image]

**adurosmart_siren.js** :
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['Smart Siren'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '81868', // Vendor model number, look on the device for a model number
    vendor: 'AduroSmart Eria', // Vendor of the device (only used for documentation and startup logging)
    description: 'Smart Siren', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.battery, fz.ias_wd, fz.ias_enroll, fz.ias_siren],
	toZigbee: [tz.warning_simple, tz.ias_max_duration, tz.warning],
	configure: async (device, coordinatorEndpoint, logger) => {
		const endpoint = device.getEndpoint(1);
		await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic', 'ssIasZone', 'ssIasWd']);
		await endpoint.read('ssIasZone', ['zoneState', 'iasCieAddr', 'zoneId']);
		await endpoint.read('ssIasWd', ['maxDuration']);
	},
	exposes: [e.battery_low(), e.tamper(), e.warning(),
		exposes.numeric('max_duration', ea.ALL).withUnit('s').withValueMin(0).withValueMax(600).withDescription('Duration of Siren'),
		exposes.binary('alarm', ea.SET, 'ON', 'OFF').withDescription('Manual start of siren')],
};

module.exports = definition;
```

**configuration.yaml** :
```
....
external_converters:
  - adurosmart_siren.js
...
```

snakuzzo, 2024-11-06:
Thank you @jvanderzande.
Can you create a PR ?

jvanderzande, 2024-11-07:
> > Thank you @jvanderzande. Can you create a PR ?
> 
> Feel free to do so when you know how that works. ;)

I ended up spending some time to see what is required to add this and made a PR.