Koenkk/zigbee2mqtt #7692: QS-Zigbee-D02-TRIAC-2C-L  (dimmer function not available)

RafaelDuncan, 2021-06-03:
## What happened
I just added the dimmer to Domiticz and it is recognized immediately by Zigbee2mqtt.
The device is recognized within Domoticz as a Light/Switch. and get 2 types of functions.
- Link Quality
- State (On/Off)

strangely enough, I do not see a dimmer function, while the device does have a dimming function.

## What did you expect to happen
I would expect that the device would also show a separate dimming function.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.19.0
Adapter hardware:  CC2652Rb
Adapter firmware version:

Koenkk, 2021-06-04:
Can you provide the data/database.db entry of this device?

RafaelDuncan, 2021-06-06:
hello @Koenkk, thanks for your quick response. I hereby send you the requested information. it's about ID9

{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0021cc3b2c","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null}
{"id":2,"type":"EndDevice","ieeeAddr":"0xbc33acfffe5273d3","nwkAddr":4971,"manufId":4619,"manufName":"_TZ1800_fcdjzz3s","powerSource":"Battery","modelId":"TY0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,1280,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TY0202","manufacturerName":"_TZ1800_fcdjzz3s","powerSource":3,"zclVersion":1,"appVersion":65,"stackVersion":2,"hwVersion":1,"dateCode":"2019.12.12"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0021cc3b2c","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc3b2c","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":2,"hwVersion":1,"dateCode":"2019.12.12","zclVersion":1,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1622964307668}
{"id":3,"type":"Router","ieeeAddr":"0x60a423fffe5e2870","nwkAddr":53436,"manufId":4098,"manufName":"_TZ3000_zmy1waw6","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_zmy1waw6","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc3b2c","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1622959354824}
{"id":4,"type":"Router","ieeeAddr":"0x842e14fffe3a4886","nwkAddr":29162,"manufId":4098,"manufName":"_TZ3000_g5xawfcq","powerSource":"Mains (single phase)","modelId":"TS0121","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,6,1794,2820],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0121","manufacturerName":"_TZ3000_g5xawfcq","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,110]}},"haElectricalMeasurement":{"attributes":{"acVoltageMultiplier":1,"acVoltageDivisor":1,"acCurrentMultiplier":1,"acCurrentDivisor":1000,"acPowerMultiplier":1,"acPowerDivisor":1,"rmsVoltage":230,"rmsCurrent":0,"activePower":0}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0001575efd","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b0001575efd","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1622964316528}
[... cut ...] utes":{"64512":255,"currentLevel":236}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc3b2c","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0021cc3b2c","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"appVersion":66,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-419401481},"lastSeen":1622755852036}

Koenkk, 2021-06-07:
This looks to be an error on the domoticz side, if you see the dimmer slider through the frontend (https://www.zigbee2mqtt.io/information/frontend.html) you should create an issue here instead: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues

RafaelDuncan, 2021-06-07:
good afternoon Ken,

 

thanks for your comment. I just created a ticket on the other page. thanks for thinking along.

 

Gr. Rafael

closed 2021-06-07 by Koenkk