Koenkk/zigbee2mqtt #7693: failed with error: 'SRSP - UTIL - ledControl after 6000ms'

acuadulza, 2021-06-03:
## What happened
- please note that the zigbee network works well.
- however, this morning I noticed that clicking on "permit join" I get  the error shown in the attached picture.
[image]

## Debug log:
[image]

## Fixing attempts made:
- restarted the coordinator
- unistalled and reinstalled zigbee2mqtt
- rebooted Home Assistant 
- added "disable_led: false" to the zigbee2Mqtt configuration as read in previous issues (I also tried with disable_led: true)

## Debug info
HASSIO Version core-2021.5.4
Zigbee2MQTT version 1.19.0 
Coordinator type  zStack12   (CC2530 + CC2592 )
Coordinator revision 20190608
Frontend version 0.4.16

## Network devices
coordinator zStack12   (CC2530 + CC2592)
3 x Aquara temperature/Humidity sensors WSDCGQ11LM
1 x Aquara PIR sensor RTCGQ11LM
1 x Aquara door sensor MCCGQ11LM
[image]

Fridolin34, 2021-06-04:
I don't know, where this is coming from, but occasionally I see the same error. What helps in my case is to shut down HA, unplug the coordination (I'm using cc2531 and CC1352P-2 in two different locations) for like 30 seconds and then restart everything.

acuadulza, 2021-06-04:
Unfortunately this does not work in my case. The zigbee network works well,
but I cannot add further devices....

ykiv63, 2021-06-05:
Hi. I've the same Issue with Cc2530

Koenkk, 2021-06-05:
Fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-06-05 by Koenkk