Koenkk/zigbee2mqtt #7695: Support for "Parkside Smart Watering Timer"

DerElch87, 2021-06-04:
Dear all,
I tried to create support for the Parkside Smart Watering Timer, but failed catastrophically :)

I created a .js file for Tuya like explained in the tutorial - but then Z2M will not start anymore.
I copied the content on the bottom of the message.
If anyone is willing to help, I'd highly appreciate.

Thanks and best regards

## Information about the device + link
Parkside Smart Watering Timer
https://www.lidl.de/de/parkside-smarter-bewaesserungscomputer-zigbee-smart-home/p375570

## PSBZS.js
_const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_c88teujp'
        },
    ],
    model: 'PSBZS A1',
    vendor: 'Lidl',
    description: 'Smart Watering Timer',
    supports: 'thermostat, temperature',
    fromZigbee: [
        fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
    ],
    onEvent: tuya.setTime, // Add this if you are getting no converter for 'commandSetTimeRequest'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
};

module.exports = definition;_

fluppie, 2021-06-04:
I also have this module :-) https://www.lidl-shop.be/nl-BE/PARKSIDE-Besproeiingscomputer-Smart-Home-PSBZS-A1/p100325201

[image]

```
Friendly name
    0x847127fffea96df6
Device type
    EndDevice
Zigbee model
    TS0601
Support status

    Unsupported
IEEE address
    0x847127fffea96df6
Network address
    0xFDB7
Power source
    Battery 
Interview completed
    Yes
```

cru8602, 2021-06-07:
I created a converter file that at least worked to switch the valve on and off:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'HG06875', // Vendor model number, look on the device for a model number
    vendor: 'PARKSIDE', // Vendor of the device (only used for documentation and startup logging)
    description: 'Smart Watering Timer', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off], // We will add this later
    toZigbee: [tz.on_off], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.switch()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
	configure: async (device, coordinatorEndpoint, logger) => {
    	const endpoint = device.getEndpoint(1);
    	await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;`

The valve doesn't seem to report back if it is switched or not and also it switches of autmaticly after about 60 sec. This time intervall can possibly be changed using the lidl gateway but I didn't found any messages that gave me a hint how to do this in z2m. I also tryed to integrate the battery state but this also doesn't seems to be reported.

TheOriginal92, 2021-06-08:
How did you manage to pair it? The Pressing the button for a few seconds until the left led is blinking does not seem to work for me. Im using an CC2652-P Stick.

fluppie, 2021-06-08:
Short press to "activate" the module and then long press untill the red led on the left starts blinking.
@cru8602 where did you store this file? Or did you add this somewhere in the web UI of Z2M?

TheOriginal92, 2021-06-08:
> Short press to "activate" the module and then long press untill the red led on the left starts blinking.
> @cru8602 where did you store this file? Or did you add this somewhere in the web UI of Z2M?

Thank you, did just that. Left led starts blinking but nothing happening in Z2M in debug log

stefangries, 2021-06-08:
> Thank you, did just that. Left led starts blinking but nothing happening in Z2M in debug log 
> 

You can try to reset the device by holding the button while inserting the battery.

cru8602, 2021-06-08:
@fluppie I named the file HG06875.js and added it to the data folder in which the configuration.ymal is located. 
After that I used the front-end to add "HG06875.js" in the external converters section.

Main-pinguin, 2021-06-09:
Seems to be a bit more complex ... I have setup a device via Tuya gateway with some intervals. This intervals still ongoing even Tuya gateway is out of range and device is paired with my testbed based on a CC2531. The GenOn seem to have the same behaviour as the manuall watering Button on the Device... This button activate the valve even for about 60 sec. Unfortunately I didn't get the Tuya datapoint debuging eorking on my test bed... However will continue to work on

BiasF, 2021-06-10:
same here, didn't get the debugging working.
The valve has only one button but doesn't send any messages via zigbee while pressing the button. 

Maybe thats why the datapoint debugging doesn't help....

enzzo-ux, 2021-06-13:
I also have a problem with starting. 
Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_htnnfasr'

digdas1, 2021-06-14:
Hi,

I got the device myself. I do see it with deconz, and also the switching off after 60 seconds. Can I help?
My setup:
Conbee 2 stick, with deconz (which sees it as a light)

BiasF, 2021-06-15:
I think it's not usefull to sniff traffic between Conbee and the watering timer

You have to sniff the traffic between the original LIDL Gateway and the watering timer, to see which commands the original gateway is sending to the device.

My Lidl Gateway should arrive today,

BiasF, 2021-06-16:
argh, I now have a LIDL Gateway and the watering timer.... but ZSHARK doesn't work with a Conbee II, only with Conbee I -,-

I also have a "Slaesh's CC2652RB stick" but it's my production network stick and I dont want to destroy my Zigbee network with flashing this stick. 

Any Ideas how to sniff with one of theese devices without destroing my Zigbee network ?

dasfuu, 2021-06-16:
Only a cheap CC2531 stick comes to mind. I have a generic Tuya Gateway and Watering Timer here.
Never sniffed zigbee before.  I will try in the next days too.

VictorvandenHoven, 2021-06-16:
@cru8602 Thanks for the converter file. At least I can use it as a switch now in Home Assistant.

fschaefer, 2021-06-17:
> I have setup a device via Tuya gateway
> I have a generic Tuya Gateway
@Main-pinguin @dasfuu 

Which gateways do you guys use?

Main-pinguin, 2021-06-17:
> 
> 
> > I have setup a device via Tuya gateway
> > I have a generic Tuya Gateway
> > @Main-pinguin @dasfuu
> 
> Which gateways do you guys use?

I didn't have any of this gateways.. I tested it at riend of mine on Lidl gateway... When I visit my friend next time I will try to sniff the zigbee traffic between GW and valve  ... But could't promise any timeline...

dasfuu, 2021-06-17:
> > I have setup a device via Tuya gateway
> > I have a generic Tuya Gateway
> > @Main-pinguin @dasfuu
> 
> Which gateways do you guys use?

I have this one: https://www.amazon.de/gp/product/B08PF4ZYNY/
Valve works perfectly with all features.

BiasF, 2021-06-20:
Now I am able to sniff with a CC2531 between the original LIDL Gateway and the Smart Watering Timer.

Here is a example PCAP file from WireShark, if this is usefull for implementing the device in Zigbee2Mqtt I can provide more capture files for specific actions :)

[LIDL-GW-Watering.txt](https://github.com/Koenkk/zigbee2mqtt/files/6682585/LIDL-GW-Watering.txt)

VentiloFred, 2021-06-20:
> argh, I now have a LIDL Gateway and the watering timer.... but ZSHARK doesn't work with a Conbee II, only with Conbee I -,-
>
> I also have a "Slaesh's CC2652RB stick" but it's my production network [...]

Hello, work for me with Conbee II

Home Assistant version core-2021.6.4
deCONZ version 6.8.0

With HA i can switch On, no feedback
With Node Red I can switch On / Off but automatically Off after 60s!

VentiloFred, 2021-06-20:
For this moment i use this solution:
I send every 10sec (for this moment) "true" with Node Red to the Parkside valve until when i want to stop the valve

[image]

Main-pinguin, 2021-06-21:
I have as well some taces from Wireshark. As they are from an in-production network I will them not share all. As example
[Button pressed On_F.txt](https://github.com/Koenkk/zigbee2mqtt/files/6686574/Button.pressed.On_F.txt)
[Set 10min and on via GW.txt](https://github.com/Koenkk/zigbee2mqtt/files/6686575/Set.10min.and.on.via.GW.txt)
 Button pressed On_F.txt (need to be renamed is the trace when button on valve is presse twice top open valve
Set 10min and on via GW.txt is the trace of data exchanged between GW and valve if in application a 10 min watering time is set and the valve is activated.

MattWestb, 2021-06-22:
For looking inside the data the current network key is needed or only getting "data" and no information.

BiasF, 2021-06-22:
Hi,

for my testing network the key is 9e7d5dd06bd96b5479fe631cc230c9d5

MattWestb, 2021-06-22:
Sorry not working ;-(((

BiasF, 2021-06-22:
Did you add the TrustCenter key?

`Add the Trust Center link key by going to to Edit -> Preferences -> Protocols -> ZigBee. Set Security Level to AES-128 Encryption, 32-bit Integrity Protection and click on Edit. Click on + and add 5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39 with Byte Order Normal.`

MattWestb, 2021-06-22:
Looks witch key its being in use in the `ZigBee Security Header`:
```
ZigBee Network Layer Command, Dst: Broadcast, Src: 0x0000
    Frame Control Field: 0x1209, Frame Type: Command, Discover Route: Suppress, Security, Extended Source Command
    Destination: 0xfffc
    Source: 0x0000
    Radius: 1
    Sequence Number: 213
    Extended Source: SiliconL_ff:fe:cc:6c:ea (58:8e:81:ff:fe:cc:6c:ea)
    ZigBee Security Header
        Security Control Field: 0x28, Key Id: Network Key, Extended Nonce
        Frame Counter: 136740
        Extended Source: SiliconL_ff:fe:cc:6c:ea (58:8e:81:ff:fe:cc:6c:ea)
        Key Sequence Number: 0
        Message Integrity Code: 0dbb429c
        [Key: -x-x-x-x-x-x]
        [Key Label: LIDL ZBGW]
    Command Frame: Link Status
```
`Key` is "patched".

I have 8 active working network keys plus the "normal master" keys in the table.

BiasF, 2021-06-22:
Mhhh should be the right key

[image]

MattWestb, 2021-06-22:
With:
[image]

Is getting:
[image]

BiasF, 2021-06-22:
you opened not my pcap file ;) it's the file from Main-pinguin

MattWestb, 2021-06-22:
110% true !!!
:-)))

And is with one paring  so the transport of the network key is inside the sniff :-))

MattWestb, 2021-06-22:
Its looks like its setting up the device with one command for using tuya commands:
```
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
    Frame Control Field: Data (0x00)
    Destination Endpoint: 1
    Cluster: Basic (0x0000)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 1
ZigBee Cluster Library Frame, Command: Write Attributes, Seq: 1
    Frame Control Field: Profile-wide (0x10)
    Sequence Number: 1
    Command: Write Attributes (0x02)
    Attribute Field, Uint8: 13
        Attribute: Unknown (0xffde)
        Data Type: 8-Bit Unsigned Integer (0x20)
        Uint8: 13 (0x0d)
```
All the rest is "normal tuya commands" and no Zigbee standard commands more then the handshaking with default response for the no ZCL commands.

MattWestb, 2021-06-22:
@BiasF do you have one sniff then sending one "on command" from the tuya ZBGW ?
It can its only sending tuya commands and not normal zigbee "on" that we have being using but is not getting the status reported back if the device is on or off.

Main-pinguin, 2021-06-22:
> Its looks like its setting up the device with one command for using tuya commands:
>
> ```
> ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
>     Frame Control Field: Data (0x00) [...]

That's even my assumption here a cut out of trace "button on valve pressed for open valve" 

Frame 2: 12 bytes on wire (96 bits), 10 bytes captured (80 bits) on interface \\.\pipe\zboss_sniffer_COM5, id 0
IEEE 802.15.4 Command, Dst: 0x2ebe, Src: 0x053e

No.     Time           Source                Destination           Protocol Length Info                                                            
      3 8.499833       0x053e                0x0000                ZigBee HA 58     Unknown Command: 0x02, Seq: 98                                  

Frame 3: 58 bytes on wire (464 bits), 56 bytes captured (448 bits) on interface \\.\pipe\zboss_sniffer_COM5, id 0
IEEE 802.15.4 Data, Dst: 0x2ebe, Src: 0x053e
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x053e
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x09)
    Sequence Number: 98
    Command: _**Unknown (0x02)**_
Data (10 bytes)

0000  00 c7 06 02 00 04 00 00 00 3c                     .........<
    Data: 00c7060200040000003c
    [Length: 10]

No.     Time           Source                Destination           Protocol Length Info                                                            
      4 8.600815       0x053e                0x2ebe                IEEE 802.15.4 12     Data Request                                                    

Frame 4: 12 bytes on wire (96 bits), 10 bytes captured (80 bits) on interface \\.\pipe\zboss_sniffer_COM5, id 0
IEEE 802.15.4 Command, Dst: 0x2ebe, Src: 0x053e

No.     Time           Source                Destination           Protocol Length Info                                                            
      5 8.603538       0x0000                0x053e                ZigBee HA 50     ZCL: Default Response, Seq: 98                                  

Frame 5: 50 bytes on wire (400 bits), 48 bytes captured (384 bits) on interface \\.\pipe\zboss_sniffer_COM5, id 0
IEEE 802.15.4 Data, Dst: 0x053e, Src: 0x2ebe
ZigBee Network Layer Data, Dst: 0x053e, Src: 0x0000
ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
ZigBee Cluster Library Frame, Command: Default Response, Seq: 98
    Frame Control Field: Profile-wide (0x00)
    Sequence Number: 98
    Command: Default Response (0x0b)
    Response to Command: 0x02
    Status: Success (0x00)

BiasF, 2021-06-22:
@MattWestb 
Here comes the TUYA_ON Sniff

[TUYA_ON.txt](https://github.com/Koenkk/zigbee2mqtt/files/6693282/TUYA_ON.txt)

MattWestb, 2021-06-22:
If taking all normal handshaking away and only looking of the strange data being sent is like this:
```
Source > Destination "tuya data" (Sequence number)
0x0000 > 0xb6c3 00b6050200040000000f0101000101 (65)
0xb6c3 > 0x0000 00b7060200040000000f (8)
0xb6c3 > 0x0000 00b80101000101050200040000000f (9)
0xb6c3 > 0x0000 00b90b0200040000005a (10)
0xb6c3 > 0x0000 00ba060200040000000e (11)
```
I think its "normal tuya commands" one to the device and the device is reporting back attribute changes but i think some other users can interpreting it better then i have no deep experience of tuyas protocol.
The good thing is that the device is looking reporting much data back so its very likely getting real status from it :-)))

mgrom, 2021-06-22:
Hey guys, I've just started decoding messages that are being send between gateway and the device. I've found that the you can switch on the device but still there is auto off defined (default 1minute). 
Please be patient :D work in progress, but looks promissing.

MattWestb, 2021-06-22:
The 1 minute is one safety future and can not (very likely) being overridden. 
The automation is easy to making in HA only triggering one on command every 55 seconds as long the water period is active.

The the status is very interesting to getting working :-))

dasfuu, 2021-06-22:
> The 1 minute is one safety future and can not (very likely) being overridden.
That status can be overriden. In the SmartLife App you can specifiy that intervall/timeout. It even stays like that if you then pair it with another network.

Also you can store rules on the device to enable it on certain days and times, with a given interval and pauses.

MattWestb, 2021-06-22:
@Smanar have start diving in the tuya commands so its start moving :-)))

mgrom, 2021-06-23:
is it only me or the device connected to zigbee2mqtt is not sending any "Zigbee HA" messages? When connected to Tuya Gateway I can sniff all messages and it works great, but when I want to try it on zigbee2mqtt there are no proper messages.

mgrom, 2021-06-23:
@MattWestb did you manage to sniff any messages between zigbee2mqtt dongle and the device? I can sniff messages between Tuya GW and the device but the only message that the device is sending when connected to zigbee2mqtt dongle is request for local time, after that nothing...

divemasterjm, 2021-06-23:
> @fluppie I named the file HG06875.js and added it to the data folder in which the configuration.yaml is located.
> After that I used the front-end to add "HG06875.js" in the external converters section.

i tried but when z2m starts it throws an error and keeps restarting

MattWestb, 2021-06-23:
The interesting is wot the tuya ZBGW is doing with  the device then pairing and also then sending commands and changing the configuration like adding scheduled and extending the on time parameters.

For the moment Z2M is treating it as one "normal zigbee device" so no magic is being made.
Need waiting for @Smanar and other devs is finding and can implanting it in the system for getting it working OK.

mgrom, 2021-06-23:
@Koenkk sorry for disturbing :) but I have some trouble with this device. 
What I've done so far:
- sniffed messages between the device and Tuya Gateway
- device is working as it should together with TGW
- paired the device with zigbee2mqtt coordinator

Problem:
Device when linked to TGW is sending commands:
[image]
but when linked to zigbee2mqtt there is radio silence.... so there is no possible way to use tuya_data_point_dump because there are no messages.

Where to look for answers?

MattWestb, 2021-06-23:
You need sending the initiating command from the sniffed https://github.com/Koenkk/zigbee2mqtt/issues/7695#issuecomment-865867667 the first command is from the coordinator to the device.

mgrom, 2021-06-23:
I don't think so. First message with some tuya-like data is sent by the device. The only message that is being sent before this is coordinators command:
[image]
But I've already tried to send it, and with no changes...

Smanar, 2021-06-23:
There is too a write attribute in Basic Cluster, attribute 0xffde, value 0x0D.
You realy don't have "tuya request" at all ?

And the device ask for gateway node descriptor (request 188), that is not realy good, it s the way used by Xiaomi to check device compatibility, some recents devices have same reaction, like the Aqara S1.
To bypass this check (the conbee have not correct stack), an user include the device on the orignal gateway, use the same MAC adress and network key on the conbee, turn off the original gateway, and turn on the conbee, like that reporting is working again. The check is only done during inclusion.

mgrom, 2021-06-23:
It looks like you described:
[image]

Smanar, 2021-06-23:
Harf, we are not using the same sniff.
On this one https://github.com/Koenkk/zigbee2mqtt/issues/7695#issuecomment-864570013

After the gateway ask for the attribute 0xffde and before the first tuya request, you have the device that ask for gateway node descriptor (and not the reverse)

mgrom, 2021-06-23:
I'm going to reflash dongle and start over :)
Maybe it's something wrong with mine configuration

mgrom, 2021-06-23:
> Its looks like its setting up the device with one command for using tuya commands:
>
> ```
> ZigBee Application Support Layer Data, Dst Endpt: 1, Src Endpt: 1
>     Frame Control Field: Data (0x00) [...]

I had the same feeling that this command is important. But I'm not able to crack it. Value of the attribute in my case was 19 - sure I tried to replicate it but with no luck... I've tried to use different values, no luck either.

Smanar, 2021-06-23:
On my sniff it s 13 (0x0D). Time based ?
No answer yet for tests using the conbee on my side. But I realy think it can come from answer from the gateway to the node descriptor request.

mgrom, 2021-06-24:
I don't think it's time based. No matter how many times I'm pairing the device with GW, every time this attribute is equal to 19.

mgrom, 2021-06-24:
OK, I can confirm two things:
DP: 0x01 - value:0 => OFF, value:1 => ON
DP: 0x05 - with raw data [0,0,0,1] sets auto off timer to 1min, [0,0,02] to two minutes etc.

Still no luck with device reporting it's own status...

Big4SMK, 2021-06-25:
That's very encouraging news. Being able to set the timer would be really helpful over sending continues "on" commands like w do right now. Great work!

divemasterjm, 2021-06-25:
Today I will receive mine, can someone help me to add it to z2m?

liselotte111, 2021-06-25:
Sorry, I know this is not the right place to ask. But I would love to try it out. If run supervised HA and I've added a .js file to the /share/zigbee2mqtt folder with the converter suggested by @cru8602. When I add it to the config.yaml I get an error saying:
Cannot find module 'zigbee-herdsman-converters/lib/extend

Could someone point me in the right direction?

VictorvandenHoven, 2021-06-25:
Did you add the extension as well?
You have to use the "extension" button in the zigbee2mqtt to add the .js file as an extension.

[image]

liselotte111, 2021-06-25:
Thank you Victor, but I tried that. It appears that I was running the add-on from the old repo. I will first try to switch to the new one.

[235 more replies]

closed 2022-11-21 by github-actions[bot]