Koenkk/zigbee2mqtt #7696: Home Assistant 2021.6 breaks zigbee2mqtt cover devices

middelink, 2021-06-04:
## What happened
Upgraded to 2021.6.1 from 2021.5.5

At startup home assistant logs show:
```
2021-06-04 11:46:59 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_cover_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/L1_Livingroom_Curtain_Left/availability'}], 'command_topic': 'zigbee2mqtt/L1_Livingroom_Curtain_Left/set', 'device': {'identifiers': ['zigbee2mqtt_0x04cf8cdf3c814dba'], 'manufacturer': 'Xiaomi', 'model': 'Aqara B1 curtain motor  (ZNCLDJ12LM)', 'name': 'L1_Livingroom_Curtain_Left', 'sw_version': 'Zigbee2MQTT 1.18.1'}, 'json_attributes_topic': 'zigbee2mqtt/L1_Livingroom_Curtain_Left', 'name': 'L1_Livingroom_Curtain_Left', 'position_topic': 'zigbee2mqtt/L1_Livingroom_Curtain_Left', 'set_position_template': '{ "position": {{ position }} }', 'set_position_topic': 'zigbee2mqtt/L1_Livingroom_Curtain_Left/set', 'unique_id': '0x04cf8cdf3c814dba_cover_zigbee2mqtt', 'value_template': '{{ value_json.position }}', 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable
    return schema(data)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/cover.py", line 118, in validate_options
    raise vol.Invalid(
voluptuous.error.Invalid: 'value_template' must be set together with 'state_topic'.
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/mixins.py", line 160, in async_discover
    config = schema(discovery_payload)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 218, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 341, in _exec
    raise e if self.msg is None else AllInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 337, in _exec
    v = func(v)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 276, in __call__
    raise er.MultipleInvalid([e])
voluptuous.error.MultipleInvalid: 'value_template' must be set together with 'state_topic'.
```
for all three of my blinds/covers

checking documentation on HA mqtt discovery on [covers](https://www.home-assistant.io/integrations/cover.mqtt/), it indeed says 
```
value_template string (optional)
Defines a template that can be used to extract the payload for the state_topic topic.
```
It seems one needs to use a state_topic instead of json_sttribute_topic now?
However, that might be a recent change :)

## What did you expect to happen
Cover devices to work

## How to reproduce it (minimal and precise)
Upgrade to 2021.6.1 is sufficient.

## Debug info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: CC26X2R1
Adapter firmware version: no idea, is not shown in the logs.

middelink, 2021-06-04:
Addedum: reverting back to 2021.5.5 fixes the problem.

Koenkk, 2021-06-04:
Dupe of https://github.com/Koenkk/zigbee2mqtt/issues/7680 (fixed in 1.19.0+)

closed 2021-06-04 by Koenkk