Koenkk/zigbee2mqtt #7700: Tuya smart metering plug 'TS011F' '_TZ3000_cphmq0q7' not reporting / readings incorrect

oetelaar24, 2021-06-04:
## What happened
I was using 1.18.3 with a custom external converter for this, with polling at 5 sec. All was working great.
code: https://github.com/Koenkk/zigbee-herdsman-converters/issues/2557

Updated to 1.19.1, disabling the custom external converter as support was added now (see: https://github.com/Koenkk/zigbee-herdsman-converters/pull/2587)

However, even after reconfiguring or removing the device and rejoining them. No readings are reported after the initial. 
also the current reading is incorrect and the energy reading stays null:
[image]

Hitting the refresh button does seem to trigger new data, however the data seems stale:
[image]

## What did you expect to happen
All the data to be correct, as I had it in 1.18.3 with the custom external converter with polling enabled:
[image]

updating every 5sec:
[image]

## How to reproduce it (minimal and precise)

1. Have a TuYa TS011F _TZ3000_cphmq0q7  smart plug.
2. Use custom external converter from: https://github.com/Koenkk/zigbee-herdsman-converters/issues/2557 with polling at 5 sec configured
3. upgrade to 1.19.1 and remove custom external converter.

## Debug info
Zigbee2MQTT version: 1.19.1
Adapter hardware: CC2650R
Adapter firmware version: 20210120

oetelaar24, 2021-06-04:
Logging from 1.18.3 with custom external converter:
`Zigbee2MQTT:info  2021-06-04 12:31:29: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":36.52,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":242}'
,Zigbee2MQTT:info  2021-06-04 12:31:24: Configuring 'MeterkastUPSPwrMtr'
,Zigbee2MQTT:info  2021-06-04 12:31:24: Successfully configured 'MeterkastUPSPwrMtr'
,Zigbee2MQTT:info  2021-06-04 12:31:34: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":36.52,"linkquality":177,"power":73,"power_outage_memory":"on","state":"ON","voltage":242}'
,Zigbee2MQTT:info  2021-06-04 12:31:39: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":36.52,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":242}'
,Zigbee2MQTT:info  2021-06-04 12:31:44: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":36.52,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":242}'
,Zigbee2MQTT:info  2021-06-04 12:31:49: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":36.52,"linkquality":177,"power":73,"power_outage_memory":"on","state":"ON","voltage":242}'
,Zigbee2MQTT:info  2021-06-04 12:31:54: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.4,"energy":36.52,"linkquality":180,"power":73,"power_outage_memory":"on","state":"ON","voltage":242}'
`

and with 1.19.1, without customs converter:
`Zigbee2MQTT:info  2021-06-04 12:16:01: Configuring 'MeterkastUPSPwrMtr'
,Zigbee2MQTT:info  2021-06-04 12:16:01: Successfully configured 'MeterkastUPSPwrMtr'
,Zigbee2MQTT:info  2021-06-04 12:16:02: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"MeterkastUPSPwrMtr"},"status":"ok","transaction":"df13s-1"}'
,Zigbee2MQTT:info  2021-06-04 12:16:25: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":36.51,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":241}'
,Zigbee2MQTT:info  2021-06-04 12:24:44: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":36.52,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":241}'
`

Koenkk, 2021-06-04:
Can you provide the data/database.db entry of your device?

oetelaar24, 2021-06-04:
Hi Koen,

Sure, this is the 1.18.3 version entry:

`{"id":51,"type":"Router","ieeeAddr":"0x847127fffe8db1c4","nwkAddr":50077,"manufId":4098,"manufName":"_TZ3000_cphmq0q7","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,1794,2820,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_cphmq0q7","powerSource":1,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"rmsCurrent":383,"activePower":71,"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":234,"acVoltageMultiplier":1,"acVoltageDivisor":1,"acPowerMultiplier":1,"acPowerDivisor":1}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":[0,3724]}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b00215ff75d","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x00124b00215ff75d","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b00215ff75d","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":[1,1]}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1622845877441}
`

I can upgrade to 1.19.1 again and see if anything has changed. Would you like me to?

Koenkk, 2021-06-05:
@oetelaar24 yes, I would expect it works correctly, in the database you see `"acCurrentDivisor":1000,"acCurrentMultiplier":1`, this means the current will be divided by 1000 which should yield the current in amperes.

McGiverGim, 2021-06-07:
I have the same device (or similar, because with Tuya there is a lot of clones), it is identified as _TZ3000_cphmq0q7 TS011F, and it works ok for me.
I never had an external converter, y used the edge version and moved to the stable when released. Maybe this makes some difference? 
Did you do a "force" remove or only a remove?

oetelaar24, 2021-06-07:
It's a bit strange. Let me explain:

So the device is: 0x847127fffe8db1c4 aka MeterkastUPSPwrMtr

On 1.18.3, with the custom stuff, and reporting on every 5sec 
```
  '0x847127fffe8db1c4':
    friendly_name: MeterkastUPSPwrMtr
    retrieve_state: true
    measurement_poll_interval: 5

```
it seems happy:
```
,Zigbee2MQTT:info  2021-06-07 07:40:04: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":241}'
,Zigbee2MQTT:info  2021-06-07 07:40:04: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":241}'
,Zigbee2MQTT:info  2021-06-07 07:40:09: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":241}'
,Zigbee2MQTT:info  2021-06-07 07:40:09: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":241}'
,Zigbee2MQTT:info  2021-06-07 07:40:14: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":241}'
,Zigbee2MQTT:info  2021-06-07 07:40:14: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":241}'
,Zigbee2MQTT:info  2021-06-07 07:40:19: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":73,"power_outage_memory":"on","state":"ON","voltage":241}'
,Zigbee2MQTT:info  2021-06-07 07:40:19: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":73,"power_outage_memory":"on","state":"ON","voltage":241}'
,Zigbee2MQTT:info  2021-06-07 07:40:24: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":73,"power_outage_memory":"on","state":"ON","voltage":241}'
,Zigbee2MQTT:info  2021-06-07 07:40:24: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":73,"power_outage_memory":"on","state":"ON","voltage":241}'
,Zigbee2MQTT:info  2021-06-07 07:40:29: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":73,"power_outage_memory":"on","state":"ON","voltage":241}'
,Zigbee2MQTT:info  2021-06-07 07:40:29: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":180,"power":73,"power_outage_memory":"on","state":"ON","voltage":241}'
,Zigbee2MQTT:info  2021-06-07 07:40:34: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":238}'
,Zigbee2MQTT:info  2021-06-07 07:40:34: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":238}'
,Zigbee2MQTT:info  2021-06-07 07:40:39: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.39,"energy":41.33,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":238}'
,Zigbee2MQTT:info  2021-06-07 07:40:39: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.39,"energy":41.33,"linkquality":180,"power":71,"power_outage_memory":"on","state":"ON","voltage":238}'
,Zigbee2MQTT:info  2021-06-07 07:40:44: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":238}'
,Zigbee2MQTT:info  2021-06-07 07:40:44: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":238}'
,Zigbee2MQTT:info  2021-06-07 07:40:49: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":238}'
,Zigbee2MQTT:info  2021-06-07 07:40:49: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":238}'
,Zigbee2MQTT:info  2021-06-07 07:40:54: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":238}'
,Zigbee2MQTT:info  2021-06-07 07:40:54: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.38,"energy":41.33,"linkquality":177,"power":71,"power_outage_memory":"on","state":"ON","voltage":238}'
,Zigbee2MQTT:info  2021-06-07 07:40:59: MQTT publish: topic 'zigbee2mqtt/MeterkastUPSPwrMtr', payload '{"current":0.4,"energy":41.33,"linkquality":177,"power":73,"power_outage_memory":"on","state":"ON","voltage":239}'
[... cut ...] tackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":680236614},"lastSeen":1623055246281}
```

It does seem to work okay now. I'm not sure why all of a sudden the configuration for the device got updated and corrected the current reading?
Also, the updates from the device seem to be very infrequent, I liked being able to get it to report every 5 secs, however, maybe this is just how it needs to be? So it only reports when a reading has changed?

McGiverGim, 2021-06-07:
You can configure when it must update the state, in configuration or in the UI:

[image]

If I'm not wrong, you can change the `min rep interval` (min time between updates) `max rep interval` (max time between updates) or the `min rep change` (minimum change in the value to update) if you want more updates.

oetelaar24, 2021-06-07:
Aah, yes, that works well. I didn't know this already worked. Every time I try to change a value, the value immediatly seems to be changed back (like a constant refresh in the background?) and it was very tricky to get it to save my changes.

But it worked. I now get updates more frequent, and setting a min rep change is pretty awesome!

oetelaar24, 2021-06-14:
Is this ability to change the reporting thresholds going to come to every device that supports reporting?
I see only a handfull of devices with this table filled in. Ofcourse I can try to replicate these settings for a different smart metering plug, but I don't know if that would work?

closed 2021-07-23 by github-actions[bot]