Koenkk/zigbee2mqtt #7703: HGZB-07A (LXT56-LS27LX1.7) now recognised as 3A12S-15

ozadr1an, 2021-06-05:
## What happened
Since update to zigbee2mqtt@1.19.1-dev or zigbee2mqtt@1.19.1 my HZGB-07A RGBW Downlight is now recognised as 3A12S-15 Lightstrip Controller
## What did you expect to happen
It to remain as per previous releases (worked ok in 1.18.1)

## How to reproduce it (minimal and precise)
Install either 1.19.1 or 1.19.1-dev

## Debug info
Zigbee2MQTT version: 1.19.1-dev
Adapter hardware: CC1352P-2
Adapter firmware version: 20210120

ozadr1an, 2021-06-05:
May be in relation to fix for https://github.com/Koenkk/zigbee2mqtt/issues/7292

ozadr1an, 2021-06-05:
For reference, debug output when joining the device on 1.19.1-dev. 

info  2021-06-05 16:10:05: Device '0x00158d0004774cc0' joined
info  2021-06-05 16:10:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0004774cc0","ieee_address":"0x00158d0004774cc0"},"type":"device_joined"}'
info  2021-06-05 16:10:05: Starting interview of '0x00158d0004774cc0'
info  2021-06-05 16:10:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0004774cc0","ieee_address":"0x00158d0004774cc0","status":"started"},"type":"device_interview"}'
info  2021-06-05 16:10:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00158d0004774cc0"},"type":"device_connected"}'
info  2021-06-05 16:10:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00158d0004774cc0"},"type":"pairing"}'
debug 2021-06-05 16:10:05: Device '0x00158d0004774cc0' announced itself
info  2021-06-05 16:10:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0004774cc0","ieee_address":"0x00158d0004774cc0"},"type":"device_announce"}'
info  2021-06-05 16:10:05: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d0004774cc0"},"type":"device_announced"}'
debug 2021-06-05 16:10:05: Received Zigbee message from '0x00158d0004774cc0', type 'readResponse', cluster 'genBasic', data '{"modelId":"LXT56-LS27LX1.7"}' from endpoint 1 with groupID 0
debug 2021-06-05 16:10:06: Received Zigbee message from '0x00158d0004774cc0', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"3A Smart Home DE"}' from endpoint 1 with groupID 0
debug 2021-06-05 16:10:06: Received Zigbee message from '0x00158d0004774cc0', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
debug 2021-06-05 16:10:06: Received Zigbee message from '0x00158d0004774cc0', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2021-06-05 16:10:06: Received Zigbee message from '0x00158d0004774cc0', type 'readResponse', cluster 'genBasic', data '{"appVersion":1}' from endpoint 1 with groupID 0
debug 2021-06-05 16:10:06: Received Zigbee message from '0x00158d0004774cc0', type 'readResponse', cluster 'genBasic', data '{"stackVersion":1}' from endpoint 1 with groupID 0
debug 2021-06-05 16:10:06: Received Zigbee message from '0x00158d0004774cc0', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2021-06-05 16:10:06: Received Zigbee message from '0x00158d0004774cc0', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20190604"}' from endpoint 1 with groupID 0
debug 2021-06-05 16:10:06: Received Zigbee message from '0x00158d0004774cc0', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"1000-0001"}' from endpoint 1 with groupID 0
info  2021-06-05 16:10:06: Successfully interviewed '0x00158d0004774cc0', device has successfully been paired
info  2021-06-05 16:10:06: Device '0x00158d0004774cc0' is supported, identified as: Nue / 3A Smart Zigbee 3.0 strip light controller (3A12S-15)
info  2021-06-05 16:10:06: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart Zigbee 3.0 strip light controller","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0},{"access":7,"description":"Color temperature of this light","name":"color_temp","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500}],"property":"color_temp","type":"numeric","unit":"mired","value_max":500,"value_min":150},{"access":7,"description":"Color temperature after cold power on of this light","name":"color_temp_startup","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500},{"description":"Restore previous color_temp on cold power on","name":"previous","value":65535}],"property":"color_temp_startup","type":"numeric","unit":"mired","value_max":500,"value_min":150},{"description":"Color of this light in the CIE 1931 color space (x/y)","features":[{"access":7,"name":"x","property":"x","type":"numeric"},{"access":7,"name":"y","property":"y","type":"numeric"}],"name":"color_xy","property":"color","type":"composite"}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","name":"effect","property":" [... cut ...] alse,"vendor":"Nue / 3A"},"friendly_name":"0x00158d0004774cc0","ieee_address":"0x00158d0004774cc0","status":"successful","supported":true},"type":"device_interview"}'
info  2021-06-05 16:10:06: Configuring '0x00158d0004774cc0'
debug 2021-06-05 16:10:06: Setup reporting for '0x00158d0004774cc0' - 1 - genOnOff
debug 2021-06-05 16:10:06: Received Zigbee message from '0x00158d0004774cc0', type 'readResponse', cluster 'lightingColorCtrl', data '{"colorCapabilities":31}' from endpoint 1 with groupID 0

ozadr1an, 2021-06-05:
And debug output from 1.18.1:

debug 2021-06-05 17:54:10: Device 'Bathroom Downlight 1' announced itself
info  2021-06-05 17:54:10: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Bathroom Downlight 1","ieee_address":"0x00158d0004768d5a"},"type":"device_announce"}'
info  2021-06-05 17:54:10: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Bathroom Downlight 1"},"type":"device_announced"}'
info  2021-06-05 17:54:11: Successfully setup reporting for '0x00158d0004774cc0' - 1 - lightingColorCtrl
info  2021-06-05 17:54:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"RGBW Downlight","friendly_name":"0x00158d0004774cc0","model":"HGZB-07A","supported":true,"vendor":"Smart Home Pty"},"type":"pairing"}'
debug 2021-06-05 17:54:11: Received Zigbee message from 'Bathroom Downlight 1', type 'readResponse', cluster 'genBasic', data '{"modelId":"LXT56-LS27LX1.7"}' from endpoint 1 with groupID 0
debug 2021-06-05 17:54:11: Received Zigbee message from 'Bathroom Downlight 1', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"3A Smart Home DE"}' from endpoint 1 with groupID 0
debug 2021-06-05 17:54:11: Received Zigbee message from 'Bathroom Downlight 1', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
debug 2021-06-05 17:54:11: Received Zigbee message from 'Bathroom Downlight 1', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2021-06-05 17:54:11: Received Zigbee message from 'Bathroom Downlight 1', type 'readResponse', cluster 'genBasic', data '{"appVersion":1}' from endpoint 1 with groupID 0
debug 2021-06-05 17:54:11: Received Zigbee message from 'Bathroom Downlight 1', type 'readResponse', cluster 'genBasic', data '{"stackVersion":1}' from endpoint 1 with groupID 0
debug 2021-06-05 17:54:12: Received Zigbee message from 'Bathroom Downlight 1', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2021-06-05 17:54:12: Received Zigbee message from 'Bathroom Downlight 1', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20190604"}' from endpoint 1 with groupID 0
debug 2021-06-05 17:54:12: Received Zigbee message from 'Bathroom Downlight 1', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"1000-0001"}' from endpoint 1 with groupID 0
info  2021-06-05 17:54:12: Successfully interviewed 'Bathroom Downlight 1', device has successfully been paired
info  2021-06-05 17:54:12: Device 'Bathroom Downlight 1' is supported, identified as: Smart Home Pty RGBW Downlight (HGZB-07A)
info  2021-06-05 17:54:12: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"RGBW Downlight","exposes":[{"features":[{"access":7,"description":"On/off state of this light","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"},{"access":7,"description":"Brightness of this light","name":"brightness","property":"brightness","type":"numeric","value_max":254,"value_min":0},{"access":7,"description":"Color temperature of this light","name":"color_temp","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500}],"property":"color_temp","type":"numeric","unit":"mired","value_max":500,"value_min":150},{"access":7,"description":"Color temperature after cold power on of this light","name":"color_temp_startup","presets":[{"description":"Coolest temperature supported","name":"coolest","value":150},{"description":"Cool temperature (250 mireds / 4000 Kelvin)","name":"cool","value":250},{"description":"Neutral temperature (370 mireds / 2700 Kelvin)","name":"neutral","value":370},{"description":"Warm temperature (454 mireds / 2200 Kelvin)","name":"warm","value":454},{"description":"Warmest temperature supported","name":"warmest","value":500},{"description":"Restore previous color_temp on cold power on","name":"previous","value":65535}],"property":"color_temp_startup","type":"numeric","unit":"mired","value_max":500,"value_min":150},{"description":"Color of this light in the CIE 1931 color space (x/y)","features":[{"access":7,"name":"x","property":"x","type":"numeric"},{"access":7,"name":"y","property":"y","type":"numeric"}],"name":"color_xy","property":"color","type":"composite"}],"type":"light"},{"access":2,"description":"Triggers an effect on the light (e.g. make light blink for a few seconds)","name":"effect","property":"effect","type":"enum","values":["blink","breathe","okay","channel_change","finish_effect","stop_effect"]},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"HGZB-07A","supports_ota":false,"vendor":"Smart Home Pty"},"friendly_name":"Bathroom Downlight 1","ieee_address":"0x00158d0004768d5a","status":"successful","supported":true},"type":"device_interview"}'
info  2021-06-05 17:54:12: Configuring 'Bathroom Downlight 1'
[... cut ...]
info  2021-06-05 17:54:13: Successfully configured 'Bathroom Downlight 1'
info  2021-06-05 17:54:13: Successfully setup reporting for '0x00158d0004768d5a' - 1 - genOnOff
debug 2021-06-05 17:54:13: Setup reporting for '0x00158d0004768d5a' - 1 - genLevelCtrl
info  2021-06-05 17:54:13: Successfully setup reporting for '0x00158d0004768d5a' - 1 - genLevelCtrl
debug 2021-06-05 17:54:13: Setup reporting for '0x00158d0004768d5a' - 1 - lightingColorCtrl
info  2021-06-05 17:54:14: Successfully setup reporting for '0x00158d0004768d5a' - 1 - lightingColorCtrl

bayview252, 2021-06-25:
HGZB-DLC4-N12B is also now being recognised as 3A12S-15.  Any thoughts on how to fix?  They have different features unfortunatly.

bayview252, 2021-06-25:
Nope,I'm not even a little bit as clever as I thought. Edit doesn't survive restart.
My Nue 12W RGBW Downlight has a controller that is labelled HGZB-DLC4-N12B, but returns the ModelId 'LXT56-LS27LX1.7'
There is a device already in the file for HGZB-DLC4-N12B but is only set to catch a ModelId of 'NUET56-DL27LX1.2'  Hopefully we can get this fixed in the next release. It's really just the Icon / Description in the Gui that is different.  Thanks for your work.

ozadr1an, 2021-08-02:
So just fired up 1.21.0 and problem still remains. Guess I'll have to write a custom extension.

closed 2021-09-10 by github-actions[bot]