Koenkk/zigbee2mqtt #7704: QS-Zigbee-C01 cover entity is unavailable after update

brizoo, 2021-06-05:
# Bug Report
## What happened
After last update my "cover" entity is marked as unavailable into HA. 
I'm using Lonsonho | QS-Zigbee-C01 modules which where working great until last update of both HA and Zigbee2mqtt. 
Now I can see only 4 entities instead of 5, and the ne missing is the cover. Which now makes the module useless since I cannot control the curtains anymore without the "cover" entity. 

I have tried to remove them and bind them again after having manually deleted all entities into HA and after a successful bind, I now only still have 4 entities without the "cover". 

## Debug Info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: CC2531
Adapter firmware version: latest

Logs here: https://hastebin.com/sutonabaku.apache
(name of the modules are "Volet Roulant TV" and "Volet Roulant Cuisine"

kasper1221, 2021-06-05:
Same here with the E1757 Ikea blinds after update, only 4 entities available.

but i can control them trough the dashboard manually.

Koenkk, 2021-06-05:
Indeed already fixed in 1.19.0

closed 2021-06-05 by Koenkk