Koenkk/zigbee2mqtt #7706: TUYA Light intensity sensor TS0222 problem, no reporting

HAwidmo, 2021-06-05:
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

sjorge, 2021-06-06:
I briefly had one of those before giving it away, I couldn't configure the reporting settings either. It just send the info when it felt like it (making it hard to get nice consistent graphs). 

Also it max out pretty quickly around 10k IIRC.

hankanman, 2021-06-12:
Facing the same issue here, can pair them and interview but z2m cannot configure them it seems, haven't seen either of the two I've got report any value other than lqi

mgrom, 2021-06-27:
Looks like more tuya devices are going to work that way so maybe it's time to figure it out :) I think the best way is to have possibility to replicate 100% identical communication that Tuya GW is sending to the device during configuration. I've spent some time on #7695 and till now no luck with reporting, but tuya gw is able to "force" the device to report.

btouchard, 2021-06-30:
Same problem
I have a CC2652RB, paring with TS0222 working fine, device appear on devices list.
But no one data on "expose" panel

[image]

Ben

elgerg, 2021-07-30:
I have a similar Moes Lux display and while i can get reporting working (looks like):
[image]

I get some proper random results:
[image]

[image]

Does anyone know what the reporting values should be to make it report the status correctly?

I also suspect that the battery percentage reporting just doesnt work as I did have what i thought was a low battery but it's reporting as 100%..

Any ideas?

Thanks!

iglu-sebastian, 2021-08-13:
I've got mine to reporting after trying to bind some random endpoints, then it suddenly started.
[image]

closed 2021-10-03 by github-actions[bot]