Koenkk/zigbee2mqtt #7712: Tamper info doesn't work with smoke detector Heiman HS1SA-E

foxDloulou, 2021-06-06:
Hello.

I just included my Heiman HS1SA-E smoke detector in zigbee2mqtt (conbee 2), and then in Jeedom (using plugin zigbeelinker)

The inclusion went well, I have all the info controls

## What happened
the tamper info never toggles when I undo the detector from its socket.
It still remains to be tamper: "Clear" in the zigbee2mqtt interface (Exposes column) and "tamper": false (State column) 
## What did you expect to happen
the state of this info should change when tampered
## How to reproduce it (minimal and precise)
undo the detector from its socket
## Debug info
Zigbee2MQTT version:     1.18.3 commit: 669489fe
Adapter hardware: Conbee II
Adapter firmware version: 0x26680700

sjorge, 2021-06-06:
@Koenkk I can confirm both my smoke and co alarms from HEIMAN don't seem to support tamper.

Actually 'test', 'trouble', 'ac_status' also don't seem to ever change, unlike my Develoco smoke alarm which does for example toggle 'test' to true when hitting the test button on the device, the HEIMAN's do not.

I dug a bit deaper to see where it gets these values from:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/e58938c8ca604e501458397243b694d094747442/converters/fromZigbee.js#L778-L794

Looks like these are generic ias_xxx from zigbee converters, I guess not all devices implement all those fields (or update them I guess).

I think the correct fix here is removing the tamper expose for the HEIMAN smoke and co alarms at least (those are the ones I have access too and could verify they don't seem to toggle tamper (or test, that one is easier to test hehe) I tried unscrewing from the base, shaking, pulling the battery and reinserting rapidly.

Koenkk, 2021-06-07:
@sjorge agree, the expose can be removed.

sjorge, 2021-06-07:
> @sjorge agree, the expose can be removed.

I'll do a PR later this week unless your or someone else gets to it first.

sjorge, 2021-06-07:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/2667

Koenkk, 2021-06-08:
Can this be closed?

closed 2021-06-08 by foxDloulou