Koenkk/zigbee2mqtt #7714: Error when install zigbee2mqtt

vuminh90, 2021-06-06:
Hello Sirs,
I have a problem when install zigbee2mqtt on Pi 3+
Cordinator: cc2350
Config:
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
  user: vuminh90
  password: minh1990
serial:
  port: /dev/ttyUSB0
advanced:
  rtscts: false
  channel: 16
  
This error is detail:
https://hastebin.com/onuzurijek.kotlin
https://hastebin.com/xisayijasa.sql
Please help me
Thanks!

wroadd, 2021-06-06:
I have a similar problem after update:
I’m sure I was on original PAN ID (6754) but after the update the PAN ID on the Coordinator reads as 6755 (I didn't change it, it was 6754 be for the update).

Koenkk, 2021-06-07:
You should set the `pan_id` to the `actual` one in your `configuration.yaml`

e.g.

```yaml
advanced:
  pan_id: 6755
```

wroadd, 2021-06-07:
> You should set the `pan_id` to the `actual` one in your `configuration.yaml`
> 
> e.g.
> 
> ```yaml
> advanced:
>   pan_id: 6755
> ```

OK, but it was 6754 before the update on stick and also on confit file, and it is still 6754(i downgrade back to1.18.3-1)
After I updated to 1.19.1-1 I got the message under z2m logs, that the PAN ID is on the stick 6755 (and is not matching with the PAN ID 6754 which is in the config file), but the PAN ID is still 6754 on the stick

Koenkk, 2021-06-07:
@wroadd the old z2m version does not read the pan id correctly, 1.19.1 fixes this.

wroadd, 2021-06-07:
> @wroadd the old z2m version does not read the pan id correctly, 1.19.1 fixes this.

Great, then all I need to do to change the PAN ID in config, when the service is down after update and restart it, right?

Koenkk, 2021-06-07:
yes

vuminh90, 2021-06-07:
> ```yaml
> pan_id: 6755
> ```

I have config panid:
advanced:
  rtscts: false
  channel: 16
  pan_id: 6756
  
  But i have error when start zigbee2mqtt as below:
  https://hastebin.com/ucuwubezin.yaml
  And i have try install zigbee2mqtt on NUC used Ubuntu 20.04
And i have this error????

Koenkk, 2021-06-07:
`permission denied, unlink '/opt/zigbee2mqtt/data/log/2021-06-07.04-00-02/log.txt'` permission error, fix permissions (e.g. `sudo chmod 777 -R /opt/zigbee2mqtt/data`

closed 2021-07-16 by github-actions[bot]