Koenkk/zigbee2mqtt #7716: After OTA update Ikea E1524/E1810 can't be bound to group

andreypolyak, 2021-06-06:
## What happened

1. I upgraded Ikea E1524/E1810 remote to the latest version using Zigbee2mqtt — 2.3.075
2. Deleted remote and readded it back (this was not related to OTA update so not sure how important is it to reproduce the issue)
3. Tried to setup the binding between the remote and light group as I previously had. It doesn't work

I read about the same issue in ZHA related issue: https://github.com/home-assistant/core/issues/43969#issuecomment-854951280. So looks like the problem is reproducable.

## What did you expect to happen

Is there any way to downgrade the firmware? I have bunch of E1524/E1810 remotes that are on 2.3.014 firmware and they are working great.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.19.1
Adapter hardware: CC2652
Adapter firmware version:

strobel1x, 2021-06-06:
I'm having the same issue also with the updated (to version 2.3.075) IKEA E1743 remote switches.

antonholmstedt, 2021-06-07:
Same problem for me, after updating E1743 to 2.3.075, build date 20210526. Force removed the remote, readded it and interview is complete, but after that no messages from clicks/actions are coming.

siklosi, 2021-06-07:
Same here  group id 0 after update. Tried removing and pairing again but can not bind to group. Otherwise works and I temporarily until resolved made script that toggles lights using from domoticz instead directly.

MattWestb, 2021-06-07:
deCONZ and ZHA can doing firmware downgrading of but need to knowing how to do it.

supercharley, 2021-06-07:
Same problem with the E1812!

martin131, 2021-06-07:
I have same problem on my E1812. No actions from button are recognized by Z2M in Home Assistant.

MattWestb, 2021-06-07:
Updated IKEA remotes is still sending all commands with unicast to 0x0000 = the coordinator if the coordinator is listen to the commands you is getting them ([sniffed with wireshark](https://github.com/zigpy/zigpy/discussions/660#discussioncomment-827229)).

Do one firmware downgrading with deCONZ or ZHA and you is getting the group functionality back or waiting some days and IKEA is very likely (but not 110% sure) putting one fixed firmware in the test feed.

ruttromp, 2021-06-08:
Have the same problem with the E1812 remote. Keeping an eye on this issue

MattWestb, 2021-06-08:
If some like downgrading with local file (possible with deCONZ and patched ZHA) is the 5 buttom remote and 2 button in the zip file attached.
[remotes.zip](https://github.com/Koenkk/zigbee2mqtt/files/6617997/remotes.zip)

shirou93, 2021-06-08:
Same issue IKEA E1812

AlexKay88, 2021-06-08:
Maybe too easy, but did you press a button on remote to wake it up during (re-)configuration?
Any error messages? Timeout?

shirou93, 2021-06-08:
> Maybe too easy, but did you press a button on remote to wake it up during (re-)configuration?
> Any error messages? Timeout?
> Checked if legacy: false for remote?

Repair not solve issue. Repair also.

heggink, 2021-06-10:
I noticed that, of my 5 IKEA remotes, only 1 is still bound to a group and working. All others are unbound (report group 0) even though I only upgraded 1 remote to the latest firmware. Whatever I try, I cannot even bind the previously working remotes to the original groups even though nothing changed in the firmware. It almost seems as if something changed in z2m rather than the firmware update causing it. My E1524 should be always bound to 901 and now that also reports group 0 :-(. No idea where I am going wrong.

MattWestb, 2021-06-10:
Request the cluster attributes "sw_build_id (id: 0x4000)" and you is getting the "real version" not what Z2M is "thinking".

With the updated firmware is not possible for Z2m to changing the group to 901 and the remote is not sending commands  to gropes only to the device with address 0x0000 = coordinator in your system.

Downgrading the firmware and you is getting the function back.

shirou93, 2021-06-10:
After upgrade firmware switch don't pair with oryginal IKEA gateway. This firmware is broken

brhahlen, 2021-06-10:
Yeah, I have the same problem here. After upgrade, no clicks are being registered anymore.

MattWestb, 2021-06-10:
@shirou93  Do you knowing if one original IKEA GW is supporting software downgrading ?
If you have it running with IKEA test firmware is very likely can do that but i dont knowing how to doing that.

deCONZ and ZHA with "puddly patch" can doing selecting downgrading if like going back to stable firmware.

heggink, 2021-06-10:
> Request the cluster attributes "sw_build_id (id: 0x4000)" and you is getting the "real version" not what Z2M is "thinking".
>
> With the updated firmware is not possible for Z2m to changing the group [...]

Bizarrely, I only upgraded one device but now find that I have 2 on old FW and 3 actually on new. I never upgraded them myself.  Does z2m upgrade FW itself?

Also, any advice on how to easiest downgrade the FW?

shirou93, 2021-06-10:
> @shirou93 Do you knowing if one original IKEA GW is supporting software downgrading ?
> If you have it running with IKEA test firmware is very likely can do that but i dont knowing how to doing that. [...]

I dont have ikea gateway. I tested in shop and employe change this switch for another one.

MattWestb, 2021-06-10:
@shirou93  If you is family member you can always do that without any reason and infinity time in most country and after that i can baying it for 3€ in the recycle hub :-))

Dont forgetting disabling the test feed until its reported being stable !!!!

shirou93, 2021-06-10:
> @shirou93 If you is family member you can always do that without any reason and infinity time in most country and after that i can baying it for 3€ in the recycle hub :-))
> 
> Dont forgetting disabling the test feed until its reported being stable !!!!

:) I disabled IKEA test server in config. In stable server no ota update info.

MattWestb, 2021-06-10:
deCONZ only picking the old file in the file dialogue and upgrading and waking the remote so it can getting the command.

ZHA [OTA-Device-Firmware-Updates](https://github.com/zigpy/zigpy/wiki/OTA-Device-Firmware-Updates#downgrading-your-device-firmware).

Z2M putt one future request for implanting it.

@shirou93 Then you was getting one with updated firmware so no need updating it and you can putting your groups back in it :-)))

heggink, 2021-06-10:
@MattWestb Thx!. The one thing I don;t get is that the E1524 no longer works even though the FW is not updated.

MattWestb, 2021-06-10:
@heggink I think its upgraded butt Z2M is not reporting it OK.
Perhaps restart Z2M and doing one new attribute report request and see is its updated or not.

heggink, 2021-06-10:
@MattWestb that's what I thought as well but then z2m would have updated it without my involvement. After restarting z2m, it still reports as FW 2.3.014 which is, according to the normal ikea FW site, the current version:

  {
    "fw_binary_url": "http://fw.ota.homesmart.ikea.net/global/GW1.0/01.14.024/bin/159699-5.1-TRADFRI-remote-control-2.3.014.ota.ota.signed",
    "fw_file_version_LSB": 17969,
    "fw_file_version_MSB": 8961,
    "fw_filesize": 182590,
    "fw_image_type": 4545,
    "fw_manufacturer_id": 4476,
    "fw_type": 2
  },

MattWestb, 2021-06-10:
Then i think the best is deleting it in Z2M and resetting the remote and doing one new pairing.

If messing up the binding its normally not possible getting it back OK without repairing /resetting the device.

Giancky79, 2021-06-10:
i'm try to bind with lamp and working, with ota we don't need pairing with a group? i have a E1743

MattWestb, 2021-06-10:
The E1743 is one dimmer switch = shall sending command to groups of lights and need doing it with broadcast (=Zigbee groups).
Before it was not possible binding the remote to one light only groups and I have not testing if its possible with the test version.
The test version is sending all commands to 0x0000 = the coordinator in you mesh.

strobel1x, 2021-06-10:
> deCONZ and ZHA with "puddly patch" can doing selecting downgrading if like going back to stable firmware.

I am trying the "ZHA with puddly patch" option to downgrade, but with no success yet:
- the TRADFRI on/off switch (E1743) does not stop to flash (after 100% and SUCCESS it starts again) and still reports the test-version (2.3.075)
- the TRADFRI remote control (E1524) flashes with SUCCESS, but still reports the test-version (2.3.075)
I did flash the latest stable versions (2.2.010 for E1743, 2.3.014 for E1524) downloaded from the test server, completely remove, reset (even with battery removed) and re-pair.

Did anyone successfully downgrade from this test-version yet?

MattWestb, 2021-06-10:
I have don many lights and also the wireless dimmer.
What is the log saying the version the in the device and the version it staring downloading (on of the first lines then starting  the OTA requests) ?

Best is disabling the normal IKEA OTA and only using the local file you like so cant getting wrong version upgraded.

strobel1x, 2021-06-10:
> I have don many lights and also the wireless dimmer.
> What is the log saying the version the in the device and the version it staring downloading (on of the first lines then starting the OTA [...]

I only have the logs of the remote control (E1524) at hand.
At the beginning it says:
```
OTA query_next_image handler for 'IKEA of Sweden TRADFRI remote control': field_control=1, manufacture_id=4476, image_type=4545, current_file_version=587683377, hardware_version=1
OTA image version: 587286065, size: 182590. Update needed: True
Updating: IKEA of Sweden TRADFRI remote control
```
At the end:
```
OTA upgrade_end handler for 'IKEA of Sweden TRADFRI remote control': status=Status.SUCCESS, manufacturer_id=4476, image_type=4545, file_version=587286065
```
I only have the local OTA folder enabled in ZHA (containing only the two OTA files), no update URL.
z2m is still reporting the test-version (2.3.075) and I keep having the group-binding issue.

MattWestb, 2021-06-10:
I is doing one fast test with one 5 button remote.
Its taking some hours but its have started downloading the file.
```
2021-06-10 21:44:16 DEBUG (MainThread) [zigpy.zcl] [0x3d53:1:0x0019] OTA query_next_image handler for 'IKEA of Sweden TRADFRI remote control': field_control=1, manufacture_id=4476, image_type=4545, current_file_version=587683377, hardware_version=1
2021-06-10 21:44:16 DEBUG (MainThread) [zigpy.zcl] [0x3d53:1:0x0019] OTA image version: 587286065, size: 182590. Update needed: True
2021-06-10 21:44:16 INFO (MainThread) [zigpy.zcl] [0x3d53:1:0x0019] Updating: IKEA of Sweden TRADFRI remote control
2021-06-10 21:44:26 DEBUG (MainThread) [zigpy.zcl] [0x3d53:1:0x0019] OTA image_block handler for 'IKEA of Sweden TRADFRI remote control': field_control=0, manufacturer_id=4476, image_type=4545, file_version=587286065, file_offset=0, max_data_size=63, request_node_addr=Noneblock_request_delay=None
2021-06-10 21:44:26 DEBUG (MainThread) [zigpy.zcl] [0x3d53:1:0x0019] OTA upgrade progress: 0.0
```

MattWestb, 2021-06-10:
Half time :-)))
```
2021-06-10 22:40:01 DEBUG (MainThread) [zigpy.zcl] [0x3d53:1:0x0019] OTA image_block handler for 'IKEA of Sweden TRADFRI remote control': field_control=0, manufacturer_id=4476, image_type=4545, file_version=587286065, file_offset=91480, max_data_size=63, request_node_addr=Noneblock_request_delay=None
2021-06-10 22:40:01 DEBUG (MainThread) [zigpy.zcl] [0x3d53:1:0x0019] OTA upgrade progress: 50.1
```

MattWestb, 2021-06-10:
Looks like the device is downloading the file OK and replaying it was done OK and rebooting and rejoining the system = OK
```
2021-06-10 23:34:48 DEBUG (MainThread) [zigpy.zcl] [0x3d53:1:0x0019] OTA image_block handler for 'IKEA of Sweden TRADFRI remote control': field_control=0, manufacturer_id=4476, image_type=4545, file_version=587286065, file_offset=182560, max_data_size=63, request_node_addr=Noneblock_request_delay=None
2021-06-10 23:34:48 DEBUG (MainThread) [zigpy.zcl] [0x3d53:1:0x0019] OTA upgrade progress: 100.0
2021-06-10 23:34:48 DEBUG (MainThread) [zigpy.zcl] [0x3d53:1:0x0019] OTA upgrade_end handler for 'IKEA of Sweden TRADFRI remote control': status=Status.SUCCESS, manufacturer_id=4476, image_type=4545, file_version=587286065
```
But its still having the test firmware and like getting one update to the old :-((
```
2021-06-10 23:36:29 DEBUG (MainThread) [zigpy.zcl] [0x3d53:1:0x0019] OTA query_next_image handler for 'IKEA of Sweden TRADFRI remote control': field_control=1, manufacture_id=4476, image_type=4545, current_file_version=587683377, hardware_version=1
2021-06-10 23:36:29 DEBUG (MainThread) [zigpy.zcl] [0x3d53:1:0x0019] OTA image version: 587286065, size: 182590. Update needed: True
2021-06-10 23:36:29 INFO (MainThread) [zigpy.zcl] [0x3d53:1:0x0019] Updating: IKEA of Sweden TRADFRI remote control
```
I letting it eating batteries over the night and see how its working (if the batteries is surviving it) tomorrow.

I can being the the firmware protection that is implanted in the blinds and signal repeater is blocking the old firmware to being flashed but the firmware is having OK signature and normally being flashed OK and the device ave replaying it success.

strobel1x, 2021-06-11:
> I can being the the firmware protection that is implanted in the blinds and signal repeater is blocking the old firmware to being flashed but the firmware is having OK signature and normally being flashed OK and the device ave replaying it success.

I'm having exactly the same effect (firmware OTA update SUCCESS, but after rejoining reporting again the previous (test-) firmware. My remotes are directly connected with the coordinator (CC2531) - no repeaters.

strobel1x, 2021-06-11:
Maybe this is not a bug, but a feature:
Instead of binding the remote switch (E1743) to one group of two lamps (FLOALT panels), I am able to bind the remote switch directly to each of the lamps.
I've never tried that before, but according to the z2m [documentation](https://www.zigbee2mqtt.io/devices/E1743.html#binding), that should not have worked before.

MattWestb, 2021-06-11:
The zigbee class of controlling devices hall being able sending  commands with braodcast = zigbee groups. Possibility sending unicast is also possible but normally not used with light controllers then its not making any sin then all with lights is groups also with touch linking and setting up the network.

And its receiving the the binding command and replaying back success but is not doing the real binding inside itself (sniffed with wireshark) https://github.com/zigpy/zigpy/discussions/660#discussioncomment-827229.

My remote is still living and have doing 75% af the last try (4 or 5 time then i was sleeping) and its still on test FW. Then its finises i trying  downgrading to one 1.2 firmware from 2018 but i think its not working because the firmware protection but need to trying.

I was thinking if the problems for some weeks ago then IKEA was roiling out the blinds and signal repeater but it was working OK after first test feed fixed version it was certified and released and looks working (user have doing positive feed back).

heggink, 2021-06-11:
> Maybe this is not a bug, but a feature:
> Instead of binding the remote switch (E1743) to one group of two lamps (FLOALT panels), I am able to bind the remote switch directly to each of the lamps. [...]

Brilliant!!! Just did the same and it works wonderful without having to downgrade anything. Thanks!!

heggink, 2021-06-11:
Latest stable E1524 (dimmer) remote FW also works the same: I can directly connect it to a bunch of devices instead of a group.

clyso, 2021-06-11:
I can confirm @heggink findings: binding the remote directly to the device (tested with ikea buld and osram power plug) instead of the group works as expected. thanks @strobel1x for this brilliant idea. i find this way much better as I don't need to create groups for each single device bound to one control. thanks @heggink for trying out
what i also noticed is that the led disappeared and joining works only after resetting control

MattWestb, 2021-06-11:
Looks being one good work around but its very likely not lasting then the firmware need being fixed then the group binding must working for light groups.

Large light network cant being controlled with unicast it must being done with braodcast (group sending) or the network is collapsing because of router discovery communications for every single command sent to etch device.

I have starting downgrading with one 1.2 and see is the test is accepting it or not. In 2 hours we is knowing if its working or not (i think not).

MattWestb, 2021-06-11:
Undocumented future i like (=Bugs). 
:-)))

MattWestb, 2021-06-11:
@strobel1x If you like getting the "default scenes" that is controlled with the 5 button remote you can downgrading IKEA lights and then upgrading then and the firmware is restoring them and is working with your remote until you is resetting the light but  you need adding the light to the "magic group" for getting it working :-))

mateusz-bak, 2021-06-11:
How do you successfully bind e1743 to a bulb?
I select my button, go to Bind tab, add the LED as destination, select OnOff and LevelCtrl, click Bind and all i get is errors.
E1743 is on 2.2.010

[image]

Log:
`debug Received MQTT message on 'zigbee2mqtt/bridge/request/device/bind' with data '{"clusters":["genOnOff","genLevelCtrl"],"from":"IKEA_E1743_kitchen_1/1","to":"IKEA_LED1836G9_kitchen_1/1","transaction":"4shzx-14"}'`

`debug binding cluster 'genOnOff' from 'IKEA_E1743_kitchen_1' to 'IKEA_LED1836G9_kitchen_1'`

`error Failed to bind cluster 'genOnOff' from 'IKEA_E1743_kitchen_1' to 'IKEA_LED1836G9_kitchen_1' (Error: Bind 0x847127fffec986aa/1 genOnOff from '0x5c0272fffeaaf4c1/1' failed (AREQ - ZDO - bindRsp after 10000ms))`

`info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genOnOff","from":"IKEA_E1743_kitchen_1","to":"IKEA_LED1836G9_kitchen_1"},"type":"device_bind_failed"}'`

`debug binding cluster 'genLevelCtrl' from 'IKEA_E1743_kitchen_1' to 'IKEA_LED1836G9_kitchen_1'`

`error Failed to bind cluster 'genLevelCtrl' from 'IKEA_E1743_kitchen_1' to 'IKEA_LED1836G9_kitchen_1' (Error: Bind 0x847127fffec986aa/1 genLevelCtrl from '0x5c0272fffeaaf4c1/1' failed (AREQ - ZDO - bindRsp after 10000ms))`

`info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"genLevelCtrl","from":"IKEA_E1743_kitchen_1","to":"IKEA_LED1836G9_kitchen_1"},"type":"device_bind_failed"}'`

`info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/bind', payload '{"data":{"clusters":[],"failed":["genOnOff","genLevelCtrl"],"from":"IKEA_E1743_kitchen_1/1","to":"IKEA_LED1836G9_kitchen_1/1"},"error":"Failed to bind","status":"error","transaction":"4shzx-14"}'`

`error Failed to bind`

heggink, 2021-06-12:
I think you need to update that firmware to the latest beta. The 2.2 release does not support direct device binding

BuildTheRobots, 2021-07-04:
> Maybe this is not a bug, but a feature:
> Instead of binding the remote switch (E1743) to one group of two lamps (FLOALT panels), I am able to bind the remote switch directly to each of the lamps. [...]

It's quite nice being able to bind a switch to multiple devices  - and I thought after rebinding everything I'd fixed any issues that this firmware brought me, but there's a hidden gotcha:

The switches now seem to be keeping track of their state and not broadcasting multiple events. EG, if you're looking at the mqtt and press the off switch on the remote, you see an off message. No matter how many times you press off again, you don't see another button press event. Same goes for switching it on; it broadcasts the first on event, but no subsiquent ones.

As I'd just set up node-red to detect double and tripple taps on the buttons (eg double tap on the bed-side button switches off all lights in the house), this is really annoying as it's completely broken things. If anyone has thoughts on getting the old (mqtt for every press) behaviour back, i'd greatly appreciate it.

brhahlen, 2021-07-13:
Did anyone fix this somehow? I have one remote (config mishap) that I accidentally put on the Test OTA (Firmware build date: 20190401; Firmware version 2.3.075)
It just won't work anymore now... Can pair it, but no button clicks are registering at all...

MattWestb, 2021-07-13:
IKEA have blocking rolling back the firmware in there controlling so its not possible doing "Over The Air".

They have not releasing any new or moving the test to  production but they have certificated 3 device by Zigbee Alliance for some weeks ago.

Still recommending waiting and see wot they is doing in some weeks.

In extreme cases its possible flashing one older version with SWD probe but for the 5 button remote you must destroying it for doing it but the rest is possible without "hard tools".

KF042020, 2021-07-19:
> In extreme cases its possible flashing one older version with SWD probe but for the 5 button remote you must destroying it for doing it but the rest is possible without "hard tools".

Hi,
Have you tried this with the shortcut or On/Off button? If so can you link some description, please?

MattWestb, 2021-07-19:
The E1743 On/Off button is easy flashing if you have one SWD probe and can shouldering some wires and you is not destroying the device / case.
[image] and for flashing and dumping IKEA firmware https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module.

Morphy99, 2021-08-05:
I've noticed at some point all my Ikea remotes (E1524) are no longer bound to my bulbs. They're on fw 2.3.014.  When HA group discovery was added in the latest update I removed all my groups forgetting the binds with the remotes. Since then I've been trying to get them to bind again with no success. I even rolled back to before deleting the groups and they still don't work. 

I've just tried removing one remote and re-pairing and the binds look different to the others:
[image]
[image]

So I think the default_bind_group was still there and after removing it, they bound to the group OK.

MattWestb, 2021-08-14:
Some new is that IKEA was taking the "bad firmware" from the test feed for some week ago so they is knowing its not working OK.

And they have certified the new version 2.3.079 of the 5 button remote, On/Off, Open/Close and the blinds.
Its still not in the test feed but very likely is coming there in one week or 2.

Shortcut and Symphonosik was also getting "bad ones" but still no new OTA files have showing up in the feed or certificate documents.

Have little patios with testing them then they is showing up for not braking your devices and i is trying them out then i can getting the new OTA files for the remotes but i dont have any blinds but they was working OK after the last update for 2 month ago.

I keep posting the status of the new remotes OTA files here for user that have getting in trouble.

MattWestb, 2021-09-14:
All "bad" firmware have getting updates certified but they is not released in the test feed but is likely being made very soon.
https://zigbeealliance.org/zigbee_products/?product_type=certified_product&se=ikea

cpuks, 2021-09-24:
I've got this issue on E1812 - firmware 2.3.075 and CC2652 - pairing is not a problem but then I cannot get anything out of this shortcut. Is it a matter of waiting for new firmware?

havardthom, 2021-10-30:
My E1524 IKEA trådfri remote stopped working a few days ago, it does not communicate with my light group anymore. I have a light group with 2 trådfri drivers (kitchen) bind with the IKEA trådfri remote. The firmware on my devices seem to be quite old:
[image]
As you can see I do not have the new firmware, but still the E1524 cannot communicate with the drivers (I can control the drivers directly/manually).

I've tried binding the E1524 remote both to the device group and directly to the two driver devices, nothing seem to work anymore. 

I've tried to check the devices for newest firmware, so I can update and connect directly to devices as reported in this issue, but I get  `Device didn't respond to OTA request` error even though my configuration has `ikea_ota_use_test_url: true`

```
Debug Received MQTT message on 'zigbee2mqtt/bridge/request/device/ota_update/check' with data '{"id":"Kjøkken Lyskontroll","transaction":"7icik-1"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'Kjøkken Lyskontroll'","meta":{"device":"Kjøkken Lyskontroll","status":"checking_if_available"},"type":"ota_update"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'Kjøkken Lyskontroll' (Device didn't respond to OTA request)","meta":{"device":"Kjøkken Lyskontroll","status":"check_failed"},"type":"ota_update"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"Kjøkken Lyskontroll"},"error":"Failed to check if update available for 'Kjøkken Lyskontroll' (Device didn't respond to OTA request)","status":"error","transaction":"7icik-1"}'
```

Anyone have suggestions what I can do? Can I upgrade device firmware manually somehow?

MattWestb, 2021-10-30:
I have not testing all new controllers but the E1743 (on/off) have loosing group binding (and likely all other new device firmware) and can only being bonded too one device, that is very bad the its not possible having more lights/derives that is working then the host system is down / remote.

Also user that have upgraded to the bad test firmware cant upgrading to the new version then its rolling the bad back on reboot and downgrading is blocked in the firmware.

I have flashing 5 E1743 with one old dumped firmware and and then its updating to the last production one OK (but still loosing the group binding) but shall being OK then new firmware is released.

I trying extracting the last good firmware and repacking it to S37 files so user can manual flashing there device with Simplicity commander with one SWD probe without braking them, but i need more testing for getting it working OK.
I keep posting then i finding more interesting things.

cpuks, 2021-10-30:
So if my E1743 are on .75 firmware which is considered bad - what can I do to upgrade those to .79 ? All I have in mind now is to try with IKEA gateway see if that helps...

muddy79, 2021-10-30:
Just for the record: I've updated some of my E1524 to "stable" 2.3.080 and I lost group binding capability. What a shame...
It seems "test" feed still supplies 2.3.014. Is there any way to force downgrade to the older version via z2m? I've updated z2m configuration to use test feed for IKEA OTA but OTA is not possible due to the error message: firmware version is newer than OTA version.

havardthom, 2021-10-30:
I am currently upgrading my E1524 remote to v2.3.080 through deconz, it's taking forever though, currently at 8% after 15 minutes. Hopefully I'll get it working again after the update. It seems that the ikea test feed (http://fw.test.ota.homesmart.ikea.net/feed/version_info.json) only had v2.3.014 which is probably why Zigbee2mqtt couldn't find the newest firmware. The stable feed (http://fw.ota.homesmart.ikea.net/feed/version_info.json) has v.2.3.080 though and luckily deconz had that version downloaded in it's `/data/ota` folder

havardthom, 2021-10-30:
> Just for the record: I've updated some of my E1524 to "stable" 2.3.080 and I lost group binding capability. What a shame... It seems "test" feed still supplies 2.3.014. Is there any way to force [...]

I had v2.3.014 on my E1524 and could not communicate with my light group anymore, so I don't think downgrading helps. It seems you can connect directly with the devices on the newest firmware though.

[121 more replies]

closed 2022-02-13 by github-actions[bot]