Koenkk/zigbee2mqtt #7720: Error. Failed to call 'Configure' after upgrading to v.1.19.1

criadoperez, 2021-06-07:
## What happened
Error after upgrading from Zigbee2mqtt v.1.18.3 to v.1.19.1

## What did you expect to happen
Application to keep working. Instead, now you get the following error after starting up Zigbee2mqtt. Using v.1.18.3 it does not happen.
```
error 2021-06-07 13:09:55: Failed to call 'Configure' 'onZigbeeStarted' (TypeError: zhc.getConfigureKey is not a function
    at Configure.shouldConfigure (/opt/zigbee2mqtt/lib/extension/configure.js:46:59)
    at Configure.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/configure.js:104:22)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:386:44))
```

## How to reproduce it (minimal and precise)
Start zigbee2mqtt v.19.1

## Debug info
Zigbee2MQTT version: 1.19.1 commit #9bd46932
Adapter hardware: Slaesh's CC2652RB stick
Adapter firmware version: revision": Coordinator version 20210120
Hardware: Raspberry pi 4 - 4GB
OS: Linux Raspberry pi OS

Koenkk, 2021-06-07:
You are using an incorrect zigbee-herdsman-converters. Make sure to remove the manual overridden one.

closed 2021-06-07 by Koenkk

criadoperez, 2021-06-07:
Thank you @Koenkk for such a fast reply. I'm glad to see this is not a general bug and the issue is closed.
However I don't understand you answer. Could you be so kind to elaborate a bit more or point me in the right direction? If I'm overriding anything manually, I'm not aware of it.
Thanks in advance.

Koenkk, 2021-06-07:
After doing the upgrade, did you execute `npm ci` in the z2m folder?

criadoperez, 2021-06-07:
That was it, thank you.
I was updating with a "git pull" instead of using the script "update.sh" which does the pull, the npm ci and a few other things. I'll stick to using the script from now on.
Thanks a million.