Koenkk/zigbee2mqtt #7726: Philips Hue wall switch module right_press not detected

rorie1982, 2021-06-07:
# Bug Report
## What happened
The Philips Hue wall switch module has two inputs to connect a wall switch. Pressing the left switch generates a message  with a "action":"left_press_release" or "action":"left_press". Nothing happens when pressing the right switch. 
## What did you expect to happen
A mqtt message is send with "action":"right_press_release" or "action":"right_press".
## How to reproduce it (minimal and precise)
Pressing the right switch.
## Debug Info
Zigbee2MQTT version:
Adapter hardware: Texas Instruments LAUNCHXL-CC26X2R1
Adapter firmware version: 1.18.2

JDFS404, 2021-06-08:
Having the same problem with one (of five) Wall Switch Module, `right_press` is not working and changing `device_mode` to `dual_push_button` is also not working.

rorie1982, 2021-06-08:
So the other four are working? Maybe it is a defect module?

JDFS404, 2021-06-08:
No, got this message earlier and after upgrading and losing my devices `right_press` was working till yesterday.

rorie1982, 2021-06-08:
To summarize, you updated to the latest version. After the update your lost all your devices, some how you get it working again and after that the wall switch stopped sending the right pressed command? Before your upgrade the wall switch was working as expected.

JDFS404, 2021-06-08:
I was on version 1.18.3 and everything was working fine. Then Docker / Unraid auto updated my Zigbee2MQTT and when I opened Z2M the next day, all my devices were gone. So had to add them again (i.e. physically removing five wall switches, which was a major pain in the ass). Next day I had to do it again... So I re-flashed my CC2531 and now I don't have any more problems, except for the `right_press` and `right_press_hold` and I think `right_press_hold_release` which are not working.

So een ezel stoot zich niet twee keer aan dezelfde steen and I will stay on 1.18.3 for a long time now.

rorie1982, 2021-06-09:
Updating to the latest version 1.19.1 and setting the device mode to dual_rocker fixed the issue for me.

closed 2021-06-09 by rorie1982